srsran/srsRAN_4G #673: Unable to connect UE with eNodeB using Docker Compose

ShubhamTatvamasi, 2021-05-20:
I am trying to run these step inside docker but it's failing.
https://docs.srslte.com/en/latest/app_notes/source/zeromq/source/index.html

```yaml
version: "3.9"

services:
  srsenb:
    image: shubhamtatvamasi/srsran-demo
    container_name: srsenb
    ipc: shareable
    cap_add:
      - SYS_NICE
    networks:
      srs_net:
        ipv4_address: 172.50.50.10
    command:
      - srsenb
      - --enb.mme_addr=192.168.60.142
      - --enb.gtp_bind_addr=172.50.50.10
      - --enb.s1c_bind_addr=172.50.50.10
      - --rf.device_name=zmq
      - --rf.device_args="fail_on_disconnect=true,tx_port=tcp://*:2000,rx_port=tcp://0.0.0.0:2001,id=enb,base_srate=23.04e6"

  srsue:
    image: shubhamtatvamasi/srsran-demo
    container_name: srsue
    depends_on: 
      - srsenb
    pid: service:srsenb
    ipc: service:srsenb
    cap_add:
      - NET_ADMIN
      - SYS_NICE
    devices:
      - /dev/net/tun
    networks:
      srs_net:
        ipv4_address: 172.50.50.11
    command:
      - srsue
      - --usim.algo=milenage
      - --usim.opc=63BFA50EE6523365FF14C1F45F88737D
      - --rf.device_name=zmq
      - --rf.device_args="tx_port=tcp://*:2001,rx_port=tcp://0.0.0.0:2000,id=ue,base_srate=23.04e6"

networks:
  srs_net:
    ipam:
      driver: default
      config:
        - subnet: "172.50.50.0/24"
```

```
Attaching to srsenb, srsue
srsenb    | ---  Software Radio Systems LTE eNodeB  ---
srsenb    | 
srsenb    | Reading configuration file /etc/srsran/enb.conf...
srsenb    | 
srsenb    | Built in Release mode using 21.04.0.
srsenb    | 
srsenb    | Opening 1 channels in RF device=zmq with args="fail_on_disconnect=true,tx_port=tcp://*:2000,rx_port=tcp://0.0.0.0:2001,id=enb,base_srate=23.04e6"
srsue     | Reading configuration file /etc/srsran/ue.conf...
srsue     | 
srsue     | Built in Release mode using 21.04.0.
srsue     | 
srsue     | Opening 1 channels in RF device=zmq with args="tx_port=tcp://*:2001,rx_port=tcp://0.0.0.0:2000,id=ue,base_srate=23.04e6"
srsue     | CHx base_srate=23.04e6"
srsue     | CHx id=ue
srsue     | Current sample rate is 1.92 MHz with a base rate of 23.04 MHz (x12 decimation)
srsue     | CH0 rx_port=tcp://0.0.0.0:2000
srsue     | CH0 tx_port=tcp://*:2001
srsenb    | CHx base_srate=23.04e6"
srsenb    | CHx id=enb
srsenb    | Current sample rate is 1.92 MHz with a base rate of 23.04 MHz (x12 decimation)
srsenb    | CH0 rx_port=tcp://0.0.0.0:2001
srsenb    | CH0 tx_port=tcp://*:2000
srsenb    | CH0 fail_on_disconnect=true
srsenb    | Current sample rate is 11.52 MHz with a base rate of 23.04 MHz (x2 decimation)
srsenb    | Current sample rate is 11.52 MHz with a base rate of 23.04 MHz (x2 decimation)
srsenb    | Setting frequency: DL=2680.0 Mhz, UL=2560.0 MHz for cc_idx=0 nof_prb=50
srsenb    | 
srsenb    | ==== eNodeB started ===
srsenb    | Type <t> to view trace
srsenb    | Closing stdin thread.
srsue     | Waiting PHY to initialize ... done!
srsue     | Attaching UE...
srsue     | Closing stdin thread.
```

It worked if I run them on same docker image.

jwijenbergh, 2021-05-20:
Hello,

I am using the following docker-compose file on srsLTE 20.10.1 with zeromq: https://github.com/jwijenbergh/srslte-docker-emulated/blob/master/docker-compose.yml

Do note that the Dockerfile points to my current fork (see https://github.com/jwijenbergh/srslte-docker-emulated/blob/513d62a1c5421ea3b61c4141d0e8c6d9da5854ab/Dockerfile#L32) so you might need to change this. Hope this helps

ShubhamTatvamasi, 2021-05-20:
Cool, I have update the compose file. got new logs:
```
srsenb    | ---  Software Radio Systems LTE eNodeB  ---
srsenb    | 
srsenb    | Reading configuration file /etc/srsran/enb.conf...
srsue     | Reading configuration file /etc/srsran/ue.conf...
srsue     | 
srsenb    | 
srsenb    | Built in Release mode using 21.04.0.
srsenb    | 
srsue     | Built in Release mode using 21.04.0.
srsue     | 
srsue     | Opening 1 channels in RF device=zmq with args="tx_port=tcp://172.50.50.11:2001,rx_port=tcp://172.50.50.10:2000,id=ue,base_srate=23.04e6"
srsue     | CHx base_srate=23.04e6"
srsue     | CHx id=ue
srsue     | Current sample rate is 1.92 MHz with a base rate of 23.04 MHz (x12 decimation)
srsue     | CH0 rx_port=tcp://172.50.50.10:2000
srsue     | CH0 tx_port=tcp://172.50.50.11:2001
srsenb    | Opening 1 channels in RF device=zmq with args="fail_on_disconnect=true,tx_port=tcp://172.50.50.10:2000,rx_port=tcp://172.50.50.11:2001,id=enb,base_srate=23.04e6"
srsenb    | CHx base_srate=23.04e6"
srsenb    | CHx id=enb
srsenb    | Current sample rate is 1.92 MHz with a base rate of 23.04 MHz (x12 decimation)
srsenb    | CH0 rx_port=tcp://172.50.50.11:2001
srsenb    | CH0 tx_port=tcp://172.50.50.10:2000
srsenb    | CH0 fail_on_disconnect=true
srsenb    | Current sample rate is 11.52 MHz with a base rate of 23.04 MHz (x2 decimation)
srsenb    | Current sample rate is 11.52 MHz with a base rate of 23.04 MHz (x2 decimation)
srsenb    | Setting frequency: DL=2680.0 Mhz, UL=2560.0 MHz for cc_idx=0 nof_prb=50
srsenb    | 
srsenb    | ==== eNodeB started ===
srsenb    | Type <t> to view trace
srsenb    | Closing stdin thread.
srsue     | Waiting PHY to initialize ... done!
srsue     | Attaching UE...
srsue     | Closing stdin thread.
srsue     | Current sample rate is 1.92 MHz with a base rate of 23.04 MHz (x12 decimation)
srsue     | Current sample rate is 1.92 MHz with a base rate of 23.04 MHz (x12 decimation)
srsue     | .
srsue     | Found Cell:  Mode=FDD, PCI=1, PRB=50, Ports=1, CFO=-0.2 KHz
srsue     | Current sample rate is 11.52 MHz with a base rate of 23.04 MHz (x2 decimation)
srsue     | Current sample rate is 11.52 MHz with a base rate of 23.04 MHz (x2 decimation)
srsue     | Found PLMN:  Id=00101, TAC=7
srsenb    | RACH:  tti=6931, cc=0, preamble=34, offset=0, temp_crnti=0x46
srsenb    | Disconnecting rnti=0x46.
srsue     | Attach failed (attempt 1/5)
srsue     | Current sample rate is 1.92 MHz with a base rate of 23.04 MHz (x12 decimation)
srsue     | Current sample rate is 1.92 MHz with a base rate of 23.04 MHz (x12 decimation)
srsue     | .
srsue     | Found Cell:  Mode=FDD, PCI=1, PRB=50, Ports=1, CFO=-0.2 KHz
srsue     | Current sample rate is 11.52 MHz with a base rate of 23.04 MHz (x2 decimation)
srsue     | Current sample rate is 11.52 MHz with a base rate of 23.04 MHz (x2 decimation)
srsue     | Found PLMN:  Id=00101, TAC=7
srsue     | Random Access Transmission: seq=21, tti=8341, ra-rnti=0x2
srsenb    | RACH:  tti=8341, cc=0, preamble=21, offset=0, temp_crnti=0x47
srsue     | RRC Connected
srsue     | Random Access Complete.     c-rnti=0x47, ta=0
srsue     | Network attach successful. IP: 192.168.128.17
srsenb    | User 0x47 connected
```

ShubhamTatvamasi, 2021-05-20:
But I am not able to ping the IP

jwijenbergh, 2021-05-20:
Do you not have the core network (srsepc) in the docker-compose file? Pinging the UE ip can be done with `docker exec -i -t virtual-srsepc ping 172.16.0.2`. Also see https://github.com/jwijenbergh/srslte-docker-emulated/blob/master/start.sh

ShubhamTatvamasi, 2021-05-20:
I don't have srsepc, I am using Magma as my Core Network.
Updating my code on this repo:
https://github.com/ShubhamTatvamasi/srsRAN-demo

ShubhamTatvamasi, 2021-05-20:
Magma's Access Gateway is on VirtualBox.
I am able to ping Access Gateway's using VirtualBox IP. Like this:
```bash
docker exec -it srsue ping 192.168.129.1
```

But the new Attached UE IP is inside container. How to access that?

ShubhamTatvamasi, 2021-05-21:
So I switched the network to host and it's working now. Thanks for your help @jwijenbergh :)

closed 2021-05-21 by ShubhamTatvamasi

jwijenbergh, 2021-05-22:
No problem @ShubhamTatvamasi. Switching to network host is indeed a good idea if not everything is inside the docker compose file. Good that it works now!

ping2balaji, 2021-06-01:
Hi @ShubhamTatvamasi , i am looking to test magma LTE-Core with srsran ue/enb simulation. Could you pls provide some insights on the same? i saw your branch on this demo. Just like to know where to punch-in mme-ip/gtp-ip in srsran to successfully test a simple attach/detach call. Any help highly appreciated.

ShubhamTatvamasi, 2021-06-01:
> Hi @ShubhamTatvamasi , i am looking to test magma LTE-Core with srsran ue/enb simulation. Could you pls provide some insights on the same? i saw your branch on this demo. Just like to know where to [...]

You can update that here: https://github.com/ShubhamTatvamasi/srsRAN-demo/blob/master/docker-compose.yaml

shalinjshah18, 2021-07-12:
> So I switched the network to host and it's working now. Thanks for your help @jwijenbergh :)

@Shubham can you please describe what it means as Switch the network to host?