srsran/srsRAN_4G #675: Incoordination in function declaration and implementation.

cld4h, 2021-05-27:
## Issue Description ##
Incoordination in function declaration and implementation.

In function `srsran_predecoding_diversity_multi`, 
the [implementation](https://github.com/srsran/srsRAN/blob/89f16eed2b7f2e6c45e675298f5829c230698e1d/lib/src/phy/mimo/precoding.c#L783
) is:

```c
int srsran_predecoding_diversity_multi(cf_t*  y[SRSRAN_MAX_PORTS],
                                       cf_t*  h[SRSRAN_MAX_PORTS][SRSRAN_MAX_PORTS],
                                       cf_t*  x[SRSRAN_MAX_LAYERS],
                                       float* csi[SRSRAN_MAX_CODEWORDS],
                                       int    nof_rxant,
                                       int    nof_ports,
                                       int    nof_symbols,
                                       float  scaling)
```
the [declaration](https://github.com/srsran/srsRAN/blob/89f16eed2b7f2e6c45e675298f5829c230698e1d/lib/include/srsran/phy/mimo/precoding.h#L92
) is:

```c
SRSRAN_API int srsran_predecoding_diversity_multi(cf_t*  y[SRSRAN_MAX_PORTS],
                                                  cf_t*  h[SRSRAN_MAX_PORTS][SRSRAN_MAX_PORTS],
                                                  cf_t*  x[SRSRAN_MAX_LAYERS],
                                                  float* csi[SRSRAN_MAX_LAYERS],
                                                  int    nof_rxant,
                                                  int    nof_ports,
                                                  int    nof_symbols,
                                                  float  scaling);
```

## Setup Details ##

OS: archlinux 5.12.3-arch1-1 x86_64
cmake version: 3.20.2

## Expected Behavior ##

run `make` success

## Actual Behaviour ##

run `make` failed

```
Consolidate compiler generated dependencies of target srsran_mimo
[ 10%] Building C object lib/src/phy/mimo/CMakeFiles/srsran_mimo.dir/precoding.c.o
/mnt/share/MS/srsRAN/lib/src/phy/mimo/precoding.c:783:47: error: argument 4 of type ‘float *[2]’ with mismatched bound [-Werror=array-parameter=]
  783 |                                        float* csi[SRSRAN_MAX_CODEWORDS],
      |                                        ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /mnt/share/MS/srsRAN/lib/src/phy/mimo/precoding.c:28:
/mnt/share/MS/srsRAN/lib/include/srsran/phy/mimo/precoding.h:92:58: note: previously declared as ‘float *[4]’
   92 |                                                   float* csi[SRSRAN_MAX_LAYERS],
      |                                                   ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~
/mnt/share/MS/srsRAN/lib/src/phy/mimo/precoding.c: In function ‘srsran_predecoding_type’:
/mnt/share/MS/srsRAN/lib/src/phy/mimo/precoding.c:1914:16: error: ‘srsran_predecoding_diversity_multi’ accessing 32 bytes in a region of size 16 [-Werror=stringop-overflow=]
 1914 |         return srsran_predecoding_diversity_multi(y, h, x, csi, nof_rxant, nof_ports, nof_symbols, scaling);
      |                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/mnt/share/MS/srsRAN/lib/src/phy/mimo/precoding.c:1914:16: note: referencing argument 4 of type ‘float **’
/mnt/share/MS/srsRAN/lib/src/phy/mimo/precoding.c:780:5: note: in a call to function ‘srsran_predecoding_diversity_multi’
  780 | int srsran_predecoding_diversity_multi(cf_t*  y[SRSRAN_MAX_PORTS],
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
make[2]: *** [lib/src/phy/mimo/CMakeFiles/srsran_mimo.dir/build.make:90: lib/src/phy/mimo/CMakeFiles/srsran_mimo.dir/precoding.c.o]
Error 1
make[1]: *** [CMakeFiles/Makefile2:4342: lib/src/phy/mimo/CMakeFiles/srsran_mimo.dir/all] Error 2
make: *** [Makefile:166: all] Error 2
```

## Steps to reproduce the problem ##

Follow the [official instruction](https://docs.srsran.com/en/latest/general/source/1_installation.html#installation-from-source).

cld4h, 2021-05-27:
> ## Issue Description
> Incoordination in function declaration and implementation.
>
> In function `srsran_predecoding_diversity_multi`,
> the [...]

## Other similar issues:

### `srsran_cqi_value_pack`
https://github.com/srsran/srsRAN/blob/89f16eed2b7f2e6c45e675298f5829c230698e1d/lib/src/phy/phch/cqi.c#L121
https://github.com/srsran/srsRAN/blob/89f16eed2b7f2e6c45e675298f5829c230698e1d/lib/include/srsran/phy/phch/cqi.h#L146

alvasMan, 2021-05-27:
Hi @cld4h, thanks for reporting this.

I had actually noticed this recently, and there is now a fix for it in the https://github.com/srsran/srsRAN/tree/fix_issue675 branch.

Could have a look and say if this fixes your issue?

cld4h, 2021-05-27:
Hi @alvasMan , thanks for your quick respond!

I checked out the https://github.com/srsran/srsRAN/tree/fix_issue675 branch and did a clean build.

The problem I mentioned above seems fine now, however I still can't compile the project successfully.

Error messages I got:
```
In file included from /usr/include/c++/11.1.0/memory:63,
                 from /mnt/share/MS/srsRAN/lib/include/srsran/srslog/detail/support/any.h:25,
                 from /mnt/share/MS/srsRAN/lib/include/srsran/srslog/srslog.h:25,
                 from /mnt/share/MS/srsRAN/lib/include/srsran/common/srsran_assert.h:25,
                 from /mnt/share/MS/srsRAN/lib/include/srsran/asn1/asn1_utils.h:25,
                 from /mnt/share/MS/srsRAN/lib/include/srsran/asn1/rrc/common.h:31,
                 from /mnt/share/MS/srsRAN/lib/include/srsran/asn1/rrc/rr_common.h:31,
                 from /mnt/share/MS/srsRAN/lib/include/srsran/asn1/rrc/common_ext.h:31,
                 from /mnt/share/MS/srsRAN/lib/include/srsran/asn1/rrc/meascfg.h:31,
                 from /mnt/share/MS/srsRAN/lib/include/srsran/asn1/rrc/bcch_msg.h:31,
                 from /mnt/share/MS/srsRAN/lib/include/srsran/asn1/rrc.h:31,
                 from /mnt/share/MS/srsRAN/lib/test/asn1/../../../srsue/hdr/stack/rrc/rrc_cell.h:25,
                 from /mnt/share/MS/srsRAN/lib/test/asn1/../../../srsue/hdr/stack/rrc/rrc.h:25,
                 from /mnt/share/MS/srsRAN/lib/test/asn1/srsran_asn1_rrc_ul_dcch_test.cc:22:
In static member function ‘static _Tp* std::__copy_move<_IsMove, true, std::random_access_iterator_tag>::__copy_m(const _Tp*, const
_Tp*, _Tp*) [with _Tp = asn1::rrc::inter_freq_band_info_s; bool _IsMove = false]’,
    inlined from ‘_OI std::__copy_move_a2(_II, _II, _OI) [with bool _IsMove = false; _II = asn1::rrc::inter_freq_band_info_s*; _OI = asn1::rrc::inter_freq_band_info_s*]’ at /usr/include/c++/11.1.0/bits/stl_algobase.h:495:30,
    inlined from ‘_OI std::__copy_move_a1(_II, _II, _OI) [with bool _IsMove = false; _II = asn1::rrc::inter_freq_band_info_s*; _OI = asn1::rrc::inter_freq_band_info_s*]’ at /usr/include/c++/11.1.0/bits/stl_algobase.h:522:42,
    inlined from ‘_OI std::__copy_move_a(_II, _II, _OI) [with bool _IsMove = false; _II = asn1::rrc::inter_freq_band_info_s*; _OI =
asn1::rrc::inter_freq_band_info_s*]’ at /usr/include/c++/11.1.0/bits/stl_algobase.h:529:31,
    inlined from ‘_OI std::copy(_II, _II, _OI) [with _II = asn1::rrc::inter_freq_band_info_s*; _OI = asn1::rrc::inter_freq_band_info_s*]’ at /usr/include/c++/11.1.0/bits/stl_algobase.h:620:7,
    inlined from ‘void asn1::dyn_array<T>::resize(uint32_t, uint32_t) [with T = asn1::rrc::inter_freq_band_info_s]’ at /mnt/share/MS/srsRAN/lib/include/srsran/asn1/asn1_utils.h:243:18,
    inlined from ‘int pack_fail_test()’ at /mnt/share/MS/srsRAN/lib/test/asn1/srsran_asn1_rrc_ul_dcch_test.cc:161:67:
/usr/include/c++/11.1.0/bits/stl_algobase.h:431:30: error: ‘void* __builtin_memmove(void*, const void*, long unsigned int)’ writing
between 2 and 4294967295 bytes into a region of size 1 overflows the destination [-Werror=stringop-overflow=]
  431 |             __builtin_memmove(__result, __first, sizeof(_Tp) * _Num);
      |             ~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /mnt/share/MS/srsRAN/lib/include/srsran/asn1/rrc/common.h:31,
                 from /mnt/share/MS/srsRAN/lib/include/srsran/asn1/rrc/rr_common.h:31,
                 from /mnt/share/MS/srsRAN/lib/include/srsran/asn1/rrc/common_ext.h:31,
                 from /mnt/share/MS/srsRAN/lib/include/srsran/asn1/rrc/meascfg.h:31,
                 from /mnt/share/MS/srsRAN/lib/include/srsran/asn1/rrc/bcch_msg.h:31,
                 from /mnt/share/MS/srsRAN/lib/include/srsran/asn1/rrc.h:31,
                 from /mnt/share/MS/srsRAN/lib/test/asn1/../../../srsue/hdr/stack/rrc/rrc_cell.h:25,
                 from /mnt/share/MS/srsRAN/lib/test/asn1/../../../srsue/hdr/stack/rrc/rrc.h:25,
                 from /mnt/share/MS/srsRAN/lib/test/asn1/srsran_asn1_rrc_ul_dcch_test.cc:22:
/mnt/share/MS/srsRAN/lib/include/srsran/asn1/asn1_utils.h: In function ‘int pack_fail_test()’:
/mnt/share/MS/srsRAN/lib/include/srsran/asn1/asn1_utils.h:241:15: note: destination object of size 1 allocated by ‘operator new []’
  241 |       data_ = new T[cap_];
      |               ^~~~~~~~~~~
In file included from /usr/include/c++/11.1.0/memory:63,
                 from /mnt/share/MS/srsRAN/lib/include/srsran/srslog/detail/support/any.h:25,
                 from /mnt/share/MS/srsRAN/lib/include/srsran/srslog/srslog.h:25,
                 from /mnt/share/MS/srsRAN/lib/include/srsran/common/srsran_assert.h:25,
                 from /mnt/share/MS/srsRAN/lib/include/srsran/asn1/asn1_utils.h:25,
                 from /mnt/share/MS/srsRAN/lib/include/srsran/asn1/rrc/common.h:31,
                 from /mnt/share/MS/srsRAN/lib/include/srsran/asn1/rrc/rr_common.h:31,
                 from /mnt/share/MS/srsRAN/lib/include/srsran/asn1/rrc/common_ext.h:31,
                 from /mnt/share/MS/srsRAN/lib/include/srsran/asn1/rrc/meascfg.h:31,
                 from /mnt/share/MS/srsRAN/lib/include/srsran/asn1/rrc/bcch_msg.h:31,
                 from /mnt/share/MS/srsRAN/lib/include/srsran/asn1/rrc.h:31,
                 from /mnt/share/MS/srsRAN/lib/test/asn1/../../../srsue/hdr/stack/rrc/rrc_cell.h:25,
                 from /mnt/share/MS/srsRAN/lib/test/asn1/../../../srsue/hdr/stack/rrc/rrc.h:25,
[... cut ...]
/mnt/share/MS/srsRAN/lib/include/srsran/asn1/asn1_utils.h:241:15: note: destination object of size 1 allocated by ‘operator new []’
  241 |       data_ = new T[cap_];
      |               ^~~~~~~~~~~
cc1plus: all warnings being treated as errors
make[2]: *** [lib/test/asn1/CMakeFiles/srsran_asn1_rrc_ul_dcch_test.dir/build.make:76: lib/test/asn1/CMakeFiles/srsran_asn1_rrc_ul_dcch_test.dir/srsran_asn1_rrc_ul_dcch_test.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:7355: lib/test/asn1/CMakeFiles/srsran_asn1_rrc_ul_dcch_test.dir/all] Error 2
make: *** [Makefile:166: all] Error 2
```

frankist, 2021-05-27:
Interesting. Can you share your compiler version?

cld4h, 2021-05-27:
> Interesting. Can you share your compiler version?

Of course!

I turned on the verbose output of `cmake`, the compiile command is:

```sh
cd /mnt/share/MS/srsRAN/build/lib/test/asn1 && /usr/bin/c++ -DASSERTS_ENABLED -DENABLE_TIMEPROF -DHAVE_BACKWARD -DHAVE_MBEDTLS -DHAVE_PCSC -I/usr/include/PCSC -I/mnt/share/MS/srsRAN/build/lib/include -I/mnt/share/MS/srsRAN/lib/include -I/mnt/share/MS/srsRAN -I/mnt/share/MS/srsRAN/cmake/modules -Wall -Wno-comment -Wno-reorder -Wno-unused-variable -Wtype-limits -std=c++11 -fno-strict-aliasing -Wno-unused-but-set-variable -march=native -mfpmath=sse -mavx2 -DLV_HAVE_AVX2 -DLV_HAVE_AVX -DLV_HAVE_SSE -fvisibility=hidden -O3 -fno-trapping-math -fno-math-errno -DBUILD_TYPE_RELEASE -Werror -O3 -DNDEBUG -fPIE -MD -MT lib/test/asn1/CMakeFiles/srsran_asn1_rrc_ul_dcch_test.dir/srsran_asn1_rrc_ul_dcch_test.cc.o -MF CMakeFiles/srsran_asn1_rrc_ul_dcch_test.dir/srsran_asn1_rrc_ul_dcch_test.cc.o.d -o CMakeFiles/srsran_asn1_rrc_ul_dcch_test.dir/srsran_asn1_rrc_ul_dcch_test.cc.o -c /mnt/share/MS/srsRAN/lib/test/asn1/srsran_asn1_rrc_ul_dcch_test.cc
```

So the compiler is `/usr/bin/c++`

Then I run `/usr/bin/c++ -v` and got these messages:

```
Using built-in specs.
COLLECT_GCC=/usr/bin/c++
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-pc-linux-gnu/11.1.0/lto-wrapper
Target: x86_64-pc-linux-gnu
Configured with: /build/gcc/src/gcc/configure --prefix=/usr --libdir=/usr/lib --libexecdir=/usr/lib --mandir=/usr/share/man --infodir=/usr/share/info --with-bugurl=https://bugs.archlinux.org/ --enable-languages=c,c++,ada,fortran,go,lto,objc,obj-c++,d --with-isl --with-linker-hash-style=gnu --with-system-zlib --enable-__cxa_atexit --enable-cet=auto --enable-checking=release --enable-clocale=gnu --enable-default-pie --enable-default-ssp --enable-gnu-indirect-function --enable-gnu-unique-object --enable-install-libiberty --enable-linker-build-id --enable-lto --enable-multilib --enable-plugin --enable-shared --enable-threads=posix --disable-libssp --disable-libstdcxx-pch --disable-libunwind-exceptions --disable-werror gdc_include_dir=/usr/include/dlang/gdc
Thread model: posix
Supported LTO compression algorithms: zlib zstd
gcc version 11.1.0 (GCC)
```

cld4h, 2021-05-27:
And by the way, you forgot updating your mailing list link in

https://github.com/srsran/srsRAN/blob/master/.github/ISSUE_TEMPLATE.md#L3

cld4h, 2021-05-28:
Update: 
I successfully compiled the whole project on a Ubuntu 20.04 machine (with gcc version 9.3.0 (Ubuntu 9.3.0-17ubuntu1~20.04) ) today.

andrepuschmann, 2021-05-28:
@cld4h What OS was the original issue on? Or was it a manual gcc11 installation?

alvasMan, 2021-05-28:
@andrepuschmann this is the information that was on the original issue:
> OS: archlinux 5.12.3-arch1-1 x86_64

@cld4h I'm on arch as well, but I'm afraid I can't replicate this.
Can you make sure your system is up to date and do a clean build? 

If the problem persists can you share the output of the CMake?

cld4h, 2021-05-28:
@alvasMan I had my archlinux up to date and did a clean build, but the problem still persists.

The output of `cmake ../`:
```

CMake Deprecation Warning at CMakeLists.txt:33 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- The C compiler identification is GNU 11.1.0
-- The CXX compiler identification is GNU 11.1.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- CMAKE_SYSTEM: Linux-5.12.6-arch1-1
-- CMAKE_SYSTEM_PROCESSOR: x86_64
-- CMAKE_CXX_COMPILER: /usr/bin/c++
-- Build type not specified: defaulting to Release.
-- Performing Test HAVE_CXX_ATOMICS_WITHOUT_LIB
-- Performing Test HAVE_CXX_ATOMICS_WITHOUT_LIB - Success
-- Performing Test HAVE_CXX_ATOMICS64_WITHOUT_LIB
-- Performing Test HAVE_CXX_ATOMICS64_WITHOUT_LIB - Success
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE
-- Found PkgConfig: /usr/bin/pkg-config (found version "1.7.3")
-- Checking for module 'fftw3f >= 3.0'
--   Found fftw3f , version 3.3.9
-- FFTW3F LIBRARIES: /usr/lib/libfftw3f.so
-- FFTW3F STATIC LIBRARIES: FFTW3F_STATIC_LIBRARY-NOTFOUND
-- FFTW3F INCLUDE DIRS: /usr/include
-- Found FFTW3F: /usr/lib/libfftw3f.so
-- FFT_LIBRARIES: /usr/lib/libfftw3f.so
-- Checking for module 'polarssl'
--   Package 'polarssl', required by 'virtual:world', not found
-- POLARSSL LIBRARIES: POLARSSL_LIBRARIES-NOTFOUND
-- POLARSSL STATIC LIBRARIES: POLARSSL_STATIC_LIBRARIES-NOTFOUND
-- POLARSSL INCLUDE DIRS: POLARSSL_INCLUDE_DIRS-NOTFOUND
-- Could NOT find Polarssl (missing: POLARSSL_LIBRARIES POLARSSL_INCLUDE_DIRS)
-- Checking for module 'mbedtls'
--   Package 'mbedtls', required by 'virtual:world', not found
-- MBEDTLS LIBRARIES: /lib/libmbedcrypto.so
-- MBEDTLS STATIC LIBRARIES: /lib/libmbedcrypto.a
-- MBEDTLS INCLUDE DIRS: /usr/include
-- Found MbedTLS: /lib/libmbedcrypto.so
-- Checking for module 'libpcsclite'
--   Found libpcsclite, version 1.9.1
-- Found PCSCLite: /usr/lib/libpcsclite.so
-- PCSC LIBRARIES: /usr/lib/libpcsclite.so
-- PCSC INCLUDE DIRS: /usr/include/PCSC
-- Building with PCSC support.
-- UHD LIBRARIES /lib/libuhd.so
-- UHD INCLUDE DIRS /usr/include
-- Found UHD: /lib/libuhd.so
[... 66 lines cut ...]
-- No post-build command defined
-- Building with srsENB
-- Found LibConfig++: /usr/lib/libconfig++.so
-- static LibConfig++ path: LIBCONFIGPP_STATIC_LIBRARY-NOTFOUND
-- Found LibConfig: /usr/lib/libconfig.so
-- static LibConfig path: LIBCONFIG_STATIC_LIBRARY-NOTFOUND
-- Checking for module 'sctp'
--   Package 'sctp', required by 'virtual:world', not found
-- SCTP LIBRARIES: /usr/lib/libsctp.so
-- SCTP INCLUDE DIRS: /usr/include
-- No post-build-ENB command defined
-- Building with srsEPC
-- Found LibConfig++: /usr/lib/libconfig++.so
-- static LibConfig++ path: LIBCONFIGPP_STATIC_LIBRARY-NOTFOUND
-- Found LibConfig: /usr/lib/libconfig.so
-- static LibConfig path: LIBCONFIG_STATIC_LIBRARY-NOTFOUND
-- Checking for module 'sctp'
--   Package 'sctp', required by 'virtual:world', not found
-- SCTP LIBRARIES: /usr/lib/libsctp.so
-- SCTP INCLUDE DIRS: /usr/include
-- No post-build-EPC command defined
-- Configuring done
-- Generating done
-- Build files have been written to: /mnt/share/MS/srsRAN/build
```

godfuzz3r, 2021-06-27:
The same problem with arch 5.12.13-arch1-2 x86_64

drago-sp, 2021-07-08:
I've had good success compiling the `fix-issue675` branch on latest Arch (5.12.14-arch1-1) with gcc 11.1.0 when SoapySDR support is disabled. Yet, when `ENABLE_SOAPYSDR=ON` I get a similar compile error :

```
/home/drago/src/srsLTE/lib/src/phy/rf/rf_soapy_imp.c:793:44: error: argument 2 of type ‘void *[4]’ with mismatched bound [-Werror=array-parameter=]
  793 |                                   void*    data[SRSRAN_MAX_PORTS],
      |                                   ~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~
In file included from /home/drago/src/srsLTE/lib/src/phy/rf/rf_soapy_imp.c:28:
/home/drago/src/srsLTE/lib/src/phy/rf/rf_soapy_imp.h:79:47: note: previously declared as ‘void **’
   79 | rf_soapy_recv_with_time_multi(void* h, void** data, uint32_t nsamples, bool blocking, time_t* secs, double* frac_secs);
      |                                        ~~~~~~~^~~~
```

sparklespdx, 2021-10-02:
I'm having the same issue with GCC11. Running the `fix_issue675` also fixes my build.

zikkuratti, 2022-08-30:
yeah manually change LAYERS to CODEWORD in that function