srsran/srsRAN_4G #679: VoLTE with srsenb and Amarisoft Callbox IMS doesn't work

NorbieE, 2021-06-10:
Hi,

we are running several 4G and 5G  Networks using srsLTE/srsRAN and Amarisoft as testbeds for mobile communications at our university.

Since srsLTE /srsRAN doesn't include an IMS for VoLTE we had the idea, that we could only use  srsenb and connect it to the MME and IMS from the Amarisoft Callbox.

To check if the Amarisoft MME is able to connect multiple eNodeB, we first connected another Amarisoft eNodeB to it.
UE registration and  VoLTE worked fine with this configuration.

After this we tried to connect srsenb to the Amarisoft MME.

Right now we are able to connect UEs to the core Network with this configuration and to browse the Internet with this UEs.

If we try to setup a voice call, the UE tries to connect to the IMS but gets kicked out of the network after some time.

Looking into the PCAP files generated by srsenb (see attached files)  it seems that there is an issue while connecting to the IMS.

[image]

The UE tries to connect to the ims and then repeats the connection request, resulting in an reject, because multiple connections to on APN aren't allowed.

My feeling is, that there is a missing part ( either an accept or a reject message) in the communication between UE and IMS.)
and if so, could this be a routing problem? ( I tried to setup srsenb with open5gs an Kamailio-IMS but failed to do so. But there was on step, where a static route had to be configured for the eNodeB)

Maybe someone more experienced then me and my team is able to find the problem with our configuration and could give us a hint, as to where to look next. 

[srslogs.zip](https://github.com/srsran/srsRAN/files/6631371/srslogs.zip)

alvasMan, 2021-06-10:
I migh be wrong, but looking at the cause in the E-RAB setup response, it seems to me that the QCI does not exist in eNB.

See the following code where that cause seems to be set:

```
  auto qci_it = cfg->qci_cfg.find(qos.qci);
   if (qci_it == cfg->qci_cfg.end() or not qci_it->second.configured) {
     logger->error("QCI=%d not configured", qos.qci);
     cause.set_radio_network().value = asn1::s1ap::cause_radio_network_opts::not_supported_qci_value;
     return SRSRAN_ERROR;
   }

```

Can you post your drb.conf and the PCAP?

NorbieE, 2021-06-10:
Hi,

thanks for your quick reaction.

As i already wrote, uploading the zip file fails somehow.
When i drag the file into the command, i see a message that the file is beeing uploaded, but afterwards i see the message "is not included in the list" at the end of the comment, and the attachment is gone. Im trying to figure out why this happens. (Already tried to use another Browser)

However, as the drb.conf isn't to big i can post it here:

// All times are in ms. Use -1 for infinity, where available

qci_config = (

{
  qci=7;
  pdcp_config = {
    discard_timer = -1;                
    pdcp_sn_size = 12;                  
  }
  rlc_config = {
    ul_um = {
      sn_field_length = 10; 
    };
    dl_um = {
      sn_field_length = 10; 
      t_reordering    = 45;             
    };
  };
  logical_channel_config = {
    priority = 13; 
    prioritized_bit_rate   = -1; 
    bucket_size_duration  = 100; 
    log_chan_group = 2; 
  };
},
{
  qci=9;
  pdcp_config = {
    discard_timer = 150;
    status_report_required = true;
  }
  rlc_config = {
    ul_am = {
      t_poll_retx = 120;
      poll_pdu = 64;
      poll_byte = 750;
      max_retx_thresh = 16;
    };
    dl_am = {
      t_reordering = 50;
      t_status_prohibit = 50;
    };
  };
  logical_channel_config = {
    priority = 11; 
    prioritized_bit_rate   = -1; 
    bucket_size_duration  = 100; 
    log_chan_group = 3; 
  };
}

);

As you can see, it is the untouched file that comes with the srsRAN repository.

We didn't alter it, because we thought that this has to do with "Quality of Service", and that QoS simply wouldn't be used, if there was no entry. 

As said i will post the PCAP and Log files as soon as possible.

NorbieE, 2021-06-11:
@alvasMan @andrepuschmann Thanks a lot for your comments!! 

Sometimes you need someone to put you on the right track again.

After setting up the the drb.config VoLTE now works with srsenb.

There seem to be some issues with some commercial UEs but i think these can be overcome too.

closed 2021-06-12 by NorbieE

NavnathNavIC, 2024-12-20:
Hey NorbiE,
I'm also trying the same thing what you have done, I'm using Amarisoft EPC core and SRS LTE eNB, but IMS is not working with SRS eNB

Will you please share me details how you have resolved this issue.

NorbieE, 2024-12-20:
Hello NavnathNavIC ,

as far as i remember i had to create additional sections for qci=1 and qci=2 in the drb.cfg file as @andrepuschmann hinted.

As i'm out of office for hollidays i can't look into the systems right now.
If needed i will provide more information when i am back in office mid january.