srsran/srsRAN_4G #680: Cannot add PPA: 'ppa:~srsran/ubuntu/releases'.

vikramkapoor, 2021-06-13:
## Issue Description ##
$ sudo add-apt-repository ppa:srsran/releases
Cannot add PPA: 'ppa:~srsran/ubuntu/releases'.
ERROR: '~srsran' user or team does not exist.
## Setup Details ##
[Specify details of the test setup. This would help us reproduce the problem reliably]
e.g. Network configuration, Operation System, Hardware, RF front-end, library and driver versions

## Expected Behavior ##
[What you expect to happen]

## Actual Behaviour ##
[What happens instead e.g. error message]

## Steps to reproduce the problem ##
[Tell us how to reproduce this issue e.g. RF setup, application config files]

## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]

vikramkapoor, 2021-06-13:
What works is sudo add-apt-repository ppa:softwareradiosystems/srsran

suttonpd, 2021-06-14:
Where did you find instructions to use ppa:srsran/releases?

vikramkapoor, 2021-06-14:
They were here
Installation Guide — srsRAN 21.04 documentation

| 
| 
|  | 
Installation Guide — srsRAN 21.04 documentation

 |

 |

 |

suttonpd, 2021-06-14:
Thanks - should be corrected now.

closed 2021-06-14 by suttonpd