srsran/srsRAN_4G #683: [Question] LimeSDR-mini connect with srsUE to commercial network

Radiomix2000, 2021-06-15:
Anybody had success to connect to commercial network with LimeSDR-mini and srsUE?
Mine cannot connect. Always has the same result:
Could not find Home PLMN Id=00101, trying to connect to PLMN Id=XXXXX
Random Access Transmission: seq=23, tti=9151, ra-rnti=0x2
LRandom Access Transmission: seq=11, tti=9171, ra-rnti=0x2
LRandom Access Transmission: seq=5, tti=9191, ra-rnti=0x2
LRandom Access Transmission: seq=12, tti=9211, ra-rnti=0x2
LRandom Access Transmission: seq=14, tti=9231, ra-rnti=0x2
LRandom Access Transmission: seq=11, tti=9251, ra-rnti=0x2
LRandom Access Transmission: seq=5, tti=9271, ra-rnti=0x2
LRandom Access Transmission: seq=10, tti=9291, ra-rnti=0x2
RF status: O=0, U=0, L=8
LRandom Access Transmission: seq=2, tti=9311, ra-rnti=0x2
LRandom Access Transmission: seq=24, tti=9331, ra-rnti=0x2
RF status: O=0, U=0, L=2
Attach failed (attempt 1/5)

What's the problem? The srsUE config is default config, except earfcn, USIM conf and device conf section.

andrepuschmann, 2021-06-23:
Could be many things. Likely Tx power/config isn't sufficiant. Provide logs if you can.

Radiomix2000, 2021-06-23:
Good, so first I will try with more RF power, if this doesn't help, I will send the logs.

closed 2023-04-14 by ismagom