srsran/srsRAN_4G #684: UHD-4.0 error on USRP X310

shuimoshusheng, 2021-06-18:
root@ubuntu:~# sudo srsenb
---  Software Radio Systems LTE eNodeB  ---

Reading configuration file /root/.config/srsran/enb.conf...
WARNING: Could not verify cpu0 scaling governor
WARNING: Could not verify cpu1 scaling governor

Built in Release mode using commit 89f16eed2 on branch master.

Opening 1 channels in RF device=default with args=default
[INFO] [UHD] linux; GNU C++ version 7.5.0; Boost_106501; UHD_4.0.0.0-154-gb061af4f
[INFO] [LOGGING] Fastpath logging disabled at runtime.
Opening USRP channels=1, args: type=x300,master_clock_rate=184.32e6
[INFO] [UHD RF] RF UHD Generic instance constructed
[INFO] [X300] X300 initialization sequence...
[INFO] [X300] Maximum frame size: 1472 bytes.
[INFO] [X300] Radio 1x clock: 184.32 MHz
Setting frequency: DL=2680.0 Mhz, UL=2560.0 MHz for cc_idx=0 nof_prb=50

==== eNodeB started ===
Type <t> to view trace
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 0.790861 >= 0. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 0.790861 >= 0. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 0.797064 >= 0. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 0.798222 >= 0. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 0.798222 >= 0. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 0.798222 >= 0. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 0.805361 >= 0. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 0.805361 >= 0. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 0.811268 >= 0. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 0.811268 >= 0. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 0.816954 >= 0. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 0.823966 >= 0. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 0.825219 >= 0. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 0.825219 >= 0. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 0.825219 >= 0. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 0.83397 >= 0. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 0.83397 >= 0. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 0.840256 >= 0. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 0.845637 >= 0. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 0.846858 >= 0. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 0.847953 >= 0. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 0.847953 >= 0. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 0.854535 >= 0. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 0.855661 >= 0. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 0.855661 >= 0. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 0.862326 >= 0. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 0.863611 >= 0. Starting new burst...
/root/srsRAN/lib/src/phy/rf/rf_uhd_imp.cc.309: UHD unhandled event code 4
[ERROR] [X300] 192.168.10.2: x300 fw communication failure #1
EnvironmentError: IOError: x300 fw poke32 - reply timed out
/root/srsRAN/lib/src/phy/rf/rf_uhd_imp.cc.1353: Error timed out while receiving samples from UHD.
/root/srsRAN/lib/src/phy/rf/rf_uhd_imp.cc.1353: Error timed out while receiving samples from UHD.
/root/srsRAN/lib/src/phy/rf/rf_uhd_imp.cc.1353: Error timed out while receiving samples from UHD.
/root/srsRAN/lib/src/phy/rf/rf_uhd_imp.cc.1353: Error timed out while receiving samples from UHD.
/root/srsRAN/lib/src/phy/rf/rf_uhd_imp.cc.1353: Error timed out while receiving samples from UHD.
/root/srsRAN/lib/src/phy/rf/rf_uhd_imp.cc.1353: Error timed out while receiving samples from UHD.
/root/srsRAN/lib/src/phy/rf/rf_uhd_imp.cc.1353: Error timed out while receiving samples from UHD.
/root/srsRAN/lib/src/phy/rf/rf_uhd_imp.cc.1353: Error timed out while receiving samples from UHD.
/root/srsRAN/lib/src/phy/rf/rf_uhd_imp.cc.1353: Error timed out while receiving samples from UHD.
/root/srsRAN/lib/src/phy/rf/rf_uhd_imp.cc.1353: Error timed out while receiving samples from UHD.
/root/srsRAN/lib/src/phy/rf/rf_uhd_imp.cc.1353: Error timed out while receiving samples from UHD.
/root/srsRAN/lib/src/phy/rf/rf_uhd_imp.cc.1353: Error timed out while receiving samples from UHD.
^CStopping ..
/root/srsRAN/lib/src/phy/rf/rf_uhd_imp.cc.1353: Error timed out while receiving samples from UHD.
^C/root/srsRAN/lib/src/phy/rf/rf_uhd_imp.cc.1353: Error timed out while receiving samples from UHD.
/root/srsRAN/lib/src/phy/rf/rf_uhd_imp.cc.1353: Error timed out while receiving samples from UHD.
---  exiting  ---
root@ubuntu:~#

andrepuschmann, 2021-06-23:
Try to increase the MTU of your network interface.

andrepuschmann, 2021-11-26:
No further feedback from reporter. Closing issue.

closed 2021-11-26 by andrepuschmann