srsran/srsRAN_4G #686: srsRAN crashed...  "Error binding SCTP socket"

lamom92, 2021-06-21:
Hello,

I compiled srsran from git (commit 89f16eed2 on branch master)
I try to launch srsepc and the soft crash during the initialization.

_$ sudo srsepc

Built in Release mode using commit 89f16eed2 on branch master.

---  Software Radio Systems EPC  ---

Reading configuration file /root/.config/srsran/epc.conf...
HSS Initialized.
Error binding SCTP socket
MME S11 Initialized
MME GTP-C Initialized
MME Initialized. MCC: 0xf001, MNC: 0xff01
SPGW GTP-U Initialized.
SPGW S11 Initialized.
SP-GW Initialized.
*** buffer overflow detected ***: terminated
srsRAN crashed... backtrace saved in './srsRAN.backtrace.crash'...
---  exiting  ---_

In the debug file srsRAN.backtrace.crash, there is

--- command='srsepc' version=21.04.0 signal=6 date='21/06/2021 21:14:17' ---
	srsepc(+0x18ab5f) [0x5604be57eb5f]
	/lib/x86_64-linux-gnu/libc.so.6(+0x46210) [0x7f7e7e62f210]
	/lib/x86_64-linux-gnu/libc.so.6(gsignal+0xcb) [0x7f7e7e62f18b]
	/lib/x86_64-linux-gnu/libc.so.6(abort+0x12b) [0x7f7e7e60e859]
	/lib/x86_64-linux-gnu/libc.so.6(+0x903ee) [0x7f7e7e6793ee]
	/lib/x86_64-linux-gnu/libc.so.6(__fortify_fail+0x2a) [0x7f7e7e71bb4a]
	/lib/x86_64-linux-gnu/libc.so.6(+0x1313e6) [0x7f7e7e71a3e6]
	/lib/x86_64-linux-gnu/libc.so.6(+0x132a8b) [0x7f7e7e71ba8b]
	srsepc(+0x3ba5d) [0x5604be42fa5d]
	srsepc(+0x2fe7d) [0x5604be423e7d]
	/lib/x86_64-linux-gnu/libpthread.so.0(+0x9609) [0x7f7e7ec28609]
	/lib/x86_64-linux-gnu/libc.so.6(clone+0x43) [0x7f7e7e70b293]

I don't understand the issue.

Thank you for your help.

andrepuschmann, 2021-06-23:
I agree we should cleanly exit in this case. But the underlaying cause of the issue is the SCTP port 36412 being used from someone else. Anything else running?

lamom92, 2021-06-23:
Thank you, with your help, I succeed to solve the problem, the issue is that open5gs-mmed was using the same port 36412.

to list the process that is using port 36412, I did:

$ sudo netstat -lpn | grep :36412
sctp                127.0.1.100:36412                               LISTEN      800/open5gs-mmed 

I uninstall open5gs-mme (I'm not using anymore open5gs as a core)
$ sudo apt-get remove open5gs-mme

closed 2021-06-23 by lamom92