srsran/srsRAN_4G #687: Stability problem: No new UEs connecting after some time

ECTPavloR, 2021-06-22:
## Issue Description ##
With the latest release 21.04, we're running srsenb+srsepc for ~25 UE, 50prb. After a certain period of time (~5-15min somewhere after rnti # > ff , but not sure this is in correlation with rnti # ), UEs starting actively dropping, not simultaneously at one moment but within another ~5-15min most of them disconnected, and in a long run (~2h) nobody connected AND newly turned-on UE is not able to connect too. Right after srsenb restart all the UEs immediately start connecting back. 
All the UEs on air are loaded by fping with small packets for the target IP range to avoid UEs going idle + a few kbps time-to-time for the other test payload.
In case only fping load - system much longer remains stable (up to 10h tested)
In case active traffic generated on top of fping, a lot of connect-disconnect messages(rnti getting bigger than ff in a couple of minutes) in console trace and system get stuck much faster. During this process, a lot (more than UEs on air) rnti with cqi 0.10 (assuming they are consuming resource blocks as well).

## Setup Details ##
srsran 21.04, 89f16eed2
RF front-end - BladeRF
50PRB
QCI 9 or 7, TM 2,3,4 (same behavior)
i5 6cores CPU
up to 25 Cat1 UEs 

## Expected Behavior ##
srsenb running stable for a long time under variable load. New UEs can connect. Radio resources cleaned from phantoms with cqi = 0.10

## Actual Behaviour ##
existing UEs getting disconnected(not due to idle most likely), no new devices able to connect. Restart for enb needed to reconnect them all. 

## Steps to reproduce the problem ##
Running >10 UEs for some time (as more UEs connected, the faster problem appears) with variable load (up to 100kbps)
## Additional Information ##
Ready to share a lot of screenshots with enb console trace and fping stat at different stages, didn't make pcaps so far.

andrepuschmann, 2021-06-23:
Thanks for flagging the issue. We need a bit more info on your setup and logs to understand whats happening.  Could you perhaps provide logs in info and PCAPs for both eNB and EPC?

ECTPavloR, 2021-06-23:
> Thanks for flagging the issue. We need a bit more info on your setup and logs to understand whats happening. Could you perhaps provide logs in info and PCAPs for both eNB and EPC?

Hello @andrepuschmann ! Here we go: https://drive.google.com/file/d/1auypHMZdl738_O6UqeAIwTCCc4d0EE0k/view?usp=sharing
at your disposal to grab more traces or check some configs
P.S. for some reason github does not allow me to attach the file right here.

ECTPavloR, 2021-06-23:
------DL-------------------------------UL--------------------------------------------
rnti cqi  ri mcs brate   ok  nok  (%)  pusch pucch phr mcs brate   ok  nok  (%)   bsr
  c2 0.10   0 3.2     0    0    0   0%    0.00.100   0.0   0     0    0    0   0%   0.0
  c3 0.10   0 0.0     0    0    0   0%    0.00.100   0.0   0     0    0    0   0%   0.0
  c5  15   0 0.0     0    0    0   0%    0.00.100   0.0   0     0    0    0   0%   0.0
 107 0.10   0 0.0     0    0    0   0%    0.00.100   0.0   0     0    0    0   0%   0.0
 149 0.10   00.10     0    0    0   0%    0.00.100   0.0   0     0    0    0   0%   0.0
 10a 0.10   0 0.0     0    0    0   0%    0.00.100   0.0   0     0    0    0   0%   0.0
 14b 0.10   0 0.0     0    0    0   0%    0.00.100   0.0   0     0    0    0   0%   0.0
 10c 1.0   0 0.0     0    0    3 100%    0.00.100   0.0   0     0    0    0   0%   0.0
  ce 0.10   0 0.0     0    0    0   0%    0.00.100   0.0   0     0    0    0   0%   0.0
  cf 4.0   0 0.0     0    0    3 100%    0.00.100   0.0   0     0    0    0   0%   0.0
 110 0.10   0 0.0     0    0    0   0%    0.00.100   0.0   0     0    0    0   0%   0.0
  d3 10.0   00.10     0    0    0   0%    0.00.100   0.0   0     0    0    0   0%   0.0
 114 4.3   0 0.0     0    0    9 100%    0.00.100   0.0   0     0    0    0   0%   0.0
  d5 0.10   0 0.0     0    0    0   0%    0.00.100   0.0   0     0    0    0   0%   0.0
 116 0.10   0 0.0     0    0    0   0%    0.00.100   0.0   0     0    0    0   0%   0.0
 117 0.10   0 0.0     0    0    0   0%    0.00.100   0.0   0     0    0    0   0%   0.0
  d8 0.10   00.10     0    0    0   0%    0.00.100   0.0   0     0    0    0   0%   0.0
 119 0.10   00.10     0    0    0   0%    0.00.100   0.0   0     0    0    0   0%   0.0
  dd 0.10   0 0.0     0    0    0   0%    0.00.100   0.0   0     0    0    0   0%   0.0
  5f  12 1.0 0.0   864    2    0   0%    0.00.100  31.0   0  4.3k    1    0   0%   0.0
  e0 4.0   00.10     0    0    3 100%    0.00.100   0.0   0     0    0    0   0%   0.0
 121 0.10   00.10     0    0    0   0%    0.00.100   0.0   0     0    0    0   0%   0.0
  e2 1.0   00.10     0    0    3 100%    0.00.100   0.0   0     0    0    0   0%   0.0
  e3 0.10   0 0.0     0    0    0   0%    0.00.100   0.0   0     0    0    0   0%   0.0
  e4 0.10   0 0.0     0    0    0   0%    0.00.100   0.0   0     0    0    0   0%   0.0
 125 0.10   0 0.0     0    0    0   0%    0.00.100   0.0   0     0    0    0   0%   0.0
  e6 6.5   0 0.0     0    0    6 100%    0.00.100   0.0   0     0    0    0   0%   0.0
  e7 2.0   0 0.0     0    0    3 100%    0.00.100   0.0   0     0    0    0   0%   0.0
  a8 0.10   0 0.0     0    0    0   0%    0.00.100   0.0   0     0    0    0   0%   0.0
  e9 8.0   0 0.0     0    0    3 100%    0.00.100   0.0   0     0    0    0   0%   0.0
  eb 0.10   0 0.0     0    0    0   0%    0.00.100   0.0   0     0    0    0   0%   0.0
  ec 0.10   0 0.0     0    0    0   0%    0.00.100   0.0   0     0    0    0   0%   0.0
 12d 0.10   0 0.0     0    0    0   0%    0.00.100   0.0   0     0    0    0   0%   0.0
  f0 0.10   0 0.0     0    0    0   0%    0.00.100   0.0   0     0    0    0   0%   0.0
  f1 0.10   0 0.0     0    0    0   0%    0.00.100   0.0   0     0    0    0   0%   0.0
  b2 0.10   00.10     0    0    0   0%    0.00.100   0.0   0     0    0    0   0%   0.0
  b3 0.10   0 0.0     0    0    0   0%    0.00.100   0.0   0     0    0    0   0%   0.0
  f4 0.10   00.10     0    0    0   0%    0.00.100   0.0   0     0    0    0   0%   0.0
  f5 0.10   0 0.0     0    0    0   0%    0.00.100   0.0   0     0    0    0   0%   0.0
  f6 0.10   0 0.0     0    0    0   0%    0.00.100   0.0   0     0    0    0   0%   0.0
  f8 0.10   0 0.0     0    0    0   0%    0.00.100   0.0   0     0    0    0   0%   0.0
  ba 6.0   0 0.0     0    0    3 100%    0.00.100   0.0   0     0    0    0   0%   0.0
  fb 0.10   0 0.0     0    0    0   0%    0.00.100   0.0   0     0    0    0   0%   0.0
  fd 0.10   0 0.0     0    0    0   0%    0.00.100   0.0   0     0    0    0   0%   0.0
  be 0.10   0 0.0     0    0    0   0%    0.00.100   0.0   0     0    0    0   0%   0.0
  bf 0.10   0 0.0     0    0    0   0%    0.00.100   0.0   0     0    0    0   0%   0.0
^CStopping ..
Saving MAC PCAP (DLT=149) to /tmp/enb.pcap
Saving S1AP PCAP file
---  exiting  ---
Latest radio trace view before stopping. In the beginning there were ~20 UEs with good cqi. BTW, what cqi 0.10 mean here? It's pretty clear when it's >1, but for 0.1 I didn't find in specs

andrepuschmann, 2021-06-25:
Your analysis is right. At some stage users aren't correctly removed from the eNB and new users can't attach anymore (limit is 64). The root cause isn't clear, we need to investigate.

ECTPavloR, 2021-06-25:
> Your analysis is right. At some stage, users aren't correctly removed from the eNB and new users can't attach anymore (limit is 64). The root cause isn't clear, we need to investigate.

Dear Andre, Thanks a lot for your attention to this issue.  How long it may take on your side, roughly? I clearly understand that you have no SLA in front of the community, but we'd like to define our strategy regarding possible workarounds with OAI eNb or digging even deeper into SRS radio code, which is a bit hard for us. Anyway, we're ready to provide any kind of support with testing/debugging. Thanks!

andrepuschmann, 2021-06-25:
We have to reproduce the issue with the current public release first and see how our internal testing with the current development version reacts to it. 

We are constantly improving the stack, especially the 4G eNB. Those improvements will be fed back to the community every 6 months. So the next update will be in October.

zsy9512, 2021-09-08:
Hello, we were running srsRAN with release 21.04, and we got a problem about multi users. We tried to connect one EPC and ENB with two UEs, but we failed. For example, we connected UE1 with EPC and ENB, it worked well. However, when we connected UE2 with EPC and ENB, UE1 was disconnected in few seconds, which meant UE1 was disconnected and UE2 was connected. We could not connect the two UEs with EPC and ENB at the same time.

We thought there were some configurations that need to be set for multiuser communication, and tried some ways, but they did not work.

Could you please help us with this problem, thank you so much!

andrepuschmann, 2021-09-08:
@zsy9512 Not sure if your issue is the same like the original one. I have my doubts. I guess you better open another ticket and provide eNB logs in info.

zsy9512, 2021-09-08:
ok, i will open a new issue and provide more detail, thanks a lot.

avalori1, 2022-03-21:
@ECTPavloR , I also faced issues when connecting multiple UEs equally stuimulated with iperf. Typically troubles appare sooner the higher the number of UEs is. Looking at the metrics I noticed that the beginning of the troubles is always happening simultaneously to a drop in the ul_bsr parameter. I understood that this parameter is used to allocate resources between UEs, and therefore makes sense that the issues are only visible with multiple UEs. I saw this drops in the ul_sr parameter also with a single UEs, but my guess is that, having a single UEs the relevance of the resources allocation is not that relevant.
@andrepuschmann , may this hypothesis make sense? Is the correlation between drop in bitrate and ul_bsr of any value in figuring out what is happening? I can provide data if useful to dig...
Regards

andrepuschmann, 2022-03-22:
Hey everyone, I am afraid we are mixing things up here. What seems to be related issues are in fact not related. @avalori1 I've commented on #837, which is an issue on the UE side rather than the eNB. This issue here is still not fully understood. But I would suggest to use the last release, i.e. 21.10, and try to reproduce there. I am closing this issue therefore. Please feel free to reopen if needed.

closed 2022-03-22 by andrepuschmann