srsran/srsRAN_4G #688: bladeRF with FPGA version 0.11.1 does appear to work with howto example

chuckixia, 2021-06-23:
## Issue Description ##
srs UE attach appears to fail when frequency changes
```
root@host:~# srsue
Reading configuration file /root/.config/srsran/ue.conf...
WARNING: Could not verify cpu0 scaling governor
WARNING: Could not verify cpu1 scaling governor
WARNING: Could not verify cpu2 scaling governor
WARNING: Could not verify cpu3 scaling governor

Built in Release mode using commit 89f16eed2 on branch master.

Opening 1 channels in RF device=default with args=default
Opening bladeRF...
Set RX sampling rate 1.92 Mhz, filter BW: 2.50 Mhz
Waiting PHY to initialize ... done!
Attaching UE...
Set RX sampling rate 1.92 Mhz, filter BW: 2.50 Mhz
set RX frequency to 2680000001
set TX frequency to 2560000001
.
Found Cell:  Mode=FDD, PCI=1, PRB=50, Ports=1, CFO=0.3 KHz
Set RX sampling rate 11.52 Mhz, filter BW: 10.00 Mhz
Found PLMN:  Id=00101, TAC=7
RF status: O=84, U=0, L=0
Set RX sampling rate 1.92 Mhz, filter BW: 2.50 Mhz
.
Found Cell:  Mode=FDD, PCI=1, PRB=50, Ports=1, CFO=0.6 KHz

```

## Setup Details ##
- Two bladeRF's x115, about a foot apart from each other
-  FPGA version 0.11.1
- firmware 2.40
- calibrated

## Expected Behavior ##
srsUE should attached

## Steps to reproduce the problem ##
follow the tutorial example here
https://docs.srsran.com/en/latest/usermanuals/source/1_setup.html#running-a-4g-end-to-end-system

## Additional Information ##
installing firmware v0.12.0 appears to produce this problem - https://github.com/Nuand/bladeRF/issues/819
And firmware 0.11.1 might be this problem https://github.com/Nuand/bladeRF/issues/778

so I'm trying to figure out if there is a way around this one way or the other. thanks.

andrepuschmann, 2021-06-23:
Hey, yeah, it's a known problem, see https://github.com/Nuand/bladeRF/issues/778 for example. I assume the eNB works well with your current setup. One thing you can try on the UE is to fix the sample rate at the radio and let srsUE do the downsampling if needed. Try:
```
[rf]
srate = 11.52e6
```
For example.

chuckixia, 2021-06-23:
it's hard to know since I'm just trying out from the getting started guide and only have two bladeRF's. But yes, it appears that the eNB is working fine. If you've got a way for me to independently verify that with a smart phone, let me know :).

chuckixia, 2021-06-23:
ok that got a lot farther.

UE
```
Reading configuration file /root/.config/srsran/ue.conf...
WARNING: Could not verify cpu0 scaling governor
WARNING: Could not verify cpu1 scaling governor
WARNING: Could not verify cpu2 scaling governor
WARNING: Could not verify cpu3 scaling governor

Built in Release mode using commit 89f16eed2 on branch master.

Opening 1 channels in RF device=default with args=default
Opening bladeRF...
Set RX sampling rate 1.92 Mhz, filter BW: 2.50 Mhz
Waiting PHY to initialize ... done!
Attaching UE...
Set RX sampling rate 11.52 Mhz, filter BW: 10.00 Mhz
set RX frequency to 2680000001
set TX frequency to 2560000001
.
Found Cell:  Mode=FDD, PCI=1, PRB=50, Ports=1, CFO=0.5 KHz
Found PLMN:  Id=00101, TAC=7
Random Access Transmission: seq=33, tti=6581, ra-rnti=0x2
Random Access Transmission: seq=2, tti=6601, ra-rnti=0x2
Random Access Transmission: seq=32, tti=6621, ra-rnti=0x2
Random Access Transmission: seq=17, tti=6641, ra-rnti=0x2
Random Access Transmission: seq=20, tti=6661, ra-rnti=0x2
RF status: O=338, U=0, L=1
/home/nobletrout/github/srsRAN/lib/src/phy/rf/rf_blade_imp.c.532: TX failed: Invalid operation or parameter
Random Access Transmission: seq=34, tti=6681, ra-rnti=0x2
Random Access Transmission: seq=10, tti=6701, ra-rnti=0x2
Random Access Transmission: seq=28, tti=6721, ra-rnti=0x2
/home/nobletrout/github/srsRAN/lib/src/phy/rf/rf_blade_imp.c.532: TX failed: Invalid operation or parameter
Random Access Transmission: seq=42, tti=6741, ra-rnti=0x2
Random Access Transmission: seq=42, tti=6761, ra-rnti=0x2
RF status: O=1055, U=0, L=1
RF status: O=750, U=0, L=0
RF status: O=448, U=0, L=0
RF status: O=611, U=0, L=0
RF status: O=181, U=0, L=0
RF status: O=451, U=0, L=0
RF status: O=479, U=0, L=0
RF status: O=663, U=0, L=0
RF status: O=629, U=0, L=0
RF status: O=290, U=0, L=0
```

enb
```
root@racoon:~# srsenb
---  Software Radio Systems LTE eNodeB  ---

Reading configuration file /root/.config/srsran/enb.conf...
WARNING: cpu0 scaling governor is not set to performance mode. Realtime processing could be compromised. Consider setting it to performance mode before running the application.

Built in Release mode using commit 89f16eed2 on branch master.

Opening 1 channels in RF device=default with args=default
Opening bladeRF...
Set RX sampling rate 1.92 Mhz, filter BW: 2.50 Mhz
Set RX sampling rate 11.52 Mhz, filter BW: 10.00 Mhz
Setting frequency: DL=2680.0 Mhz, UL=2560.0 MHz for cc_idx=0 nof_prb=50
set TX frequency to 2680000001
set RX frequency to 2560000001

==== eNodeB started ===
Type <t> to view trace
RACH:  tti=6631, cc=0, preamble=40, offset=29, temp_crnti=0x46
Disconnecting rnti=0x46.
```

chuckixia, 2021-06-23:
I've managed to get MAC and sometimes NAS messages sent/received, but it still appears to be having some kind of issue. I've done the following to get the bladeRF's past the firmware issue, but believe I now either have a timing or synch issue, which I'm struggling to find a solution for.

here is the diff from git
```
diff --git a/lib/src/phy/rf/rf_blade_imp.c b/lib/src/phy/rf/rf_blade_imp.c
index b5160ba1b..64c899d67 100644
--- a/lib/src/phy/rf/rf_blade_imp.c
+++ b/lib/src/phy/rf/rf_blade_imp.c
@@ -220,8 +220,11 @@ int rf_blade_open(char* args, void** h)
   handler->tx_stream_enabled = false;

   /* Set default sampling rates */
-  rf_blade_set_tx_srate(handler, 1.92e6);
-  rf_blade_set_rx_srate(handler, 1.92e6);
+
+  rf_blade_set_rx_srate(handler, 11.52e6);
+  rf_blade_set_tx_srate(handler, 11.52e6);
+  //rf_blade_set_rx_srate(handler, 1.92e6);
+  //rf_blade_set_rx_srate(handler, 1.92e6);

   /* Set info structure */
   handler->info.min_tx_gain = range_tx->min;
```

the calling options from srsenb
```
srsenb --expert.max_prach_offset_us=180 --rf.time_adv_nsamples=1900
```
the calling options and config from srsue
```
srsue --rf.time_adv_nsamples=200 --phy.prach_gain=-1
[rf]
freq_offset = 0
tx_gain = 80
rx_gain = 40
srate = 11.52e6
```

ue.log
```
19:40:59.749153 [PHY    ] [W] [ 2676] SFN not yet synchronized, sending out-of-sync
19:40:59.750913 [PHY    ] [W] [ 2676] SNR=1.1 dB RSRP=-91.7 dBm, sync=out-of-sync from channel estimator
19:40:59.751072 [PHY    ] [W] [ 2677] SFN not yet synchronized, sending out-of-sync
19:40:59.752431 [PHY    ] [W] [ 2677] SNR=1.1 dB RSRP=-91.7 dBm, sync=out-of-sync from channel estimator
19:40:59.752695 [PHY    ] [W] [ 2678] SFN not yet synchronized, sending out-of-sync
19:40:59.764214 [PHY    ] [W] [ 2678] SNR=1.0 dB RSRP=-91.7 dBm, sync=out-of-sync from channel estimator
19:40:59.764433 [PHY    ] [W] [ 2679] SFN not yet synchronized, sending out-of-sync
19:40:59.764860 [PHY    ] [W] [ 2679] SNR=1.0 dB RSRP=-91.7 dBm, sync=out-of-sync from channel estimator
19:40:59.765657 [PHY    ] [W] [ 2679] SNR=1.1 dB RSRP=-91.4 dBm, sync=out-of-sync from channel estimator
19:40:59.766852 [PHY    ] [W] [ 2679] SNR=1.1 dB RSRP=-91.5 dBm, sync=out-of-sync from channel estimator
```

this just goes on and on....

and enb.log
```
19:40:45.312925 [ENB    ] [I] Using binary srsenb with arguments: --expert.max_prach_offset_us=180 --rf.time_adv_nsamples=200
19:40:45.313072 [ENB    ] [I] Built in Release mode using commit 89f16eed2 on branch master.
19:40:45.350626 [RF     ] [I] Configuring physical DL channel 0 with band-pass filter (0.0, 0.0)
19:40:45.350633 [RF     ] [I] Configuring physical UL channel 0 with band-pass filter (0.0, 0.0)
19:40:45.798256 [PHY0   ] [I] [    0] Component Carrier Worker 0 configured cell 50 PRB
19:40:45.798294 [PHY0   ] [I] [    0] Worker 0 configured cell 50 PRB
19:40:45.913617 [PHY1   ] [I] [    0] Component Carrier Worker 0 configured cell 50 PRB
19:40:45.913651 [PHY1   ] [I] [    0] Worker 0 configured cell 50 PRB
19:40:46.028976 [PHY2   ] [I] [    0] Component Carrier Worker 0 configured cell 50 PRB
19:40:46.029009 [PHY2   ] [I] [    0] Worker 0 configured cell 50 PRB
19:40:46.044308 [RF     ] [I] Mapping RF channel 0 (device=0, channel=0) to logical carrier 0 on f_tx=2680.0 MHz
19:40:46.046227 [MAC    ] [I] [    0] Resetting MAC
19:40:46.046251 [RF     ] [I] Mapping RF channel 0 (device=0, channel=0) to logical carrier 0 on f_rx=2560.0 MHz
19:40:46.084462 [RRC    ] [I] CC0 SIB payload - Tx systemInformationBlockType1 (15 B)
[... 217 lines cut ...]
19:40:56.810975 [MAC    ] [I] [  504] SCHED: Msg3 tx rnti=0x4b, cc=0, pid=0, dci=(0,0), prb=[2, 5), n_rtx=0, tbs=7, bsr=0 (0-7)
19:40:56.819804 [PHY1   ] [I] [  512] PUSCH: cc=0, rnti=0x4b, rb=(2,4), nof_re=432, tbs=7, mod=2, rv=0, crc=KO, avg_iter=8.0, snr=-0.7 dB, epre=-16.6 dBfs, ta=-0.3 us, cfo=382.1 hz, t=66 us
19:40:56.819841 [MAC    ] [I] [  512] SCHED: Msg3 retx rnti=0x4b, cc=0, pid=0, dci=(0,0), prb=[2, 5), n_rtx=1, tbs=7, bsr=0 (0-7)
19:40:56.819883 [PHY1   ] [I] [  512] PHICH: rnti=0x4b, hi=0, I_lowest=2, n_dmrs=0, tti_tx_dl=516
19:40:56.827535 [PHY1   ] [I] [  520] PUSCH: cc=0, rnti=0x4b, rb=(2,4), nof_re=432, tbs=7, mod=2, rv=2, crc=KO, avg_iter=8.0, snr=0.0 dB, epre=-16.5 dBfs, ta=23.2 us, cfo=205.9 hz, t=71 us
19:40:56.827567 [MAC    ] [I] [  520] SCHED: Msg3 retx rnti=0x4b, cc=0, pid=0, dci=(0,0), prb=[2, 5), n_rtx=2, tbs=7, bsr=0 (0-7)
19:40:56.827583 [PHY1   ] [I] [  520] PHICH: rnti=0x4b, hi=0, I_lowest=2, n_dmrs=0, tti_tx_dl=524
19:40:56.835301 [PHY1   ] [I] [  528] PUSCH: cc=0, rnti=0x4b, rb=(2,4), nof_re=432, tbs=7, mod=2, rv=3, crc=KO, avg_iter=8.0, snr=-0.6 dB, epre=-16.4 dBfs, ta=0.0 us, cfo=166.6 hz, t=72 us
19:40:56.835334 [MAC    ] [I] [  528] SCHED: Msg3 retx rnti=0x4b, cc=0, pid=0, dci=(0,0), prb=[2, 5), n_rtx=3, tbs=7, bsr=0 (0-7)
19:40:56.835356 [PHY1   ] [I] [  528] PHICH: rnti=0x4b, hi=0, I_lowest=2, n_dmrs=0, tti_tx_dl=532
19:40:56.843072 [PHY1   ] [I] [  536] PUSCH: cc=0, rnti=0x4b, rb=(2,4), nof_re=432, tbs=7, mod=2, rv=1, crc=KO, avg_iter=8.0, snr=-0.8 dB, epre=-16.3 dBfs, ta=-10.0 us, cfo=613.5 hz, t=59 us
19:40:56.843085 [MAC    ] [I] [  536] SCHED: discarding UL pid=0, tti=536, maximum number of retx exceeded (4)
19:40:56.843125 [PHY1   ] [I] [  536] PHICH: rnti=0x4b, hi=0, I_lowest=2, n_dmrs=0, tti_tx_dl=540
19:40:56.888432 [RRC    ] [I] Activity timer for rnti=0x4b expired after 80 ms
19:40:56.888457 [RRC    ] [I] Disconnecting rnti=0x4b.
[... cut ...]
19:40:56.888503 [RRC    ] [I] Deallocated SR resources for time-frequency slot (0, 0)
19:40:56.888514 [RRC    ] [I] Removed user rnti=0x4b
19:40:56.897587 [MAC    ] [I] [  590] User rnti=0x4b removed from MAC/PHY
19:41:05.091391 [COMN   ] [D] [    0] RxSockets: Closing rx socket handler thread
19:41:05.092427 [COMN   ] [D] [    0] RxSockets: closed.
```

thanks for all the help. I'm just trying to a reasonable link going so i can work from there. If you think what i'm doing is just wrong, let me know.

andrepuschmann, 2021-06-25:
Hey,
that already looks a lot better.

> If you've got a way for me to independently verify that with a smart phone, let me know :).

Well, a COTS phone is always an option to calibrate the eNB side first and make sure it works. You don't need much, such a SIM card, see https://docs.srsran.com/en/latest/app_notes/source/cots_ue/source/index.html for reference.

I've noticed you are using the native bladeRF driver in your tests. To be honest I haven't used them in a while and would actually recommend the Soapy driver instead. We do all our testing with it and they work just fine.
```
device_name = soapy
device_args = driver=bladerf
```
On the eNB side we don't use any time offset parameters either.

But it should work with the native blade RF driver as well. What you're stuck at right now is likely UL issues. Maybe gain, etc. What I would recommend as well is to use `pdsch_ue/enb` first to calibrate rx/tx gains. 

Cheers
Andre

chuckixia, 2021-06-25:
I was looking into pdsch_ue/enb. But there doesn't seem to be a lot of documentation that I can find about how to use it. Can you point me to something?

ign4z, 2022-02-15:
I have the same problem here, I also tried with FPGA v0.14.0. @chuckixia  Did you manage to solve the issue?

nobletrout, 2022-04-11:
hey man: https://github.com/nobletrout/srsRAN
the SPARROW branch
This gets the bladeRF to at least receive/send with enb from UE.

closed 2023-04-14 by ismagom