srsran/srsRAN_4G #689: Cannot build 21.04 with GCC 11

FFY00, 2021-06-26:
## Issue Description ##
The latest release does not build with GCC 11. There are a few warnings, but after removing `-Werror`, I run into the following issue.

```
In file included from /build/srsran/src/srsRAN-release_21_04/lib/include/srsran/common/test_common.h:30,
                 from /build/srsran/src/srsRAN-release_21_04/lib/test/adt/observer_test.cc:23:
/build/srsran/src/srsRAN-release_21_04/lib/test/adt/observer_test.cc: In function ‘int observable_test()’:
/build/srsran/src/srsRAN-release_21_04/lib/test/adt/observer_test.cc:80:24: error: ‘class srsran::observable<M>’ has no member named ‘nof_observers’; did you mean ‘srsran::base_observable<M>::observer_list_t srsran::base_observable<M>::observers’? (not accessible from this context)
   80 |     TESTASSERT(subject.nof_observers() == 0);
      |                        ^~~~~~~~~~~~~
```

## Setup Details ##
I am using GCC 11.1.0.

## Expected Behavior ##
Code should compile :stuck_out_tongue_winking_eye: 

## Actual Behaviour ##
See log above.

## Steps to reproduce the problem ##
Just try to build with GCC 11.

## Additional Information ##
I'd recommend trying to build with an up-to-date GCC in your CI to avoid issues like this.

alvasMan, 2021-06-29:
Hi Felipe --

Thank you for reporting, but this issue is already being tracked on #675.
You can try the `fix_issue675` branch that should resolve the `nof_observers()` issue.

There is still a warning present when you compile in release mode that I still need to address, but if you are removing Werror, it might be good enough for you, for now at least.

FFY00, 2021-06-29:
I can confirm that fixes my issues. There are still a few warnings you should probably look at, though.

My issue now are failing tests. Here's the log:
https://pkgbuild.com/~ffy00/srsran-test-log.txt

andrepuschmann, 2021-11-26:
Fixed now with latest version.

closed 2021-11-26 by andrepuschmann