srsran/srsRAN_4G #693: Is it possible for COTS UE to report RSRP/RSSI back to srsenb?

knbchse, 2021-07-06:
## Issue Description ##
[Describe the issue in detail]

Hi there, I have been testing srsRAN with a bladeRF xA4 with Commercial off the shelf UE's (iPhone 6s). I have been using symsocom SIM cards. The setup is successful so far. I have been logging the srsenb and srsepc logs and looking for the parameters that are reported back from the UE to the eNB. I can see that metrics such as CQI and SNR are reported back to the eNB for each connected UE, but I am looking for more metrics. I understand that RSRP/RSSI is usually only reported back to eNB upon an A3 event (handover). Is it possible to make changes to code such that the UE reports RSRP/RSSI back to the eNB consistently? 

Thanks!

## Setup Details ##
[Specify details of the test setup. This would help us reproduce the problem reliably]
e.g. Network configuration, Operation System, Hardware, RF front-end, library and driver versions
- Running srsenb on Ubuntu 20.04 Server on Raspberry Pi 4 (followed RPi4 application note by srsRAN)
- Running srsepc on separate machine (Ubuntu 20.04)
- RF front-end: bladeRF xA4

## Expected Behavior ##
[What you expect to happen]
Reporting more RF metrics such as RSSI/RSRP etc...

## Actual Behaviour ##
[What happens instead e.g. error message]
Limited metrics reported back

## Steps to reproduce the problem ##
[Tell us how to reproduce this issue e.g. RF setup, application config files]
Run trace on srsenb and observe the results, as well as enb log files.

## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]
N/A

ph03n1xAu, 2021-07-08:
Hi @knbchse , 
This is not the answer for your question but I need some help from you for this implementation.

I'm trying to use iphone6s (sysmocom usim) from last few weeks using LimeSDR-mini as enb.  Unfortunately, my phone failed to discover the network even with different combination of mcc-mnc/dl_earfcn. I'm using ubuntu 20.04 as well. 

Would you mind to share a snippet of the config files from your setup so that I can verify what I'm missing. I believe, I did everything as per the documentation for COTS-UE setup.

Thanks...

knbchse, 2021-07-09:
Hey @ph03n1xAu 

Sure! The LimeSDR setup may be a bit different, but I primarily followed this guide for the Raspberry Pi 4: https://docs.srslte.com/en/latest/app_notes/source/pi4/source/index.html

The bladeRF setup was fairly simple. The mcc-mnc I am using is default 901-70 supplied with the uSIM cards, I didn't think they would work with a different combination. I am aware that different iPhone models have certain LTE bands they can receive. My 6S (Model A1688) allows me to connect to dl_earfcn = 1934. However when I try this with my iPhone 5S, it won't pick up the network. I often find that if I turn airplane mode on whenever I am not wanting to connect to the eNB makes it easier to connect to the network again. I also go into Network selection and turn off auto to see if I can see the Software Radio Systems network (sometimes called 901-70 in the network list). The connection process often takes a couple of minutes for me, and then the EPC log will say that it is sending S1 response. Usually the phone just connects after a couple of minutes, sometimes without having to turn off auto network selection. Search up your exact phone model and what LTE bands it can receive and then change your dl_earfcn to suit (if the band is free to use of course!).

I have attached my enb and epc config files as txt files below. It is worth noting that I am running the eNB on the Raspberry Pi 4 (Ubuntu Server 20.04) and the EPC on a separate Ubuntu 20.04 machine (connected via ethernet). 

Cheers

[epc.txt](https://github.com/srsran/srsRAN/files/6788785/epc.txt)
[enb.txt](https://github.com/srsran/srsRAN/files/6788786/enb.txt)

ph03n1xAu, 2021-07-09:
Hi @knbchse

Thanks so much for your detailed response and config files. I highly appreciate your help :-)

I couldn't find any major difference in my config file. At the moment, I'm not sure is this because of LimeSDR or not. As I'm having some stability problem using LimeSDR with srsENB and srsUE. The UE-limesdr disconnect automatically and send reattach request to enb/epc every 3/4 minutes. I'm now analyzing the debug log to understand the problem.

For COTS-UE, as you confirmed it worked in your case, I'm hopeful about that. I reprogrammed the USIM for different mcc-mnc as well to test. Looks like either I have to switch hardware or continue experiment with LimeSDR to find the issues.

Thanks again, Cheers!!

knbchse, 2021-07-09:
No worries! 

> I'm not sure is this because of LimeSDR or not. As I'm having some stability problem using LimeSDR with srsENB and srsUE. The UE-limesdr disconnect automatically and send reattach request to enb/epc every 3/4 minutes. I'm now analyzing the debug log to understand the problem.

Perhaps follow the guide I linked above. It covers how to setup Ubuntu with LimeSuite and release_19_12 of srsRAN. Perhaps it is worth starting a fresh install of srsRAN and Lime drivers?
I haven't tried setting up any SDRs with srsUE yet. I think I've primarily had time-out errors with my setup. I believe this is because the Pi is struggling to transmit the frames in time. I have had instances where the COTS-UE loses connection and reattaches, I assumed this is because of the the Raspberry Pi is struggling to keep up with the processing. I will investigate this further, and report back what I find...

> For COTS-UE, as you confirmed it worked in your case, I'm hopeful about that. I reprogrammed the USIM for different mcc-mnc as well to test. Looks like either I have to switch hardware or continue experiment with LimeSDR to find the issues.

Oh I see, you got the reprogrammable uSIM, good idea! I am yet to see if the system remains stable for longer periods of time.

Hope this helps!

ph03n1xAu, 2021-07-20:
Hi @knbchse ,

Finally my iphone6s (A1688) with sysmocom SIM worked with srsRAN :-D
I figured this is because of the frequency..at the moment it works only with dl_earfcn = 3350. I'm yet to find out why as from the specs it should work with other earfcn as well..

The other issue I'm currently having is the stability problem with LimeSDR as enb. UE loses connection and tried to reconnect every 2/3 minutes. Instability problem is same for  srsRAN_release_19_12 and latest. I assume something else is going on the background.

However, I hope someone can resolve your problem for RF metrics soon. If I found anything useful, I'll let you know

Cheers!!

knbchse, 2021-07-21:
Hey @ph03n1xAu 

Thats good news! I recently upgraded to latest release, to see if this release passed RSRP through during A3 events. I also found that I've been having stability issues. I have found that there are a few things that effect stability:
- When I try access web pages/stream data using 'Mobile Data', the EPC log will give a service reject (and Short MAC Invalid) message and will crash. I have been turning off mobile data to enhance stability.
- The higher the PRB, the less stable the system is. I have reduced the PRB down to 6 (from 15) in enb.conf. This seems to help with stability. Probably because the Raspberry Pi isn't very computationally powerful

This is just what I've noticed. It'll be interesting to see what you find with regards to enhancing stability.

Cheers

ph03n1xAu, 2021-07-21:
Hello @knbchse :-)

Thanks for sharing your experience. I saw the same behaviour while trying to browse internet from mobile but with different error message. In my case, EPC console flooded with either 'paging failure message' or 'radio connection lost with UE' in ENB console. I'm yet to figure out the root cause. Submitted a new issue with all the logs, let's see if someone can point me into right direction.

The other thing I am little bit confused about the frequency this handset selects. Why it works only when earfcn = 3350? Theoretically it should search for other supported LTE band as well, isn't it? But it sticks with the same earfcn that my commercial SIM used to connect.

I've tried with prb=6/25 before but srsUE app didn't respond that time. I'll test again with lower PRB today for iphone and will let you know the results. Additionally, I'm planning to install this on Intel NUC with Ubuntu 20.04 as host OS. I hope that will boost the performance and that will answer few questions about LimeSDR-mini as RF-frontend. I'm really enjoying all these experiments while learning a ton everyday :-)

knbchse, 2021-07-23:
> Thanks for sharing your experience. I saw the same behaviour while trying to browse internet from mobile but with different error message. In my case, EPC console flooded with either 'paging failure [...]

Fingers crossed! I only just found this srsRAN user archive list which has a heap of resources available, just search your keywords: https://lists.softwareradiosystems.com/pipermail/srslte-users/

> The other thing I am little bit confused about the frequency this handset selects. Why it works only when earfcn = 3350? Theoretically it should search for other supported LTE band as well, isn't it? But it sticks with the same earfcn that my commercial SIM used to connect.

Interesting, I haven't actually had any luck using earfcn = 3350 with my iPhone. Mine only seems to connect with earfcn = 1937. So maybe your theory is correct.

> I've tried with prb=6/25 before but srsUE app didn't respond that time. I'll test again with lower PRB today for iphone and will let you know the results. Additionally, I'm planning to install this [...]

I would be interested to see if the NUC enhances stability! Agreed! It is a lot of fun :)

nomidaepapi, 2021-08-12:
Hi, Have you achieved a X2 Handover with the Bladerfs? If so, could you share the RR.conf files?

knbchse, 2021-08-16:
> Hi, Have you achieved a X2 Handover with the Bladerfs? If so, could you share the RR.conf files?

No I haven't. From memory, ZeroMQ is the only supported way to 'simulate' handover. I haven't looked into the topic for a while. Have you managed to obtain two bladeRF's? If so, have you got the same user_db.csv file between them? I'm inclined to believe that having two nearly identical bladeRF's setup as **separate** eNB's would be enough to trigger an A3 event, and therefore report back RSRP...

joeldelosangeles, 2021-09-22:
Hi I noticed that Version X2/S1 handover in srsUE is supported since version 17.12.  Does srsenb support X2?  I saw S1 support in your handover application note.

andrepuschmann, 2021-09-22:
> Hi I noticed that Version X2/S1 handover in srsUE is supported since version 17.12. Does srsenb support X2? I saw S1 support in your handover application note.

No, X2 handover is not supported in srsENB

closed 2023-04-14 by ismagom

joeldelosangeles, 2023-08-31:
Since srsran now has NSA and X2 for gnodeb, is X2 for enodeb to enodeb handover now supported?