srsran/srsRAN_4G #694: srsue can't success Found Cell when using --rf.srate option

ShunOkahiro, 2021-07-09:
## Issue Description ##
When launching srsue with --rf.srate option,  ue can't success Found Cell.
I want to change srate of Found cell 11.52M to 1.92M so I tried "sudo srsue --rf.srate 1.92e06 ue.conf", however it doesn't work.

## Setup Details ##

[ue]
Ubunt18.04LTS
bladeRF2.0micro xA4

[enb,epc]
Ubunt18.04LTS
USRPb210

## Expected Behavior ##

Ue successes Found Cell process.

Bellow is prompt message when execute srsue without --rf.srate option

```
tcu@tcu-PC-VK27MCZCM:~/srsLTE-release_20_10_1/srsue$ sudo srsue ue.conf
[sudo] password for tcu: 
Reading configuration file ue.conf...
WARNING: cpu0 scaling governor is not set to performance mode. Realtime processing could be compromised. Consider setting it to performance mode before running the application.

Built in Release mode using 20.10.1.

Opening 1 channels in RF device=bladeRF with args=default
Opening bladeRF...
Set RX sampling rate 1.92 Mhz, filter BW: 1.92 Mhz
Waiting PHY to initialize ... done!
Attaching UE...
Set RX sampling rate 1.92 Mhz, filter BW: 1.92 Mhz
set RX frequency to 780499998
set TX frequency to 725499998
.
RF status: O=2, U=0, L=0
...
Found Cell:  Mode=FDD, PCI=1, PRB=50, Ports=1, CFO=-0.8 KHz
RF status: O=3, U=0, L=0
Set RX sampling rate 11.52 Mhz, filter BW: 9.22 Mhz
RF status: O=6, U=0, L=0
Found PLMN:  Id=90170, TAC=7
Could not find Home PLMN Id=00101, trying to connect to PLMN Id=90170
Random Access Transmission: seq=12, ra-rnti=0x2
Random Access Complete.     c-rnti=0x46, ta=10
RRC Connected
RF status: O=2, U=0, L=0
Network attach successful. IP: 172.16.0.2
Software Radio Systems LTE (srsLTE) 9/7/2021 2:2:29 TZ:0

```

## Actual Behaviour ##

Ue can't success Found Cell.
Bellow is prompt message when execute srsue with --rf.srate option

```
tcu@tcu-PC-VK27MCZCM:~/srsLTE-release_20_10_1/srsue$ sudo srsue --rf.srate 1.92e06 ue.conf
Reading configuration file ue.conf...
WARNING: cpu0 scaling governor is not set to performance mode. Realtime processing could be compromised. Consider setting it to performance mode before running the application.

Built in Release mode using 20.10.1.

Opening 1 channels in RF device=bladeRF with args=default
Opening bladeRF...
Set RX sampling rate 1.92 Mhz, filter BW: 1.92 Mhz
Waiting PHY to initialize ... done!
Attaching UE...
Set RX sampling rate 1.92 Mhz, filter BW: 1.92 Mhz
set RX frequency to 780499998
set TX frequency to 725499998
.
RF status: O=2, U=0, L=0
.
Found Cell:  Mode=FDD, PCI=1, PRB=50, Ports=1, CFO=-0.4 KHz
.
Found Cell:  Mode=FDD, PCI=1, PRB=50, Ports=1, CFO=0.4 KHz
.
Found Cell:  Mode=FDD, PCI=1, PRB=50, Ports=1, CFO=0.3 KHz
.
Found Cell:  Mode=FDD, PCI=1, PRB=50, Ports=1, CFO=0.3 KHz
.
Found Cell:  Mode=FDD, PCI=1, PRB=50, Ports=1, CFO=0.6 KHz
^CStopping ..
---  exiting  ---

```

## Steps to reproduce the problem ##
ue HW=bladeRF2.0 micro xA4
$ sudo srsue --rf.srate 1.92e06 ue.conf

## Additional Information ##

DancingSW, 2023-02-10:
In my case, I was using srate=11.52e6 but had a similar problem. After I read all the logs, I realized that the UE does not transmit UL control traffic somehow. Eventually, I could solve the problem after I changed from a native bladerf driver to soapy driver.

Originally, I used the following config:
**[ue.config]**
device_name = bladerf

Then, I changed it as follows:
**[ue.config]**
device_name = soapy
device_args = driver = bladerf

If you installed bladerf library through pybombs, you might need to[ install bladerf library and soapySDR](https://github.com/pothosware/SoapyBladeRF) again.

closed 2023-04-14 by ismagom

SeaBreezing, 2025-06-09:
@ShunOkahiro  Could you kindly share your epc.conf, enb.conf and ue.conf? I found my ue fail to attach on BladeRF and wish to test on a successful configuration.