srsran/srsRAN_4G #695: srsue can't connect Internet when turn-off wifi of UE host PC

GodRadioMagic, 2021-07-09:
## Issue Description ##
After successing launch srsue and successing ping 172.16.0.1(default gw) and successing access to the Internet via default gw(ie. ping www.google.com), once turn-off the wifi of UE host PC, ue get not to be able to connent to the Internet( but host PC can ping 172.16.0.1 and default gw is still 172.16.0.1 dev tunsrsue).
I understand that ue connects to the Internet via srsran, so it isn't necessary that ue host PC connect to the Internet via its wifi.

Why does this happen?

## Setup Details ##

[ue]
Ubunt18.04LTS
bladeRF2.0micro xA4
Host PC connects to the Internet via wifi

[enb,epc]
Ubunt18.04LTS
USRPb210
Host PC connects to the Internet via wifi

## Expected Behavior ##
Even after ue host PC turn-off the wifi, ue host pc continue to connect to the Internet (ie can ping www.google.com)

## Actual Behaviour ##
```
user~$ ping www.google.com
ping: www.google.com: Name or service not known
```

## Steps to reproduce the problem ##

[enb,epc hostPC]
wifi turn-on , connect to the Internet
```
cd ~/srsLTE-release_20_10_1
sudo srsepc_if_masq.sh wlp6s0
```

[epc]
```
cd ~/srsLTE-release_20_10_1/srsepc
sudo srsepc epc.conf

```
[enb]
```
cd ~/srsLTE-release_20_10_1/srsenb
sudo srsenb enb.conf
```

[ue host PC]
wifi turn-on , connect to the Internet

[ue]

```
cd ~/srsLTE-release_20_10_1/srsue
sudo srsue ue.conf

sudo ip route del default
sudo ip route add default via 172.16.0.1

```
Then, traceroute and ping to Google
```
~$ traceroute youtube.com
traceroute to youtube.com (172.217.175.14), 30 hops max, 60 byte packets
 1  _gateway (172.16.0.1)  24.659 ms  24.667 ms  24.657 ms
 2  * * *
...

~$ ping www.google.com
PING google.com (172.217.31.142) 56(84) bytes of data.
64 bytes from nrt20s08-in-f14.1e100.net (172.217.31.142): icmp_seq=1 ttl=112 time=62.8 ms
64 bytes from nrt20s08-in-f14.1e100.net (172.217.31.142): icmp_seq=2 ttl=112 time=82.5 ms
64 bytes from nrt20s08-in-f14.1e100.net (172.217.31.142): icmp_seq=3 ttl=112 time=100 ms

```

Then, turn off the wifi, ue host PC can't connect to the internet via gw 172.16.0.1.

```
~$ traceroute youtube.com
youtube.com: Name or service not known
Cannot handle "host" cmdline arg `youtube.com' on position 1 (argc 1)

~$ ping www.google.com
ping: www.google.com: Name or service not known

```

## Additional Information ##

mooction, 2021-07-20:
I suspect after your turning off the WiFi, the NetworkManager rewritten the route table, so your manually added rule is cleared. Try this command after you turn off the WiFi:

`ip route get 8.8.8.8`

GodRadioMagic, 2021-07-21:
You were right.
I tried set default gw via 172.16.0.1 again, then srsue can connect to the Internet via it.

Thank you very much.

### I tried. ###

```
~$ sudo srsue ue.conf
~$ sudo ip route del default
~$ sudo ip route add default via 172.16.0.1
```
disconnect the wifi of UE PC.
Then, set the default gw again;
```
~$ sudo ip route add default via 172.16.0.1
~$ traceroute www.google.com
traceroute to www.google.com (172.217.161.68), 30 hops max, 60 byte packets
 1  _gateway (172.16.0.1)  29.759 ms  29.758 ms  29.750 ms
 2  * * *

```

GodRadioMagic, 2021-09-09:
When I tried again using the method described above, I could not connect to the Internet. There may be other reasons.

ismagom, 2023-04-14:
This is unrelated to srsRAN

closed 2023-04-14 by ismagom

natasyanurulputri, 2023-06-26:
hello, i have the same problem with you. i already did the steps. But the UE still can't connect the internet
how i can solve this problem?
Thankyou