srsran/srsRAN_4G #698: Cat-1 sample configuration files

Quzard, 2021-07-12:
I have a camera which uses LTE Cat-1(LTE-TDD：B34/38/39/40/41LTE-FDD：B1/3/5/8)
 
However, it is unable to connect srsran ( dl_earfcn: 1650)

Does srsran support cat 1?

andrepuschmann, 2021-07-12:
No, we don't support cat M1

Quzard, 2021-07-13:
> No, we don't support cat M1

My camera uses  LTE Cat 1, not cat M1. You don't support LTE Cat 1 too?

andrepuschmann, 2021-07-13:
> My camera uses LTE Cat 1, not cat M1. You don't support LTE Cat 1 too?

Ahh sorry, I misread this. Will reopen it again.

Too be honest, I am not very familiar with Cat 1 and we've never tried a Cat 1 UE. At a first glance it looks the PHY is completely the same, just some limitations in TB sizes, etc., which could be addressed with MCS limitations probably. So in theory, it could work. Have you tried it? What is the eNB saying? Please provide logs if you see something.

Quzard, 2021-07-13:
> Ahh sorry, I misread this. Will reopen it again.
>
> Too be honest, I am not very familiar with Cat 1 and we've never tried a Cat 1 UE. At a first glance it looks the PHY is completely the same, just [...]

There is no connection log in the epc.pcap. I think my camera with Cat 1 don't try to connect srsran.

avalori1, 2023-09-25:
@andrepuschmann, adding here rather than opening a new issue since it is exactly the right topic, and there is already some hint that helped us.

We are trying CAT1 (FDD). 
With "standard" parameters, we manage to connect, but with very high errors. 
As you suggested, we tried to address the issue with mcs limitations (see below). Indeed, it worked, and we manage to transmit without errors. Of course, at this point the, total bandwidth is limited and if there is also a CAT4 modem is limited too. 

It seems that SRS does not recognize that the module is a CAT1 and does not modulate the mcs as required... is that a potential explanation?

Regards

andrepuschmann, 2023-09-25:
> It seems that SRS does not recognize that the module is a CAT1 and does not modulate the mcs as required... is that a potential explanation?

Yes, this sounds plausible. We don't differentiate between the Cat 1 and Cat 4 UEs in the scheduler. For reference, could you share a MAC PCAP when the UE attaches and the UE capabilities are exchanged?

avalori1, 2023-12-06:
Hi @andrepuschmann,
I guess this is the exchange of information you are referring to, correct? It seems to confirm that the UE declares its category correctly as CAT-1. 

Please let me know if yo need something else and if you have any plan to include the differentiation between categories in the scheduler.

Regards

    Frame 31: 551 bytes on wire (4408 bits), 551 bytes captured (4408 bits)
    DLT: 149, Payload: udp (User Datagram Protocol)
    User Datagram Protocol, Src Port: 48879, Dst Port: 57005
    MAC-LTE UL-SCH: (SFN=807 , SF=0) UEId=0   (Power Headroom Report) (Short BSR) (1:2 bytes) (1:152 bytes) (Padding:remainder)    ||   
        [Context (RNTI=71)]
        MAC PDU Header (Power Headroom Report) (Short BSR) (1:2) (1:152) (Padding:remainder)  [5 subheaders]
        Power Headroom Report (28 <= PH < 29)
        Short BSR (lcgid=0  BS = 0)
        RLC-LTE [UL] [AM] SRB:1  [CONTROL]  ACK_SN=6   
        RLC-LTE [UL] [AM] SRB:1  [DATA] (P) sn=7     [150-bytes]
            [Context]
            AM Header  (P) sn=7   
            [Sequence Analysis - OK]
            PDCP-LTE sn=6  
                [Configuration: (direction=Uplink, plane=Signalling)]
                [UE Security (ciphering=EEA0 (NULL), integrity=EIA2 (AES))]
                000. .... = Reserved: 0
                ...0 0110 = Seq Num: 6
                [Sequence Analysis - OK]
                LTE Radio Resource Control (RRC) protocol
                    UL-DCCH-Message
                        message: c1 (0)
                            c1: ueCapabilityInformation (7)
                                ueCapabilityInformation
                                    rrc-TransactionIdentifier: 0
                                    criticalExtensions: c1 (0)
                                        c1: ueCapabilityInformation-r8 (0)
                                            ueCapabilityInformation-r8
                                                ue-CapabilityRAT-ContainerList: 1 item
                                                    Item 0
                                                        UE-CapabilityRAT-Container
                                                            rat-Type: eutra (0)
                                                            ueCapabilityRAT-Container: c8180104cd820018083081c40b18348923064850ffff95ffe87fffcafff43fffe57ffa1f…
                                                                UE-EUTRA-Capability
                                                                    accessStratumRelease: rel10 (2)
                                                                    ue-Category: 1
                                                                    pdcp-Parameters
                                                                    phyLayerParameters
                                                                    rf-Parameters
                                                                    measParameters
                                                                    featureGroupIndicators: 7fcffeba [bit length 32, 0111 1111  1100 1111  1111 1110  1011 1010 decimal value 2144337594]
                                                                    interRAT-Parameters
                                                                    nonCriticalExtension