srsran/srsRAN_4G #701: Unhandled NAS integrity protected message LIBLTE_MME_MSG_TYPE_PDN_CONNECTIVITY_REQUEST

mooction, 2021-07-20:
## Issue Description ##

```text
Initial UE message: NAS Message Type Unknown
Received Initial UE message -- Service Request
Service request -- S-TMSI 0x15e33d1a
Service request -- eNB UE S1AP Id 6
Service Request -- Short MAC valid
Service Request -- User is ECM DISCONNECTED
UE previously assigned IP: 172.16.0.4
Generating KeNB with UL NAS COUNT: 4
UE Ctr TEID 0
Sent Initial Context Setup Request. E-RAB id 5 
Received Initial Context Setup Response
E-RAB Context Setup. E-RAB id 5
E-RAB Context -- eNB TEID 0x60c05; eNB GTP-U Address 192.168.23.139
Initial Context Setup Response triggered from Service Request.
Sending Modify Bearer Request.
Received GTP-C PDU. Message type: GTPC_MSG_TYPE_MODIFY_BEARER_REQUEST
Unhandled NAS integrity protected message LIBLTE_MME_MSG_TYPE_PDN_CONNECTIVITY_REQUEST
Unhandled NAS integrity protected message LIBLTE_MME_MSG_TYPE_PDN_CONNECTIVITY_REQUEST
Unhandled NAS integrity protected message LIBLTE_MME_MSG_TYPE_PDN_CONNECTIVITY_REQUEST
Unhandled NAS integrity protected message LIBLTE_MME_MSG_TYPE_PDN_CONNECTIVITY_REQUEST
Unhandled NAS integrity protected message LIBLTE_MME_MSG_TYPE_PDN_CONNECTIVITY_REQUEST
Unhandled NAS integrity protected message LIBLTE_MME_MSG_TYPE_PDN_CONNECTIVITY_REQUEST
UL NAS: Detach Request
Detach request -- IMSI 901011234567890
Received GTP-C PDU. Message type: GTPC_MSG_TYPE_DELETE_SESSION_REQUEST
Received UE Context Release Complete. MME-UE S1AP Id 10
UE Context Release Completed.
```

[image]

## Setup Details ##

EPC: Deepin 20.2.1 (upstream Debian 10.9), commit https://github.com/srsran/srsRAN/commit/89f16eed2b7f2e6c45e675298f5829c230698e1d

UE: Xiaomi 8 (Android 10, SDK 29)

eNB: purchased HeNB

USIM: purchased magic USIM

## Expected Behavior ##

UE attached and network access

## Actual Behaviour ##

UE attached without network access

## Steps to reproduce the problem ##

## Additional Information ##

I found some similar issues but there's no reply:

https://lists.softwareradiosystems.com/pipermail/srslte-users/2019-April/002530.html
https://lists.softwareradiosystems.com/pipermail/srslte-users/2020-November/003926.html
https://lists.softwareradiosystems.com/pipermail/srslte-users/2021-February/004112.html

mooction, 2021-07-20:
[image]

Melon08P4, 2021-11-10:
Did u resolve this problem ?