srsran/srsRAN_4G #703: Stable connectivity problem with COTS-UE

ph03n1xAu, 2021-07-21:
## Issue Description ##

I'm having a stable connectivity problem with COTS-UE. Error messages are like, radio-connection-with-ue-lost or could not page the ue. I'm using LimeSDR-mini as RF-frontend. I've attached all the logs including console messages in the attached zip file. [enb_epc_log.zip](https://github.com/srsran/srsRAN/files/6853188/enb_epc_log.zip)

Could you please have a look and suggest me is this because of LimeSDR problem or something else. 

## Setup Details ##

Ubuntu 20.04 running on vmware workstation and LimeSDR-mini as RF-frontend 

Host OS            : VM OS- Ubuntu 20.04, 8 GB RAM, Core : 6
RF Frontend    : LimeSDR Mini v1.2 ​
COTS- UE         : iphone6s (Model A1688)
SoapySDR Lib  : v0.7.2
LimeUtil Lib     : v20.10.0-ga45e482d
srsRAN             : commit 89f16eed2 on branch master

1. All default config options except frequency, dl_earfcn: 3350
2. Sysmocom USIM with iphone6s
3. All log level set as debug (attached epc/enb log)

## Expected Behavior ##
Stable connectivity between enb and UE

## Actual Behaviour ##

After successful authentication with the network, connectivity between enb and ue dropped 4/5 minutes later every time using iphone6s.  For first few minutes ping works and then epc console flooded with below messages 

```
Found UE for Downlink Notification 
MME Ctr TEID 0x1, IMSI: 901700000048363
T3413 expired -- Could not page the ue.
Received GTP-C PDU. Message type: GTPC_MSG_TYPE_DOWNLINK_DATA_NOTIFICATION_FAILURE_INDICATION
Found UE for Downlink Notification 
MME Ctr TEID 0x1, IMSI: 901700000048363
T3413 expired -- Could not page the ue.
Received GTP-C PDU. Message type: GTPC_MSG_TYPE_DOWNLINK_DATA_NOTIFICATION_FAILURE_INDICATION
Found UE for Downlink Notification 
MME Ctr TEID 0x1, IMSI: 901700000048363
T3413 expired -- Could not page the ue.
Received GTP-C PDU. Message type: GTPC_MSG_TYPE_DOWNLINK_DATA_NOTIFICATION_FAILURE_INDICATION
Found UE for Downlink Notification 
MME Ctr TEID 0x1, IMSI: 901700000048363
T3413 expired -- Could not page the ue.
Received GTP-C PDU. Message type: GTPC_MSG_TYPE_DOWNLINK_DATA_NOTIFICATION_FAILURE_INDICATION
Found UE for Downlink Notification 
MME Ctr TEID 0x1, IMSI: 901700000048363
T3413 expired -- Could not page the ue.
Received GTP-C PDU. Message type: GTPC_MSG_TYPE_DOWNLINK_DATA_NOTIFICATION_FAILURE_INDICATION
```

Enb_console_message:

```
User 0x46 connected
RACH:  tti=3141, cc=0, preamble=38, offset=12, temp_crnti=0x47
User 0x47 requesting RRC Reestablishment as 0x46. Cause: otherFailure
Disconnecting rnti=0x46.
User 0x47 connected
RACH:  tti=5981, cc=0, preamble=24, offset=12, temp_crnti=0x48
User 0x48 requesting RRC Reestablishment as 0x47. Cause: otherFailure
Disconnecting rnti=0x47.
User 0x48 connected
RACH:  tti=1221, cc=0, preamble=40, offset=12, temp_crnti=0x49
User 0x49 requesting RRC Reestablishment as 0x48. Cause: otherFailure
Disconnecting rnti=0x48.
User 0x49 connected
RACH:  tti=5681, cc=0, preamble=16, offset=12, temp_crnti=0x4a
User 0x4a requesting RRC Reestablishment as 0x49. Cause: otherFailure
Disconnecting rnti=0x49.
```

## Steps to reproduce the problem ##

1. Iphone6s with programmed sysmocom USIM
2. enb.conf:
  dl_earfcn = 3350
  tx_gain = 70
  rx_gain = 50
  device_name = soapy
	
  Host OS           : VM OS- Ubuntu 20.04, 8 GB RAM, Core : 6
  RF Frontend     : LimeSDR Mini v1.2 ​
  COTS- UE         : iphone6s (Model A1688)
  SoapySDR Lib  : v0.7.2
  LimeSuite Lib   : v20.10.0-ga45e482d
  srsRAN             : commit 89f16eed2 on branch master

3. After successful connection, wait for 2/3 minutes then ping from gateway EPC host to UE IP address.

## Additional Information ##

Same instability problem with different srsRAN release for limeSDR-mini as RF-frontend.

knbchse, 2021-07-26:
Hey @ph03n1xAu 

Are you always running the IP-masquerade command before you start epc/enb?

I find that I get this issue only when I run the IP-masquerade before starting epc and enb. I usually run them without the masquerade script.

Cheers

ph03n1xAu, 2021-08-02:
Hi @knbchse, 
Sorry for late response. Yeah for me the problem is same with or without IP-masquerade script. Connection is extremely unstable.

ph03n1xAu, 2021-08-02:
Hi @andrepuschmann ,

Could you please help me to identify the problem; I've attached all the log messages here in my first post. 

I checked this again today with different set of LimeSDR-mini, I got the same error message (T3413 expired -- Could not page the ue) after 2/3 minutes of successful attach.

ph03n1xAu, 2021-09-14:
It turns out LimeSDR mini temperature above 55c impacts the RF performance on enb. After using heatsink and cooling fan, the enb and iphone was connected for 5 hours. Looks like it's working fine and didn't receive the "T3413 expired" error during that time.

closed 2021-09-14 by ph03n1xAu