srsran/srsRAN_4G #712: compilation error: newmask.srsran::bounded_bitset may be used uninitialized in this function

kaoh, 2021-08-03:
> [ 90%] Building CXX object srsenb/src/stack/mac/CMakeFiles/srsenb_mac.dir/sched_grid.cc.o
In file included from /home/ubuntu/srsRAN/srsenb/hdr/stack/mac/sched_phy_ch/../sched_common.h:25,
                 from /home/ubuntu/srsRAN/srsenb/hdr/stack/mac/sched_phy_ch/sched_result.h:25,
                 from /home/ubuntu/srsRAN/srsenb/hdr/stack/mac/sched_grid.h:26,
                 from /home/ubuntu/srsRAN/srsenb/src/stack/mac/sched_grid.cc:22:
/home/ubuntu/srsRAN/lib/include/srsran/adt/bounded_bitset.h: In member function ‘srsenb::alloc_result srsenb::sf_grid_t::alloc_ul_data(srsenb::sched_ue*, srsenb::prb_interval, bool, bool)’:
/home/ubuntu/srsRAN/lib/include/srsran/adt/bounded_bitset.h:336:32: error: ‘newmask.srsran::bounded_bitset<100, true>::buffer[<unknown>]’ may be used uninitialized in this function [-Werror=maybe-uninitialized]
  336 |       buffer[i] &= other.buffer[i];
      |                    ~~~~~~~~~~~~^
cc1plus: all warnings being treated as errors
make[2]: *** [srsenb/src/stack/mac/CMakeFiles/srsenb_mac.dir/build.make:115: srsenb/src/stack/mac/CMakeFiles/srsenb_mac.dir/sched_grid.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:10958: srsenb/src/stack/mac/CMakeFiles/srsenb_mac.dir/all] Error 2
make: *** [Makefile:163: all] Error 2

## Issue Description ##

I cannot compile srsRAN. I'm using an  "Intel(R) Atom(TM) x5-Z8350" and have to manually compile srsRAN. The provided Ubuntu package `srsENB` is throwing an "Illegal instruction" otherwise.

Test Setup:

* Ettus USRP B205mini-i 
* Intel(R) Atom(TM) x5-Z8350 
*  Ubuntu 20.04 server

## Expected Behavior ##

Compilation succeeds.

## Actual Behaviour ##

> /home/ubuntu/srsRAN/lib/include/srsran/adt/bounded_bitset.h:336:32: error: ‘newmask.srsran::bounded_bitset<100, true>::buffer[<unknown>]’ may be used uninitialized in this function [-Werror=maybe-uninitialized]
  336 |       buffer[i] &= other.buffer[i];
      |                    ~~~~~~~~~~~~^
cc1plus: all warnings being treated as errors

## Steps to reproduce the problem ##

Follow [build instructions](https://docs.srslte.com/en/latest/general/source/1_installation.html) and execute `make`

kaoh, 2021-08-17:
A workaround for this is to edit `lib/include/srsran/adt/bounded_bitset.h` at line 335:

~~~c
#pragma GCC diagnostic push
#pragma GCC diagnostic ignored "-Wmaybe-uninitialized"
    for (size_t i = 0; i < nof_words_(); ++i) {
      buffer[i] &= other.buffer[i];
    }
#pragma GCC diagnostic pop
~~~

I don't know which assert instruction to use to prevent this compilation error otherwise.

LucasPAmaral, 2023-09-13:
## Error at ```make``` command (compilation related)

### Description 

This error is happening at the ```make``` command.

```
[ 85%] Building CXX object srsenb/src/stack/mac/CMakeFiles/srsenb_mac.dir/sched_grid.cc.o
In file included from /home/c2n/srsran/srsRAN/srsenb/hdr/stack/mac/sched_phy_ch/../sched_lte_common.h:26,
                 from /home/c2n/srsran/srsRAN/srsenb/hdr/stack/mac/sched_phy_ch/sched_result.h:25,
                 from /home/c2n/srsran/srsRAN/srsenb/hdr/stack/mac/sched_grid.h:27,
                 from /home/c2n/srsran/srsRAN/srsenb/src/stack/mac/sched_grid.cc:22:
/home/c2n/srsran/srsRAN/lib/include/srsran/adt/bounded_bitset.h: In member function ‘srsenb::alloc_result srsenb::sf_grid_t::alloc_ul_data(srsenb::sched_ue*, srsenb::prb_interval, bool, bool)’:
/home/c2n/srsran/srsRAN/lib/include/srsran/adt/bounded_bitset.h:336:32: error: ‘newmask.srsran::bounded_bitset<100, true>::buffer[<unknown>]’ may be used uninitialized in this function [-Werror=maybe-uninitialized]
  336 |       buffer[i] &= other.buffer[i];
      |                    ~~~~~~~~~~~~^
cc1plus: all warnings being treated as errors
make[2]: *** [srsenb/src/stack/mac/CMakeFiles/srsenb_mac.dir/build.make:132: srsenb/src/stack/mac/CMakeFiles/srsenb_mac.dir/sched_grid.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:12029: srsenb/src/stack/mac/CMakeFiles/srsenb_mac.dir/all] Error 2
make: *** [Makefile:166: all] Error 2
```

### Workarounds

Tried before using gcc-10 as in [here](https://github.com/srsran/srsRAN_4G/issues/1116), but without success. 
[This workarround](https://github.com/srsran/srsRAN_4G/issues/712) is curently working.