srsran/srsRAN_4G #714: srsUE hangs, RLC mutex cannot be acquired

mrlnc, 2021-08-06:
## Issue Description ##

srsUE hangs after trying to send Authentication Response. I've added a couple of debugging-prints and noticed that RLC stops sending data while waiting for the RX/TX mutexes.

`TX:write_sdu` can't deliver the Authentication Response because it cannot lock the TX mutex:
```
14:31:08.180088 [RLC    ] [I] trying to lock mutex 0x7fab7715b7e8 from int srsran::rlc_am_lte::rlc_am_lte_tx::write_sdu(srsran::unique_byte_buffer_t)
```

Seems it is still locked by `TX::get_buffer_state()`
```
14:31:08.033315 [RLC    ] [I] trying to lock mutex 0x7fab7715b7e8 from uint32_t srsran::rlc_am_lte::rlc_am_lte_tx::get_buffer_state()
14:31:08.033316 [RLC    ] [I] mutex 0x7fab7715b7e8 acquired from uint32_t srsran::rlc_am_lte::rlc_am_lte_tx::get_buffer_state()
```

`TX:get_buffer_state()` can't free the lock because it is waiting for the RX mutex:
```
14:31:08.033317 [RLC    ] [I] trying to lock mutex 0x7fab7715b870 from int srsran::rlc_am_lte::rlc_am_lte_rx::get_status_pdu_length()
```

The `RX` mutex is still locked by `RX:write_pdu()` to process the incoming Authentication Request
```
14:31:08.028544 [RLC    ] [I] mutex 0x7fab7715b870 acquired from void srsran::rlc_am_lte::rlc_am_lte_rx::write_pdu(uint8_t*, uint32_t)
```

I'm not sure what actually triggers `get_buffer_state` here.

Call stack for the STACK thread:
```
libpthread.so.0!__lll_lock_wait(int * futex, int private) (lowlevellock.c:52)
libpthread.so.0!__GI___pthread_mutex_lock(pthread_mutex_t * mutex) (pthread_mutex_lock.c:80)
srsran::rlc_am_lte::rlc_am_lte_tx::write_sdu(std::unique_ptr<srsran::byte_buffer_t, std::default_delete<srsran::byte_buffer_t> >) (Unknown Source:0)
srsran::rlc_am_lte::write_sdu(std::unique_ptr<srsran::byte_buffer_t, std::default_delete<srsran::byte_buffer_t> >) (Unknown Source:0)
srsran::rlc::write_sdu(unsigned int, std::unique_ptr<srsran::byte_buffer_t, std::default_delete<srsran::byte_buffer_t> >) (Unknown Source:0)
srsran::pdcp_entity_lte::write_sdu(std::unique_ptr<srsran::byte_buffer_t, std::default_delete<srsran::byte_buffer_t> >, int) (Unknown Source:0)
non-virtual thunk to srsran::pdcp::write_sdu(unsigned int, std::unique_ptr<srsran::byte_buffer_t, std::default_delete<srsran::byte_buffer_t> >, int) (Unknown Source:0)
srsue::rrc::send_ul_dcch_msg(unsigned int, asn1::rrc::ul_dcch_msg_s const&) (Unknown Source:0)
srsue::rrc::send_ul_info_transfer(std::unique_ptr<srsran::byte_buffer_t, std::default_delete<srsran::byte_buffer_t> >) (Unknown Source:0)
srsue::rrc::write_sdu(std::unique_ptr<srsran::byte_buffer_t, std::default_delete<srsran::byte_buffer_t> >) (Unknown Source:0)
srsue::nas::send_authentication_response(unsigned char const*, unsigned long) (Unknown Source:0)
srsue::nas::parse_authentication_request(unsigned int, std::unique_ptr<srsran::byte_buffer_t, std::default_delete<srsran::byte_buffer_t> >, unsigned char) (Unknown Source:0)
srsue::nas::write_pdu(unsigned int, std::unique_ptr<srsran::byte_buffer_t, std::default_delete<srsran::byte_buffer_t> >) (Unknown Source:0)
srsue::rrc::parse_dl_dcch(unsigned int, std::unique_ptr<srsran::byte_buffer_t, std::default_delete<srsran::byte_buffer_t> >) (Unknown Source:0)
srsue::rrc::process_pdu(unsigned int, std::unique_ptr<srsran::byte_buffer_t, std::default_delete<srsran::byte_buffer_t> >) (Unknown Source:0)
non-virtual thunk to srsue::rrc::write_pdu(unsigned int, std::unique_ptr<srsran::byte_buffer_t, std::default_delete<srsran::byte_buffer_t> >) (Unknown Source:0)
srsran::pdcp_entity_lte::handle_srb_pdu(std::unique_ptr<srsran::byte_buffer_t, std::default_delete<srsran::byte_buffer_t> >) (Unknown Source:0)
srsran::pdcp_entity_lte::write_pdu(std::unique_ptr<srsran::byte_buffer_t, std::default_delete<srsran::byte_buffer_t> >) (Unknown Source:0)
srsran::pdcp::write_pdu(unsigned int, std::unique_ptr<srsran::byte_buffer_t, std::default_delete<srsran::byte_buffer_t> >) (Unknown Source:0)
srsran::rlc_am_lte::rlc_am_lte_rx::reassemble_rx_sdus() (Unknown Source:0)
srsran::rlc_am_lte::rlc_am_lte_rx::handle_data_pdu(unsigned char*, unsigned int, srsran::rlc_amd_pdu_header_t&) (Unknown Source:0)
srsran::rlc_am_lte::rlc_am_lte_rx::write_pdu(unsigned char*, unsigned int) (Unknown Source:0)
srsran::rlc_am_lte::write_pdu(unsigned char*, unsigned int) (Unknown Source:0)
srsran::rlc::write_pdu(unsigned int, unsigned char*, unsigned int) (Unknown Source:0)
srsue::demux::process_sch_pdu(srsran::sch_pdu*) (Unknown Source:0)
srsue::demux::process_pdu(unsigned char*, unsigned int, srsran::pdu_queue::channel_t) (Unknown Source:0)
srsran::pdu_queue::process_pdus() (Unknown Source:0)
srsran::task_details::smallbuffer_table_t<srsue::mac::process_pdus()::{lambda()#1}, void>::call(void*) const (Unknown Source:0)
non-virtual thunk to srsue::ue_stack_lte::run_thread() (Unknown Source:0)
srsran::thread::thread_function_entry(void*) (Unknown Source:0)
libpthread.so.0!start_thread(void * arg) (pthread_create.c:463)
libc.so.6!clone() (clone.S:95)
```

[ue.log](https://github.com/srsran/srsRAN/files/6946586/ue.log)
[PCAPs.zip](https://github.com/srsran/srsRAN/files/6946592/PCAPs.zip)

andrepuschmann, 2021-08-30:
Thanks @mrlnc for flagging this. We actually found that ourselves a few weeks ago. May I ask how you tirgger the issue? Is it always happening for you?

mrlnc, 2021-08-30:
this was happening on every attach. Huawei 4G small cell (operator-managed; no access to logs).

mrlnc, 2021-11-03:
Works great on 21.10, thanks!

closed 2021-11-03 by mrlnc

andrepuschmann, 2021-11-03:
> Works great on 21.10, thanks!

That's awesome. Thanks for confirming. And with 21.04 you can still reproduce the hang? Just trying to rule out that the setup has changed somehow and is now hiding something ...