srsran/srsRAN_4G #723: multiuser communication, srsRAN 21.04, one EPC and ENB connects two UEs

zsy9512, 2021-09-08:
## Issue Description ##
[Describe the issue in detail]
Our goal: one EPC and ENB connects two UEs at the same time

Hello, we were running srsRAN with release 21.04, and we got a problem about multi users. We tried to connect one EPC and ENB with two UEs, but we failed. For example, we connected UE1 with EPC and ENB, it worked well. However, when we connected UE2 with EPC and ENB, UE1 was disconnected in few seconds, which meant UE1 was disconnected and UE2 was connected. We could not connect the two UEs with EPC and ENB at the same time.

We thought there were some configurations that need to be set for multiuser communication, and tried some ways, but they did not work.

## Setup Details ##
[Specify details of the test setup. This would help us reproduce the problem reliably]
e.g. Network configuration, Operation System, Hardware, RF front-end, library and driver versions

**Our orders are as follows:(we did not change ue.conf. all orders are default. srsRAN 21.04)

first:
PC1 with B210:(we consider it as a base station) 
sudo srsepc
sudo srsenb

then:
PC2 with B210:(we consider it as an UE)
sudo srsue 
(PC2 is connected)

then:
PC3 with B210:(we consider it as an UE)
sudo srsue 
(PC2 is disconnected, PC3 is connected)

so the problem is that we can not make PC2 and PC3 connect PC1 at the same time.** 

## Expected Behavior ##
[What you expect to happen]
we want PC1 can connect PC2 and PC3 at the same time.

## Actual Behaviour ##
[What happens instead e.g. error message]
only one PC can connect to PC1

## Steps to reproduce the problem ##
[Tell us how to reproduce this issue e.g. RF setup, application config files]

## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]
PC2 logs:
sudo srsue 
Reading configuration file /home/shuangc/.config/srsran/ue.conf...
WARNING: cpu0 scaling governor is not set to performance mode. Realtime processing could be compromised. Consider setting it to performance mode before running the application.

Built in Release mode using commit 1c6dd8c4a on branch HEAD.

Opening 1 channels in RF device=default with args=default
[INFO] [UHD] linux; GNU C++ version 7.5.0; Boost_106501; UHD_3.14.0.HEAD-0-g6875d061
[INFO] [LOGGING] Fastpath logging disabled at runtime.
Opening USRP channels=1, args: type=b200,master_clock_rate=23.04e6
[INFO] [UHD RF] RF UHD Generic instance constructed
[INFO] [B200] Detected Device: B210
[INFO] [B200] Operating over USB 3.
[INFO] [B200] Initialize CODEC control...
[INFO] [B200] Initialize Radio control...
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Asking for clock rate 23.040000 MHz... 
[INFO] [B200] Actually got clock rate 23.040000 MHz.
Waiting PHY to initialize ... done!
Attaching UE...
Starting plot for worker_id=0
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
[... 138 lines cut ...]
07:38:56.691976 [S1AP   ] [I] Initial Context Setup Request -- S1-U TEID 0x1. IP 127.0.1.100 
07:38:56.691977 [S1AP   ] [I] Initial Context Setup Request -- S1-U TEID 0x1. IP 127.0.1.100 
07:38:56.691983 [S1AP   ] [I] Initial Context Setup Request -- QCI 7
07:38:56.739577 [S1AP   ] [I] Received S1AP msg. Size: 67
07:38:56.739585 [S1AP   ] [I] Received Initiating PDU
07:38:56.739585 [S1AP   ] [I] Ignoring UE capability Info Indication.
07:38:56.940222 [S1AP   ] [I] Received S1AP msg. Size: 38
07:38:56.94023

PC1 enb log:
==== eNodeB started ===
Type <t> to view trace
Setting frequency: DL=2680.0 Mhz, UL=2560.0 MHz for cc_idx=0 nof_prb=50
RACH:  tti=5621, cc=0, preamble=5, offset=43, temp_crnti=0x46
RACH:  tti=5621, cc=0, preamble=7, offset=1, temp_crnti=0x47
RACH:  tti=5621, cc=0, preamble=10, offset=7, temp_crnti=0x48
Disconnecting rnti=0x48.
Disconnecting rnti=0x46.
User 0x47 connected
RACH:  tti=3861, cc=0, preamble=21, offset=43, temp_crnti=0x49
RACH:  tti=3861, cc=0, preamble=23, offset=1, temp_crnti=0x4a
RACH:  tti=3861, cc=0, preamble=26, offset=7, temp_crnti=0x4b
Disconnecting rnti=0x4b.
Disconnecting rnti=0x49.
User 0x4a connected

umapriyam, 2022-02-07:
whats the max no of UE srs supports here?

frankist, 2022-04-29:
The LTE eNB can work in real time for more than 2 UEs. Currently, the only limit is either your computer or the constant `MAX_NOF_UES`.

pemo75, 2024-01-08:
> `MAX_NOF_UES`

hello frankist 
could you please tell me where can i find MAX_NOF_UES? can i change it to any number i want?