srsran/srsRAN_4G #726: srsEPC not loading correct sib.conf

avalori1, 2021-09-20:
## Issue Description ##
srsenb does not seem to correctly pick the correct sib configuration file described in the enb.conf, but always the default one

## Setup Details ##
LimeSDRmini and Ubuntu 20.4 lts on Lenovo

## Expected Behavior ##
srsenb to pick the sib configuration file mentioned into the enb configuration file

## Actual Behaviour ##
Always picks sib.conf

## Steps to reproduce the problem ##
create a copy of `sib.conf` (e.g. `sib_lime.conf`) and add a "error line" at the top (e.g. abc)
create a copy of `enb.conf` (e.g. `enb_lime.conf`)
open `enb_lime.conf` and set `sib_config = sib_lime.conf`
run `sudo -E srsenb .config/srsran/enb_lime.conf` THIS RUNS OK DESPITE ERROR IN `sib_lime.conf`
add error line to `.config/srsran/sib.conf` (e.g. abc)
run `sudo -E srsenb .config/srsran/enb_lime.conf` NOW GIVES AN ERROR SUGGESTING THAT READS `sib.conf` RATHER THAN `sib_lime.conf`

joaquim-srs, 2021-10-05:
It would be useful to know the exact location of your `sib.conf` and `sib_lime.conf` files.

However, I am assuming that they are not located in your terminal's working directory when you are running `srsenb`. The paths specified in `enb.conf` for the `sib_config`, `rr_config` and `rb_config` files are relative to the current working directory. You can try adding the absolute path to the files, such as, `/home/<user>/.config/srsran/enb_lime.conf`. Please let me know if this solves your issue.

avalori1, 2021-10-06:
Dear Joaquim,

thanks for your reply. After a bit more research, I seems to have realized what was happening. Things seems to be a bit more "subtle"...

My config folder contains
```
enodeb@enodeb-ThinkPad-L460:~$ ls .config/srsran/
drb.conf  enb_B200.conf  enb_lime.conf  epc.conf  mbms.conf  rr.conf  sib.conf  sib_lime.conf  ue.conf  user_db.csv
```

The file sib_lime.conf contains an error (placed there on purpose).

If, in the enb_lime.conf file, I put the (full) path as you reccommended:
`sib_config = /home/enodeb/.config/srsran/sib_lime.conf`
This fails (expected behaviour) with: 
Parse error at /home/enodeb/.config/srsran/sib_lime.conf:2 - syntax error

However, **if I make a mistake in the typing of the sib_conf file name in the enb_lime.conf** file (see example, one extra inexistent `.config` folder):
`sib_config = /home/enodeb/.config/srsran/.config/sib_lime.conf`
this starts happily picking the default `/home/enodeb/.config/srsran/sib.conf` file (NOT EXPECTED BEHAVIOUR)
I believe this is what was happening when I reported the issue...

Therefore, if I am right, I have to admit that can't call this a "BUG", but maybe a check if the file mentioned in the enb.conf file exists, before reverting to something else, may be useful to avoid people wasting time on silly mistakes (as I did).

Cheers,

Andrea