srsran/srsRAN_4G #727: srsUE can not attach to eNB

fancyshirt, 2021-09-21:
## Issue Description ##
[srsUE keeps searching eNB but can not attach to eNB]

## Setup Details ##
[Specify details of the test setup. This would help us reproduce the problem reliably]
Machine-1:
Ubuntu 18.04
Running srsepc amd srsenb
UHD 3.15 installed
Connects to a USRP N310 with UHD 3.15

Machine-2:
Ubuntu 18.04
Running srsue
UHD 3.15 installed
Connects to a USRP N310 with UHD 3.15

## Expected Behavior ##
After EPC and eNB are already running, UE can attach to eNB.

## Actual Behaviour ##
srsUE keeps searching for attaching.
root@machine2:~# srsue --rf.device_args=send_frame_size=1476,recv_frame_size=1476 --rat.eutra.dl_freq=2560e6 --rat.eu                                                                                                                        tra.ul_freq=2680e6
Reading configuration file /root/.config/srsran/ue.conf...

Built in Release mode using commit 89f16eed2 on branch master.

Opening 1 channels in RF device=UHD with args=send_frame_size=1476,recv_frame_size=1476
[INFO] [UHD] linux; GNU C++ version 7.5.0; Boost_106501; UHD_3.15.0.0-release
[INFO] [LOGGING] Fastpath logging disabled at runtime.
Opening USRP channels=1, args: send_frame_size=1476,recv_frame_size=1476,type=n3xx,master_clock_rate=122.88e6
[INFO] [UHD RF] RF UHD Generic instance constructed
[INFO] [MPMD] Initializing 1 device(s) in parallel with args: mgmt_addr=192.168.11.2,type=n3xx,product=n310,seria                                                                                                                        l=3205558,claimed=False,addr=192.168.11.2,send_frame_size=1476,recv_frame_size=1476,master_clock_rate=122.88e6
[INFO] [MPM.PeriphManager] init() called with device args `send_frame_size=1476,master_clock_rate=122.88e6,time_s                                                                                                                        ource=internal,recv_frame_size=1476,product=n310,mgmt_addr=192.168.11.2,clock_source=internal'.
[INFO] [0/Replay_0] Initializing block control (NOC ID: 0x4E91A00000000004)
[INFO] [0/Radio_0] Initializing block control (NOC ID: 0x12AD100000011312)
[INFO] [0/Radio_1] Initializing block control (NOC ID: 0x12AD100000011312)
[INFO] [0/DDC_0] Initializing block control (NOC ID: 0xDDC0000000000000)
[INFO] [0/DDC_1] Initializing block control (NOC ID: 0xDDC0000000000000)
[INFO] [0/DUC_0] Initializing block control (NOC ID: 0xD0C0000000000002)
[INFO] [0/DUC_1] Initializing block control (NOC ID: 0xD0C0000000000002)
[INFO] [0/FIFO_0] Initializing block control (NOC ID: 0xF1F0000000000000)
[INFO] [0/FIFO_1] Initializing block control (NOC ID: 0xF1F0000000000000)
[INFO] [0/FIFO_2] Initializing block control (NOC ID: 0xF1F0000000000000)
[INFO] [0/FIFO_3] Initializing block control (NOC ID: 0xF1F0000000000000)
Waiting PHY to initialize ... done!
Attaching UE...
.................................................................
....................................................
..................
.........................................................................................................
...........................................
........................
.................................................................................................................                                                                                                                        ..............................................................................................................
................................................................................................
..................................................................
...............
.......................................................
.................................................................................................................                                                                                                                        .........
...........................................
.................................................................................................................                                                                                                                        ............................................................................................................
.............................................
.
.................................................................................................................                                                                                                                        ..........................
.................................................................................................................                                                                                                                        ....
.................................................................................................................                                                                                                                        .................................................................................................................                                                                                                                        ............
.................................................................................................................                                                                                                                        .................................................................................................................                                                                                                                        .................................................................................................................                                                                                                                        ......................................................................Stopping ..
.....---  exiting  ---
root@machine2:~#

## Steps to reproduce the problem ##
[Tell us how to reproduce this issue e.g. RF setup, application config files]
=== epc.conf
[mme]
mme_code = 0x1a
mme_group = 0x0001
tac = 0x0007
mcc = 001
mnc = 01
mme_bind_addr = 127.0.1.100
apn = srsapn
dns_addr = 8.8.8.8
encryption_algo = EEA0
integrity_algo = EIA1
paging_timer = 2

[hss]
db_file = user_db.csv

[spgw]
gtpu_bind_addr   = 127.0.1.100
sgi_if_addr      = 172.16.0.1
sgi_if_name      = srs_spgw_sgi
max_paging_queue = 100

[pcap]
enable   = false
filename = /tmp/epc.pcap

[log]
all_level = info
all_hex_limit = 32
filename = /tmp/epc.log

=== enb.conf
[enb]
enb_id = 0x19B
mcc = 001
mnc = 01
mme_addr = 127.0.1.100
gtp_bind_addr = 127.0.1.1
s1c_bind_addr = 127.0.1.1
n_prb = 50

[enb_files]
sib_config = sib.conf
rr_config  = rr.conf
drb_config = drb.conf

[rf]
dl_earfcn = 3350
tx_gain = 80
rx_gain = 40
device_name = UHD
device_args = type=n3xx
srate = 15.36e6

[pcap]
enable = false
filename = /tmp/enb.pcap
s1ap_enable = false
s1ap_filename = /tmp/enb_s1ap.pcap

mac_net_enable = false
bind_ip = 0.0.0.0
bind_port = 5687
client_ip = 127.0.0.1
client_port = 5847

[log]
all_level = warning
all_hex_limit = 32
filename = /tmp/enb.log
file_max_size = -1

[gui]
enable = false

=== ue.conf
[rf]
freq_offset = 0
tx_gain = 80
device_name = UHD
device_args = type=n3xx
#srate = 15.36e6
rx_gain = 40

[rat.eutra]
dl_earfcn = 3350

[pcap]
enable = none
mac_filename = /tmp/ue_mac.pcap
mac_nr_filename = /tmp/ue_mac_nr.pcap
nas_filename = /tmp/ue_nas.pcap

[log]
all_level = warning
phy_lib_level = none
all_hex_limit = 32
filename = /tmp/ue.log
file_max_size = -1

[usim]
mode = soft
algo = xor
#opc  = 63BFA50EE6523365FF14C1F45F88737D
k    = 00112233445566778899aabbccddeeff
imsi = 001010123456789
imei = 353490069873319

[gw]
#netns =
ip_devname = tun_srsue
ip_netmask = 255.255.255.0

[gui]
enable = false

I used the following command:
1. (on machine 1)
    ~# sudo srsepc
2. (on machine 1)
    ~# sudo srsenb --rf.device_args=send_frame_size=1476,recv_frame_size=1476
3. (on machine 2)
    ~# sudo srsue --rf.device_args=send_frame_size=1476,recv_frame_size=1476 --rat.eutra.dl_freq=2560e6 --rat.eutra.ul_freq=2680e6

## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]
[enb.log](https://github.com/srsran/srsRAN/files/7205964/enb.log)
[epc.log](https://github.com/srsran/srsRAN/files/7205965/epc.log)
[ue.log](https://github.com/srsran/srsRAN/files/7205966/ue.log)

andrepuschmann, 2021-09-22:
The N310, or the UHD drivers, are a bit picky and so there are still issues with it. We're trying to address them and also flagged them with Ettus, see (though this is with UHD 4.1):
* https://github.com/EttusResearch/uhd/issues/449
* https://github.com/EttusResearch/uhd/issues/486

This is a bit better on the eNB with UHD 3.15 and the N310, but the UE, as far I as I can tell also there requires special parameters to avoid retuning. Check the NSA appnote [here](https://docs.srsran.com/en/latest/app_notes/source/5g_nsa/source/index.html#srsue) and try the `srate=` param.

fancyshirt, 2021-09-22:
Thank you @andrepuschmann .
I will go through the NSA appnote you gave and see if any of those solve the issue.

fancyshirt, 2021-09-22:
@andrepuschmann I'm connecting USRP N310 to my host over a 1GigE cable. Is it required to connect them over 10GigE or 1GigE should able to work as well?

andrepuschmann, 2021-09-24:
> @andrepuschmann I'm connecting USRP N310 to my host over a 1GigE cable. Is it required to connect them over 10GigE or 1GigE should able to work as well?

Depends on your particular cell configuration. Check the bandwidth requirments, but for a typical 10Mhz cell in SISO the 1GigE should be enough.

bondadakumarsai, 2021-11-11:
Ran the srsRAN (release_20_10) with UHD 4.1.0.4 version installed. Initially, the srsENB & srsUE were able to recognize the N310s but srsUE. couldn't attach to srsENB. The steps I followed to make it work 
1. Set CPU to Performance mode
2. Change LTE sample rates to ON in the CMakeLists.txt file
3. What I found in the srsUE code (main.cc) was use_standard_lte_rates was set to false permanently, cannot be configured in ue.conf file. Changed the state to true.
4. Then built the srsRAN code and ran srsEPC, srsENB & srsUE. I was to see UE being attached to Cell and was able to ping traffic.

andrepuschmann, 2021-11-12:
Thanks for the feedback. 

> What I found in the srsUE code (main.cc) was use_standard_lte_rates was set to false permanently, cannot be configured in ue.conf file. Changed the state to true. Added lte_sample_rates=true in enb.conf file

That is only the default. You can overwrite it with `--expert.lte_sample_rates=true`

Anyway, closing the issue as I think it's solved. Also no feedback from the original author.

closed 2021-11-12 by andrepuschmann

MD-star-cloud33, 2023-03-31:
[ue.log.zip](https://github.com/srsran/srsRAN_4G/files/11120218/ue.log.zip)
[enb.log.zip](https://github.com/srsran/srsRAN_4G/files/11120225/enb.log.zip)
[aiab-30-2.zip](https://github.com/srsran/srsRAN_4G/files/11120228/aiab-30-2.zip)

MD-star-cloud33, 2023-03-31:
[image]
[ue.log.zip](https://github.com/srsran/srsRAN_4G/files/11120448/ue.log.zip)
[enb.log.zip](https://github.com/srsran/srsRAN_4G/files/11120450/enb.log.zip)

MD-star-cloud33, 2023-04-03:
Getting error -> 
Cause #16 – MSC temporarily not reachable

      This EMM cause is sent to the UE if it requests a combined EPS attach or tracking area updating in a PLMN where the MSC is temporarily not reachable via the EPS part of the network.

Cause #17 – Network failure

      This EMM cause is sent to the UE if the MME cannot service an UE generated request because of PLMN failures.

Cause #66 – Requested APN not supported in current RAT and PLMN combination

      This ESM cause is used by the network to indicate that the procedure requested by the UE was rejected as the requested APN is not supported in the current RAT and PLMN.