srsran/srsRAN_4G #728: Srsenb invalid argument on BladeRF xA4 + Biastee amps

yesimxev, 2021-09-23:
## Issue Description ##
srsenb fails at the very beginning, after biastee amplifiers are enabled 

## Setup Details ##
Latest Kali, libbladRF, srsRAN built from repo

## Expected Behavior ##
Work same as without 

## Actual Behaviour ##
Fails with 
```
---  Software Radio Systems LTE eNodeB  ---

Reading configuration file configs/enb.conf...
WARNING: cpu0 scaling governor is not set to performance mode. Realtime processing could be compromised. Consider setting it to performance mode before running the application.

Built in Release mode using commit 2645620 on branch main.

Opening 1 channels in RF device=bladeRF with args=default
Opening bladeRF...
/root/mnrt/Modmobmap/thirdparty/srsRAN/lib/src/phy/rf/rf_blade_imp.c.210: Failed to get TX gain range: Invalid operation or parameter
Error initializing radio.
```

Use bladRF.conf in current working dir with
```
fpga hostedxA4-latest.rbf
biastee_rx true
biastee_tx true
srsenb srsenb.conf
```

fllay, 2021-09-29:
Please take a look at
https://nuand.com/forums/viewtopic.php?f=6&t=9863
It uses its own configuration file. It work just fine. But they still did not reply me how to setting the external clock. It takes them very long time to answer.

yesimxev, 2021-10-02:
It's not about how to enable or keep it, it's not compatible yet as looks like biastee needs a different arg

fllay, 2021-10-03:
For may setting, I load fpga bit file from spi flash and set bias tee in the config file. I did not have any problem.

yesimxev, 2021-10-03:
You mean to flash fpga for autoload, using `bladeRF-cli -L`?

fllay, 2021-10-04:
> You mean to flash fpga for autoload, using `bladeRF-cli -L`?

Yes. That is what I have done. And bladeRF.conf will have only bias tee setting.

yesimxev, 2021-10-04:
Unfortunately that crashes my blade, using latest Kali, libbladerfr, srsRAN. Flashed fpga as you said. Even bladeRF-cli crashes upon opening device. The whole process is very simple, but I might still miss something, can you confirm your setup and/or configs please?

My error is:
[image]

fllay, 2021-10-04:
In your case, I think your Linux could not find the board. I would suggest you compile bladeRF driver from source and/or update bladerf firmware. I have tried a bladeRF board with old formware and bladeRF-cli could not probe the board. However, my error is not the same as yours. Also, I could not flash the latest firmware using my Ubuntu machine. I have to use my mac to flash it. If you install FPGA bin file correctly and it is loaded at boot, you will see all LEDs is on.

yesimxev, 2021-10-04:
It's all latest including blade source. It's finding it, all crashes when .conf file is read. Hence it doesn't find after crash. Without amps, works like a charm

fllay, 2021-10-04:
One last thing I can think of. you should try to use Ubuntu instead of Kali. I am using Ubuntu 20.04 LTS.  I have used Ubuntu 18.04 before and it is ok.  You can also using `dmesg -wH` to see when your bladeRF is kicked out from the system. 
You may ask about this issue at the bladeRF forum but it takes them months to reply. 

[IMG_1944.HEIC.pdf](https://github.com/srsran/srsRAN/files/7279655/IMG_1944.HEIC.pdf)

yesimxev, 2021-10-04:
That would really explain everything if solves it. Trying soon 🙏

Also your external power source seems more stable than usb3.0. I have ordered one a week ago, still coming

yesimxev, 2021-10-05:
Still fail on Ubuntu, dmesg shows nothing. Until I get the power supply, I'll try on another bladeRF too
```
[ 2851.981366] usb 4-1: reset SuperSpeed Gen 1 USB device number 2 using xhci_hcd
[ 2852.009445] usb 4-1: LPM exit latency is zeroed, disabling LPM.
[ 2853.728295] usb 4-1: USB disconnect, device number 2
```

yesimxev, 2021-10-05:
@fllay See the clock settings: https://github.com/Nuand/bladeRF/commit/66958d63b1d4012309b4d6dffd388325ddf88291

Credits to @rghilduta for the answer! 🙏

fllay, 2021-10-05:
> @fllay See the clock settings: [Nuand/bladeRF@66958d6](https://github.com/Nuand/bladeRF/commit/66958d63b1d4012309b4d6dffd388325ddf88291)
> 
> Credits to @rghilduta for the answer! 🙏

Thank you very much. I will give it a try.