srsran/srsRAN_4G #748: [Question] A lot of RARs

matan1008, 2021-11-02:
## Issue Description ##
Hey!
I noticed that after running the EPC and ENB for a while, the phone tries to "reconnect" to the network (the signal bar drops to the lowest level and after some seconds comes back). I took a pcap capture and it seems that in the same time there are a lot of RARs that are sent to the phone.
I don't really know if it is the expected behaviour or not, so I would really like to hear your expert opinion :smiley: 

## Setup Details ##
- Ubuntu 20.04
- USRP B210 - with power adaptor, GPSDO and USB 3
- UHD - linux; GNU C++ version 9.2.1 20200304; Boost_107100; UHD_3.15.0.0-2build5
- CPU - Intel(R) Core(TM) i7-7700 CPU @ 3.60GHz
- iPhone 7

## Expected Behavior ##
Continuous connection, with RAR once in a while. 

## Actual Behaviour ##
Sometime there is a RAR each 0.09 seconds and the connection drops.

## Steps to reproduce the problem ##
[current_enb.txt](https://github.com/srsran/srsRAN/files/7458433/current_enb.txt)
[current_enb_rbs.txt](https://github.com/srsran/srsRAN/files/7458434/current_enb_rbs.txt)
[current_enb_rr.txt](https://github.com/srsran/srsRAN/files/7458435/current_enb_rr.txt)
[current_enb_sibs.txt](https://github.com/srsran/srsRAN/files/7458436/current_enb_sibs.txt)
[current_epc.txt](https://github.com/srsran/srsRAN/files/7458437/current_epc.txt)

## Additional Information ##
Github doesn't allow pcapng files so just rename the file extension:
[multiple_rar_reduced_rename_to_pcapng.txt](https://github.com/srsran/srsRAN/files/7458458/multiple_rar_reduced_rename_to_pcapng.txt)
(The original file was too big)

I also have a huge debug log which github doesn't allow uploading, I it is not the expected behaviour I will try to upload in parts or something.

Thanks a lot!

andrepuschmann, 2021-11-19:
Hey, can you please share logs (in info) and also the stdout of the eNB (press t+Enter after starting).

matan1008, 2021-11-21:
Thanks for the quick response!
And the eNB logs are: [enb.log](https://github.com/srsran/srsRAN/files/7575846/enb.log)
The stdout is: [term.log](https://github.com/srsran/srsRAN/files/7575845/term.log)
Note that both logs start with a minute where everything was fine - in my opinion the problem starts around 2021-11-21T06:34:46.924246 in the logs (that's when I saw it in the pcap).

andrepuschmann, 2021-11-22:
Hey, I see a ton of RF error (lates, etc) in your logs. With those the phone loses sync an PRACHs again. So this is what you're seeing. You need to reduce the number of PRBs perhaps or tune your system to sustain the rates. It's not an srsRAN issue.

closed 2021-11-22 by andrepuschmann

matan1008, 2021-11-22:
Thanks a lot for the quick response!
Since I am new the USRP and you probably have way more experience than me, do you have any general recommendations for how to tune the system?

matan1008, 2022-03-03:
For future users that might face the same issue: Moving to a PC with the spec in the [docs](https://docs.srsran.com/en/latest/app_notes/source/hw_packs/source/index.html#package-3) solved my issue.

cristina-posada, 2022-06-29:
Hello!! How can I reduce the number of PRBs? Thanks!!