srsran/srsRAN_4G #751: plot unsuccessful

moon657, 2021-11-05:
I have installed srsRAN21.10 and srsGUI, and make test is successful.
when I run srsenb,
Trying to start a plot but plots are disabled (ENABLE_GUI constant in sf_worker.cc)

andrepuschmann, 2021-11-05:
well, it seems when building srsran Cmake couldn't pick up srsgui headers/libs. Post cmake output etc.

moon657, 2021-11-06:
> well, it seems when building srsran Cmake couldn't pick up srsgui headers/libs. Post cmake output etc.

-- CMAKE_SYSTEM: Linux-5.4.0-42-generic
-- CMAKE_SYSTEM_PROCESSOR: x86_64
-- CMAKE_CXX_COMPILER: /usr/bin/c++
-- Build type not specified: defaulting to Release.
-- FFTW3F LIBRARIES: /usr/lib/x86_64-linux-gnu/libfftw3f.so
-- FFTW3F STATIC LIBRARIES: /usr/lib/x86_64-linux-gnu/libfftw3f.a
-- FFTW3F INCLUDE DIRS: /usr/include
-- FFT_LIBRARIES: /usr/lib/x86_64-linux-gnu/libfftw3f.so
-- Checking for module 'polarssl'
--   No package 'polarssl' found
-- POLARSSL LIBRARIES: POLARSSL_LIBRARIES-NOTFOUND
-- POLARSSL STATIC LIBRARIES: POLARSSL_STATIC_LIBRARIES-NOTFOUND
-- POLARSSL INCLUDE DIRS: POLARSSL_INCLUDE_DIRS-NOTFOUND
-- Could NOT find Polarssl (missing: POLARSSL_LIBRARIES POLARSSL_INCLUDE_DIRS) 
-- Checking for module 'mbedtls'
--   No package 'mbedtls' found
-- MBEDTLS LIBRARIES: /usr/lib/x86_64-linux-gnu/libmbedcrypto.so
-- MBEDTLS STATIC LIBRARIES: /usr/lib/x86_64-linux-gnu/libmbedcrypto.a
-- MBEDTLS INCLUDE DIRS: /usr/include
-- Checking for module 'libpcsclite'
--   Found libpcsclite, version 1.8.23
-- Found PCSCLite: /usr/lib/x86_64-linux-gnu/libpcsclite.so  
-- PCSC LIBRARIES: /usr/lib/x86_64-linux-gnu/libpcsclite.so
-- PCSC INCLUDE DIRS: /usr/include/PCSC
-- Building with PCSC support.
-- UHD LIBRARIES /usr/lib/libuhd.so
-- UHD INCLUDE DIRS /usr/include
-- SKIQ LIBRARIES SKIQ_LIBRARY-NOTFOUND/usr/lib/x86_64-linux-gnu/libglib-2.0.a/usr/lib/x86_64-linux-gnu/libusb-1.0.a
-- SKIQ INCLUDE DIRS SKIQ_INCLUDE_DIRS-NOTFOUND
-- Could NOT find SKIQ (missing: SKIQ_INCLUDE_DIRS) 
-- Checking for module 'libbladeRF'
--   No package 'libbladeRF' found
-- libbladeRF not found.
-- FINDING SOAPY.
-- Checking for module 'SoapySDR'
--   No package 'SoapySDR' found
-- libSOAPYSDR not found.
-- FINDING ZEROMQ.
-- Checking for module 'ZeroMQ'
--   No package 'ZeroMQ' found
-- libZEROMQ not found.
-- Boost version: 1.65.1
-- Found the following Boost libraries:
--   program_options
-- SRSGUI LIBRARIES SRSGUI_LIBRARIES-NOTFOUND
-- SRSGUI INCLUDE DIRS SRSGUI_INCLUDE_DIRS-NOTFOUND
-- Could NOT find SRSGUI (missing: SRSGUI_LIBRARIES SRSGUI_INCLUDE_DIRS) 
-- Found libdw: /usr/lib/x86_64-linux-gnu/libdw.so  
-- Could NOT find libbfd (missing: LIBBFD_LIBRARY LIBBFD_INCLUDE_DIR) 
-- Could NOT find libdwarf (missing: LIBDWARF_LIBRARY LIBDWARF_INCLUDE_DIR) 
-- Found Backward: /home/ubuntu/srsRAN/cmake/modules  
-- Building with backward-cpp support
-- SSE4.1 is enabled - target CPU must support it
-- AVX is enabled - target CPU must support it
-- AVX2 is enabled - target CPU must support it
-- FMA is enabled - target CPU must support it
-- SSE4.1 is enabled - target CPU must support it
-- AVX is enabled - target CPU must support it
-- AVX2 is enabled - target CPU must support it
-- FMA is enabled - target CPU must support it
-- CMAKE_C_FLAGS is  -Wno-unused-but-set-variable -march=native -Werror=incompatible-pointer-types -Wall -Wno-comment -Wno-write-strings -Wno-unused-result -Wformat -Wmissing-field-initializers -Wtype-limits -std=c99 -fno-strict-aliasing -D_GNU_SOURCE -O3 -fno-trapping-math -fno-math-errno -DBUILD_TYPE_RELEASE -mfpmath=sse -mavx2 -DLV_HAVE_AVX2 -DLV_HAVE_AVX -DLV_HAVE_SSE -mfma -DLV_HAVE_FMA -Ofast -funroll-loops -fvisibility=hidden -Werror --param large-function-growth=1600
-- CMAKE_CXX_FLAGS is  -Wall -Wno-comment -Wno-reorder -Wno-unused-variable -Wtype-limits -std=c++11 -fno-strict-aliasing -Wno-unused-but-set-variable -march=native -mfpmath=sse -mavx2 -DLV_HAVE_AVX2 -DLV_HAVE_AVX -DLV_HAVE_SSE -fvisibility=hidden -O3 -fno-trapping-math -fno-math-errno -DBUILD_TYPE_RELEASE -Werror
-- Using install prefix: /usr/local
-- Building for version: 21.10.0
--    examples will be installed.
-- Checking for module 'sctp'
--   No package 'sctp' found
-- SCTP LIBRARIES: /usr/lib/x86_64-linux-gnu/libsctp.so
-- SCTP INCLUDE DIRS: /usr/include
-- No post-build command defined
-- No post-build command defined
-- Building with srsUE
-- No post-build command defined
-- No post-build command defined
-- No post-build-UE command defined
-- Building with srsENB
-- Found LibConfig++: /usr/lib/x86_64-linux-gnu/libconfig++.so
-- static LibConfig++ path: /usr/lib/x86_64-linux-gnu/libconfig++.a
-- Found LibConfig: /usr/lib/x86_64-linux-gnu/libconfig.so
-- static LibConfig path: /usr/lib/x86_64-linux-gnu/libconfig.a
-- Checking for module 'sctp'
--   No package 'sctp' found
-- SCTP LIBRARIES: /usr/lib/x86_64-linux-gnu/libsctp.so
-- SCTP INCLUDE DIRS: /usr/include
-- No post-build-ENB command defined
-- Building with srsEPC
-- Found LibConfig++: /usr/lib/x86_64-linux-gnu/libconfig++.so
-- static LibConfig++ path: /usr/lib/x86_64-linux-gnu/libconfig++.a
-- Found LibConfig: /usr/lib/x86_64-linux-gnu/libconfig.so
-- static LibConfig path: /usr/lib/x86_64-linux-gnu/libconfig.a
-- Checking for module 'sctp'
--   No package 'sctp' found
-- SCTP LIBRARIES: /usr/lib/x86_64-linux-gnu/libsctp.so
-- SCTP INCLUDE DIRS: /usr/include
-- No post-build-EPC command defined
-- Configuring done
-- Generating done
-- Build files have been written to: /home/ubuntu/srsRAN/build
3Q

closed 2021-11-06 by moon657