srsran/srsRAN_4G #752: COTS UE has no Internet with NSA gNB

yc541, 2021-11-05:
## Issue Description ##
Hi!

Really glad to see the NSA gNB release and big thanks to the SRS team!
I have been testing the eNB/gNB + Open5GS core with COTS UEs (with sysmocom SIMs). I first tested a Huawei P40 Pro which worked straightaway, attached to the gNB, saw the little 5G logo at the signal bar, no problem with Internet. Then I put the same SIM in a Oneplus Nord (UK version), the UE attached with gNB (I can see the 5G logo) but it only had Internet for a short time (from a split of a second to maybe 10 seconds) after attach. Then the UE lost Internet and I can't ping its IP from the core. If I switch the UE to LTE mode, it attaches to the eNB, has Internet and I can ping it from core. I also switched the core to srsepc, the UE had the same problem.

## Setup Details ##
I am using the latest master branch, also latest Open5GS v2.3.6. SRS eNB/gNB are on a physical machine with ubuntu 20.04 LTS, Open5GS is on a VM with network bridged (in the same subnet). Oneplus Nord was just bought from Amazon. I tried Android 10 and 11 but both had the same problem. The apn is set to internet for Open5GS. The SDR is a USRP X310.

## Expected Behavior ##
UE remains connected to the Internet

## Actual Behaviour ##
UE lost Internet connection almost immediately after attach
## Steps to reproduce the problem ##
My enb.conf, rr.conf are attached. Toggling the air plane mode can reproduce the problem.

## Additional Information ##
The enb logs for Huawei P40 and Oneplus Nord are attached as well. When looking at the enb(Nord).log, the UE gets rnti=0x46 (eNB) and rnti=0x4601 (gNB) after the first attach and there is some user traffic over the gNB. Shortly after the UE sends another prach and gets rnti=0x47, then it sends rrcConnectionReestablishmentRequest and complains about reconfigurationFailure. This triggers the rrcConnectionReconfiguration for rnti=0x47 (around line 86278 in enb log), compare with the rrcConnectionReconfiguration for rnti=0x46 (around line 47729) the NR cell is missing in the message for rnti=0x47. 

I just wonder if this is a configuration issue? Are there anything I can try to diagnose the problem? Any help would be appreciated!

Many thanks
Yi

[Nord.zip](https://github.com/srsran/srsRAN/files/7487660/Nord.zip)

andrepuschmann, 2021-11-07:
Hey, thanks for giving this a try. I am glad it the P40 works for you. Do you mind providing the exact model name and baseband version for this phone? We'd like to maintain a list of working basebands and don't have the P40 in it yet. Regarding the OnePlus: actually, same question. Please provide the exact model and baseband version. Note that the OnePlus Nord 2 actually has a Mediatek baseband and not the QC that the first version uses. Quite possible that this Mediatek doesn't like our RRC configuration.

Anyway, provide the models and please collect logs in info mode and provide the PCAP files as well. Thanks

yc541, 2021-11-08:
Many thanks for your reply!

I attached the pcap, logs and screenshots of UE model/BB.
When the P40 attached (I can see the 5G logo) I didn't see the nr entries
in the enb trace (it only had lte entries), is this normal? I did a
speedtest (prb=25), with 5G enabled it did about 22Mbps, with 4G it did
about 16Mbps so there is difference.

I do have the first version Nord with a Qualcomm chipset. Could you please
let me know the BB version which tested successfully? I will try flash that
and test again. Many thanks!

Regards
Yi

andrepuschmann, 2021-11-08:
> Many thanks for your reply! I attached the pcap, logs and screenshots of UE model/BB. When the P40 attached (I can see the 5G logo) I didn't see the nr entries in the enb trace (it only had lte entries), is this normal?

Ah, forget the 5G logo. It's displayed when a specific field in the SIB1 is present. That doesn't tell you anything. If the NR entries are not shown in the eNB, then it's not enabled yet.

> I did a speedtest (prb=25), with 5G enabled it did about 22Mbps, with 4G it did about 16Mbps so there is difference. I do have the first version Nord with a Qualcomm chipset. Could you please let me [...]

Use the eNB config as it is, don't change to 25 PRB, this will not work. It needs to be 10 MHz for both cells.

We use the AC2003 version of the OnePlus Nord 5G.

andrepuschmann, 2021-11-08:
Also make sure to not change the EUTRA/NR bands  if they are not supported by your phone.  Stick with the provided configs and try again. Send the same logs/PCAPs again

yc541, 2021-11-08:
The Nord (I have the AC2003 too) actually worked after tweaking the bands (and prb=50)! When I have the default eNB on B20, gNB on N3, the UE can't see the network. When both nodes are on B7/N7 (different frequencies), the UE attaches but gets rrcreconfiguration failure immediately (no Internet after). 

When eNB is on B3 and gNB is on N7, the UE attaches and has Internet, there are plenty of rrc reconfiguration and the UE gets new rnti frequently (at higher mcs), mcs=10 seems to give a steady connection.
[Nord(mcs).zip](https://github.com/srsran/srsRAN/files/7497255/Nord.mcs.zip)

andrepuschmann, 2021-11-08:
> The Nord (I have the AC2003 too) actually worked after tweaking the bands (and prb=50)! When I have the default eNB on B20, gNB on N3, the UE can't see the network. 

Mhh, could be that the phone doesn't like the PLMN or refuses to attach.  We use a test SIM card and with that the phone sees the network.

> When both nodes are on B7/N7 (different frequencies), the UE attaches but gets rrcreconfiguration failure immediately (no Internet after).

This wont work because b7/n7 is not a supported band combination for CA or NSA

yc541, 2021-11-08:
One thing I forgot when testing B20 is that the daughterboard on my USRP supports 1.2 to 6 GHz, this explains why the UE doesn't see the B20 eNB (though eNB runs without getting complained by UHD). Anyway I'm glad my Nord works, I will test other UEs once I get my hands on them. Many thanks for the help!

closed 2021-11-09 by yc541

jonaogaard, 2022-05-04:
Hi Yi,
Did you manage to get the NSA setup to work on the P40? I have tried multiple band combinations, but the UE switches back to LTE after a few seconds and no data ia sent over the NR link