srsran/srsRAN_4G #755: 5G NSA with COTSUE Using bladeRF xA9/xA4 throws error when running srsenb

nomidaepapi, 2021-11-08:
## Issue Description ##
I was testing out the 5G NSA using the bladeRF xA9/xA4 over a COTSUE (Oppo A73 with sysmocom sim cards) but the srsenb threw errors when it was run. All the steps for 5G NSA were followed as per the documentation. The only change to the config were the arguments replacing the USRP to BladeRF on both xA9/xA4. The srsEPC runs fine, the srsENB loads up but after it does, it gets stuck in a loop spamming this:
   

## **Error Message:**##
Soapy has found device #0: backend=libusb, device=0x02:0x06, driver=bladerf, instance=0, label=BladeRF #0 [4cd7f1ff..694c0e28], serial=4cd7f1fffd1c4d88b3ead6ef694c0e28,                                                                                                                                                              Selecting Soapy device: 0                                                                                                                                          [INFO] bladerf_open_with_devinfo()                                                                                                                                 [INFO] bladerf_get_serial() = 4cd7f1fffd1c4d88b3ead6ef694c0e28                                                                                                     [INFO] setSampleRate(Rx, 0, 4.000000 MHz), actual = 4.000000 MHz                                                                                                   [INFO] setSampleRate(Tx, 0, 4.000000 MHz), actual = 4.000000 MHz                                                                                                   Setting up Rx stream with 2 channel(s)                                                                                                                             Setting up Tx stream with 2 channel(s)                                                                                                                             [INFO] setSampleRate(Rx, 0, 1.920000 MHz), actual = 1.920000 MHz                                                                                                   [INFO] setSampleRate(Rx, 1, 1.920000 MHz), actual = 1.920000 MHz                                                                                                   [INFO] setSampleRate(Tx, 0, 1.920000 MHz), actual = 1.920000 MHz                                                                                                   [INFO] setSampleRate(Tx, 1, 1.920000 MHz), actual = 1.920000 MHz                                                                                                   Available device sensors:                                                                                                                                           - RFIC_TEMP                                                                                                                                                       Available sensors for Rx channel 0:                                                                                                                                 - PRE_RSSI                                                                                                                                                         - SYM_RSSI                                                                                                                                                        Available sensors for Rx channel 1:                                                                                                                                 - PRE_RSSI                                                                                                                                                         - SYM_RSSI                                                                                                                                                        State of gain elements for Rx channel 0 (AGC supported):                                                                                                            - full: 71.00 dB                                                                                                                                                  State of gain elements for Tx channel 0 (AGC not supported):                                                                                                        - dsa: -10.00 dB                                                                                                                                                  Rx antenna set to RX                                                                                                                                               Tx antenna set to TX                                                                                                                                               RF device 'soapy' successfully opened                                                                                                                                                                                                                                                                                                 ==== eNodeB started ===                                                                                                                                            Type <t> to view trace                                                                                                                                             [INFO] setSampleRate(Rx, 0, 5.760000 MHz), actual = 5.760000 MHz                                                                                                   [INFO] setSampleRate(Rx, 1, 5.760000 MHz), actual = 5.760000 MHz                                                                                                   [INFO] setSampleRate(Tx, 0, 5.760000 MHz), actual = 5.760000 MHz                                                                                                   [INFO] setSampleRate(Tx, 1, 5.760000 MHz), actual = 5.760000 MHz                                                                                                   Setting frequency: DL=806.0 Mhz, UL=847.0 MHz for cc_idx=0 nof_prb=25                                                                                              Setting frequency: DL=1842.5 Mhz, UL=1747.5 MHz for cc_idx=1 nof_prb=25                                                                                            000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000TTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTT00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000TTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTT [... cut ...]
## Expected Behavior ##
BladeRF xA9/xA4 should connect as well properly the same way USRP connects. 

## Actual Behaviour ##
The highlight of the behavior (error) is:
[ERROR @ host/libraries/libbladeRF/src/backend/usb/libusb.c:1090] Transfer timed out for buffer 0x555f1687ddf0    
getting spammed

## Steps to reproduce the problem ##
Using the bladeRF xA9/xA4 is sufficient to reproduce the problem. All the config files provided in the documentation were respectfully followed.

Thank you for your response!

andrepuschmann, 2021-11-08:
The bladeRF isn't supported with NSA I am afraid. Check the NSA appnote to see supported HW.

closed 2021-11-08 by andrepuschmann

jonaogaard, 2021-11-17:
Andre,
What is required to make the bladeRF compatible with NSA? I'm implementing a NSA network as a part of my Master's project and the only available SDR is bladeRF. Can you estimate how much implementation time that is required to run NSA on bladeRF?

fllay, 2021-12-02:
As far as I know, NSA needs both operating 4G and 5G basestations. I think that why srsenb needs two independently tune radios HW, one for 4G one for 5G. What I hope is that can we just plug two bladeRFs into two computers and one runs 4G and another one runs 5G and both are connected to the same EPC. I don't know that is possible.

andrepuschmann, 2021-12-02:
That's right. The interface between 4G eNB and 5G gNB is over X2. However, we only use dummy X2 interface but don't have the full packing/unpacking over the real interface. Therefore, we have this dual channel RF requirement.