srsran/srsRAN_4G #758: srsenb (release 21.10) doesn't start with USRP X310 on 5G NSA mode

vishvikkrishnan, 2021-11-10:
## Issue Description ##
The srsenb application in srsRAN release 21.10 fails to start properly in 5G NSA mode when running on USRP X310. I keep getting the following error on the console:

```
[ERROR] [UHD RF] RuntimeError: Multiple sampling rates upstream of RX Terminator 0: RuntimeError: Node 0/DDC_1 specifies 1.92e+06, node 0/DDC_0 specifies 1.152e+07.
Setting frequency: DL=2680.0 Mhz, UL=2560.0 MHz for cc_idx=0 nof_prb=50
[ERROR] [UHD RF] RuntimeError: Multiple sampling rates upstream of RX Terminator 0: RuntimeError: Node 0/DDC_1 specifies 1.92e+06, node 0/DDC_0 specifies 1.152e+07.
Setting frequency: DL=1842.5 Mhz, UL=1747.5 MHz for cc_idx=1 nof_prb=52
```

## Setup Details ##
System Hardware: Intel NUC
Operating System: Ubuntu 20.04.3 LTS (Linux kernel version 5.11.0-40-generic)
RF front-end: USRP X310 with UBX-160 daughterboards
Library & Driver versions: GNU C++ version 9.3.0; Boost_107100; UHD_3.15.0.0-release
Connectivity: 10Gbps ethernet connection using an SFP+ connector and a Cat. 6 RJ45 cable.

## Expected Behavior ##
srsenb starts up with both LTE and NR links operational and without any error on the console.

## Actual Behaviour ##
Error preventing srsenb from starting up. The TX LEDs (red color) on both channels of the X310 keep blinking continuously until the application is aborted. RX LEDs were stable all along. srsenb works totally fine with LTE-only eNB setup, i.e., without using the gNB on rf_port 1 (_rr.conf_ without any cells in the NR Cell List). Using a different USRP does not fix the issue.

## Steps to reproduce the problem ##
1. Follow srsRAN documentation to pull the latest code from release 21.10 and perform make install.
2. Update _rr.conf_ with the NR Cell info under the nr_cell_list as outlined in the official documentation.
3. Start srsepc as usual.
4. Start srsenb. The application seems to start, but shortly after, an error as highlighted in the description is noticed on the console. Error log and console log are attached below.

## Additional Information ##
The zmq driver approach works totally fine with the same setup. The enb.log with log_level set to info shows a lot of "late" and "Underflow" messages although the network connection is set to work with MTU 9000 and the system detects the interface to be a 10000 Mbps connection with the USRP. Disabling CPU frequency scaling doesn't help as well. _epc.conf_ was unmodified. Modified _rr.conf_ and _enb.conf_ files are attached with this issue

[console.log](https://github.com/srsran/srsRAN/files/7508913/console.log)
[enb.conf.txt](https://github.com/srsran/srsRAN/files/7508914/enb.conf.txt)
[enb.log](https://github.com/srsran/srsRAN/files/7508915/enb.log)
[rr.conf.txt](https://github.com/srsran/srsRAN/files/7508916/rr.conf.txt)

andrepuschmann, 2021-11-17:
Sorry for the delay in responding and thanks for the pointer again. Looking at your `enb.conf` it seems you're not using the right device arguments for the X310. Have a look at https://docs.srsran.com/en/latest/app_notes/source/5g_nsa_cots/source/index.html#srsenb and take the exact ones to correctly initialize the device.

vishvikkrishnan, 2021-11-17:
Thanks for getting back, Andre!

In `device_args=type=x300,clock=external,sampling_rate=11.52e6,lo_freq_offset_hz=11.52e6`, should I specifically use an external clock source for the X310 or would the internal clock work just fine?

andrepuschmann, 2021-11-17:
Yeah, you can remove the `clock` param. The important one is the `sampling_rate` to remove your UHD warnings. I am closing the issue. Let me know if you still see it happening.

closed 2021-11-17 by andrepuschmann

vishvikkrishnan, 2021-11-19:
@andrepuschmann, That resolved the issue, thanks! I appreciate your timely support.