srsran/srsRAN_4G #761: Manual Camping of phones after provisioning

kielcondi, 2021-11-12:
## Issue Description ##
Manual camping of phones after provisioning

We're running an EPC --> enodeB --> SDR  (b210) --> duplexer --> antenna . A complete set-up for a box base station. SRSENB 20.10.1 , UHD 3.15.0 .  

## Behavior ##
After provisioning the phones to the EPC, we expected the phones to automatically camp to the network at first, but it didn't happen so we tried to manual camp it to the network and it did camped. After our first uncamping of phones, we expected it to automatically camp to the network just in case there are other considerations for first camping, but as the previous experience it didn't happen. 

We tried to wait for a longer time than usual to make it latch automatically on the network but we failed to latch it. We were always resorted to manual camping. 

If you have any request of log files and other configuration files, please feel free to comment on this forum for me to get it posted. Thank you so much all for your future support.

andrepuschmann, 2021-11-12:
I actually don't understand your question. But in any case, use the newest release (21.10) and yes, explain the issue again and provide logs and PCAP which help us to see what's going wrong.

kielcondi, 2021-11-15:
Sorry for the confusion. I'm new to this software to work on, please have a patience with me haha thank you.
We're running an EPC --> enodeB --> SDR (b210) --> duplexer --> antenna . A complete set-up for a box base station. SRSENB 20.10.1 , UHD 3.15.0 .

What we encountered was the phone need to manual search for the network to camped on instead of an automatic latching on the network. 

I have here is the PCAP when the scenario is i'm trying to make the phone  latch automatically without trying to manual search the network. From the last trace in the PCAP, i tried to do the manual search of the network and camp it since it's not camping automatically. 192.168.140.6 is my UE's IP. Thank you so much.
[pcap.txt](https://github.com/srsran/srsRAN/files/7537277/pcap.txt)
.

andrepuschmann, 2021-11-19:
I actually don't think there is an issue on the eNB. Some phones just take a bit longer to attach and need manual search because the SIM card IMSI doesn't match with the PLMN or what ever. Also phones store details of last connected networks (PLMN, EARFCN) and prefer those the next time. So that all requires a bit of manual tuning. Closing the issue.

closed 2021-11-19 by andrepuschmann

kielcondi, 2021-11-22:
Sorry for the late notice, here is another trace from srsenb showing different phones trying to latch to our network and get rejected. Does this affect why phones took longer to camp? Is there any way our network don't get congested by those phones that are trying to latch? Thank you.

[Phones trying to latch.txt](https://github.com/srsran/srsRAN/files/7578783/Phones.trying.to.latch.txt)

andrepuschmann, 2021-11-22:
You probably overload the system. You don't send logs but you probably have many RF issues (lates, etc) that cause the phones to reattach. Reduce the PRBs and don't run MIMO

kielcondi, 2021-11-23:
I see. Thank you so much for your help.