srsran/srsRAN_4G #765: SDR x300 not transmiting 5G frequency

joaogro, 2021-11-26:
Hello there, I'm following the 5G NSA cots guide and for some reason the SDR is not transmiting the 5G frequency. I have a SDR X300 conected to a server by 10gb sfp conection ( I tested it with a fiber and a cat6 cable and the result seems to be the same), the server has enough resources to run the sdr application ( it has 32 Gb of RAM), but when I start the srsenb application  even though it seems to be working fine, (because the output is the same of the logs showed in the guide) it is not transmiting the 5G frequency. I say that is not transmiting because I have a spectrum analyser, and it shows the 4G frequency but not the 5G one. I tested it with de n3 and n1 bands, and for the 4G with the band 1, 3 and 20. I followed this link:https://www.sqimway.com/nr_endc.php to see if the combinations were valid. I'm sending the configuration files and some prints showing what I describe above, if anything else is needed please let me now, thank you advance for the help, i really appreciate it.
[confs.zip](https://github.com/srsran/srsRAN/files/7608741/confs.zip)
[image]
[image]
[image]

andrepuschmann, 2021-11-26:
Please check you `rr.conf` file again and make sure that you don't overwrite the DL EARFCN. I see a red line in stdout that shouldn't be there. Also note that on the NR carrier, there is only the SSB every 10ms, so there is not much to see on an idle system.

joaogro, 2021-11-26:
i will check it out, thank you!

closed 2022-04-25 by joaogro