srsran/srsRAN_4G #769: RRC-NR reconfiguration failing with srsRAN gNB-NSA

vishvikkrishnan, 2021-12-08:
## Issue Description ##
The srsue application doesn't get 5G-NSA connection since RRC-NR reconfiguration fails. UE log suggests the following warning before reconfiguration fails:
[RRC-NR ] [W] Option zp_csi_rs_res_to_add_mod_list not present
[RRC-NR ] [W] Option csi_report_cfg_to_add_mod_list not present

## Setup Details ##
Same setup and specs as in [Issue 758](https://github.com/srsran/srsRAN/issues/758)
+
USRP X310 with UBX-160 daughterboards as RF-front end for 5G UE (same system hardware for UE as in the linked issue)

## Expected Behavior ##
SDR-based 5G UE should be able to get successful RRC-NR reconfiguration.

## Actual Behaviour ##
UE gets IP address, but RRC-NR reconfiguration fails. Note that the [rat.nr] config is modified to reflect:
```
[rat.nr]
bands = 3,78
nof_carriers = 1

and 
[rrc]
release = 15

```
However, still the UE only gets LTE connection.

## Steps to reproduce the problem ##
1. Same steps as in [Issue 758](https://github.com/srsran/srsRAN/issues/758)
2. Use TDD mode in rr.conf instead of FDD mode (comment out FDD and uncomment TDD settings).
2. Make changes to UE config in the [rat.nr] section and the [rrc] sections as mentioned in the 5G NSA End-to-End app note.
3. Start UE. UE attaches to the eNB and gets an IP. But RRC-NR reconfiguration fails.

## Additional Information ##
COTS UEs are able to connect to the srsRAN gNB on FDD mode. The config files, console, and UE logs are attached. When using FDD mode for SDR-based UE, the RRC-NR reconfiguration fails stating "TDD config not present".

vishvikkrishnan, 2021-12-17:
@andrepuschmann, just following up, is there an update on this issue?

andrepuschmann, 2022-01-06:
The FDD support in srsUE 21.04 was very limited, well, actually, it wasn't present as we focused on TDD on band n78. That has changed with 21.10 and FDD works with both srsENB and srsUE. In your logs, it seems you're using srsUE 21.04 though. So the behavior is expected.

```
22:56:25.830006 [UE     ] [I] Using binary srsue with arguments: /root/.config/srsran/ue5g.conf 
22:56:25.830203 [UE     ] [I] Built in RelWithDebInfo mode using 21.04.0.
```

andrepuschmann, 2022-01-06:
I am closing the issue, please feel free to reopen if needed.

closed 2022-01-06 by andrepuschmann

santos-j, 2022-01-12:
Thank you so much, @andrepuschmann!