srsran/srsRAN_4G #770: srsUE LTE TDD BUG

Hfywtias, 2021-12-09:
## Issue Description ##
[Describe the issue in detail]
srsUe connect CMU500 in TDD earfcn, RA fail
ue: srsUE, version:srsRAN-release_19_12, usrpX310
enb/epc: cmu500

## Setup Details ##
[Specify details of the test setup. This would help us reproduce the problem reliably]
RA fail, ue send RA.msg3, but cmu500 not recieved

## Expected Behavior ##
[What you expect to happen]
RA success, attach success

## Actual Behaviour ##
[What happens instead e.g. error message]

## Steps to reproduce the problem ##
[Tell us how to reproduce this issue e.g. RF setup, application config files]

## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]
BUG repair: 
file: srsue/src/phy/sync.cc  
function: set_time_adv_sec
void sync::set_time_adv_sec(float time_adv_sec)
{
  // If transmitting earlier, transmit less samples to align time advance. If transmit later just delay next TX
  next_offset             = (int)round((this->time_adv_sec - time_adv_sec) * srslte_sampling_freq_hz(cell.nof_prb));
  this->next_time_adv_sec = time_adv_sec;

  #define __TDD__          
  #ifdef __TDD__
  this->next_time_adv_sec = time_adv_sec+624 *SRSLTE_LTE_TS;      //repair like this
  #endif

  
  Info("Applying time_adv_sec=%.1f us, next_offset=%d\n", time_adv_sec * 1e6, next_offset);
}
after repair, ue attach success

jtang613, 2022-02-03:
Hi @Hfywtias ,

Could you please generate a pull request with your fix? I'm noticing similar problems with SRSUE not connecting to my TDD equipment and I'd like to confirm.

Thanks!

andrepuschmann, 2022-02-14:
Thanks for the report and this suggestion.

ismagom, 2023-11-06:
See https://github.com/srsran/srsRAN_4G/issues/805#issuecomment-1796382633

closed 2023-11-06 by ismagom