srsran/srsRAN_4G #775: srsRAN and LimeSDR [with COTS UE]

nikolicnemanja991, 2021-12-16:
Hi,

I am running srsepc and srsenb on Ubuntu 18.04 (srsRAN is latest: 21.10).
In order to configure EPC, ENB to work with LimeSDR I have followed your srsRAN Documentation, Release 21.10 (section: 5.2 COTS UE).
I can see my private network on my mobile phone (Samsung Galaxy A9) but connection is very unstabile, please see the logs:

[image]

Could you tell me how this issue can be fixed?

I connected very good antennas with 38dBi:
https://www.aliexpress.com/item/1005002756942513.html?spm=a2g0s.9042311.0.0.45b64c4d1EUkid

Here is my configuration:
#####################################################
[rf]
#dl_earfcn = 3350
tx_gain = 80
rx_gain = 40
device_name = soapy
#device_args = auto
device_args = rxant=LNAH,txant=BAND1
#####################################################
[mme]
mme_code = 0x1a
mme_group = 0x0001
tac = 0x0007
mcc = 901
mnc = 70
mme_bind_addr = 127.0.1.100
apn = srsapn
dns_addr = 8.8.8.8
encryption_algo = EEA0
integrity_algo = EIA1
paging_timer = 2
request_imeisv = false

andrepuschmann, 2021-12-17:
hey, try to play with the gains, likely you're saturating the receiver.

nikolicnemanja991, 2021-12-17:
What are the limits for rx and tx gain?
Do I need additional power supply?
I have only USB 3.0.

UE attaches to the network first time but after it disconnect due to the signal los it is not possible to re-attach. It happens friquently.
I am not sure why.

nomidaepapi, 2021-12-21:
Hi,

Try lowering the no_prb from the ENB, it helped me.

nikolicnemanja991, 2021-12-22:
Thanks for suggestion. 
It seems that I need duplexer.

jtang613, 2022-01-27:
Your RX gain is likely too low. I've found values of around 75 to work best on LimeSDR-USB. Alternatively, commenting out the rx_gain line to allow the AGC to adjust the gain also works reasonably well on LimeSDR. A duplexer is not a hard requirement for the LimeSDR to work - it will work without it - but I've been successfully using my own Band4 and Band1 duplexers with the LimeSDR without issues either. Lastly, 50 PRB should work fine - I only have issues at 100 PRB.