srsran/srsRAN_4G #776: Permanent disconnecing RNTI, USRP X310

sergunya1979, 2021-12-17:
## Issue Description ##
UEs cannot connect srsenb. Always messages like that:
RACH:  tti=4211, cc=0, preamble=54, offset=0, temp_crnti=0x46
RACH:  tti=4231, cc=0, preamble=21, offset=0, temp_crnti=0x47
RACH:  tti=4251, cc=0, preamble=23, offset=0, temp_crnti=0x48
RACH:  tti=4271, cc=0, preamble=43, offset=0, temp_crnti=0x49
Disconnecting rnti=0x46.
RACH:  tti=4291, cc=0, preamble=42, offset=0, temp_crnti=0x4a
Disconnecting rnti=0x47.
RACH:  tti=4311, cc=0, preamble=22, offset=0, temp_crnti=0x4b
Disconnecting rnti=0x48.
RACH:  tti=4331, cc=0, preamble=53, offset=0, temp_crnti=0x4c
Disconnecting rnti=0x49.
RACH:  tti=4351, cc=0, preamble=2, offset=12, temp_crnti=0x4d
Disconnecting rnti=0x4a.
RACH:  tti=4371, cc=0, preamble=44, offset=0, temp_crnti=0x4e
Disconnecting rnti=0x4b.
RACH:  tti=4391, cc=0, preamble=39, offset=0, temp_crnti=0x4f
Disconnecting rnti=0x4c.
RACH:  tti=4411, cc=0, preamble=41, offset=0, temp_crnti=0x50

          -----------------DL----------------|-------------------------UL-------------------------
rat rnti  cqi  ri  mcs  brate   ok  nok  (%) | pusch  pucch  phr  mcs  brate   ok  nok  (%)    bsr
lte   4d  n/a   0    0      0    0    0   0% |  -0.5    n/a    0    0      0    0    4 100%    0.0
lte   4e  n/a   0    0      0    0    0   0% |  -0.6    n/a    0    0      0    0    4 100%    0.0
lte   4f  n/a   0    0      0    0    0   0% |  -0.6    n/a    0    0      0    0    2 100%    0.0
lte   50  n/a   0    0      0    0    0   0% |  -0.6    n/a    0    0      0    0    0   0%    0.0

## Setup Details ##
[Specify details of the test setup. This would help us reproduce the problem reliably]
e.g. Network configuration, Operation System, Hardware, RF front-end, library and driver versions
Core: commercial. Host:HP proliant ML310e Gen8/32Gb RAM
Bare HW Ubuntu 18.04.6 LTS/low latency kernel/performance mode
USRP 2940R/MB x310
UHD_003.009.007-0-g50839059
SRS 21_10

## Expected Behavior ##
[What you expect to happen]

## Actual Behaviour ##
[What happens instead e.g. error message]

## Steps to reproduce the problem ##
[Tell us how to reproduce this issue e.g. RF setup, application config files]

## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]

gaurishdotkdm, 2022-02-10:
@sergunya1979 were you able to fix the issue?

andrepuschmann, 2022-02-14:
Please share full logs and describe radio config. Otherwise we can't help here. Closing the issue until further info is provided.

closed 2022-02-14 by andrepuschmann