srsran/srsRAN_4G #780: 5G NSA COTS UE with iphone

Lu-Yi-Hsun, 2021-12-24:
i have been try and rebuild many time
does any people test and work 5G NSA COTS UE with iphone on b210

i only success setting lte for iphone13 pro,but nsa still can not find the signal on iphone

kbarlee, 2021-12-30:
iPhones are carrier firmware locked, so will generally not even attempt to search for NSA cells/ signal NSA UEcaps when the PLMN is not onboarded by Apple. Just Apple being evil... Have had more joy with Oppo and Huawei phones.

andrepuschmann, 2022-01-06:
We never tested with 5G NSA with Apple products ourselves. But you could try to configure the network to use a PLMN that you know your phone works with, see @kbarlee comment. But make sure to comply with your regulators restrictions and use a shielding box for it.

ismagom, 2023-04-14:
Please provide logs so that others can help you. Thanks