srsran/srsRAN_4G #782: volte issues srsran21.10 with open5gs

zxmek, 2021-12-26:
I use the srsenb with open5gs. I can use volte normally. after about 6 minutes.  I can not use volte . but I  can only  surf the Internet normally.
here is the enb.log
[enb.log](https://github.com/srsran/srsRAN/files/7776744/enb.log)

zxmek, 2021-12-26:
[image]
the max lcid is 10. here I get 11. What are the possible reasons for this result？

andrepuschmann, 2022-01-06:
Thanks for reporting this. @zxmek Can you try with srsENB 21.04 please?
@herlesupreeth have you tried VoLTE with 21.10 by any chance?

herlesupreeth, 2022-01-06:
@andrepuschmann Unfortunately not, still using the 21.04 and it works falwlessly there

andrepuschmann, 2022-01-19:
@zxmek Could you try 21.04 just to double check.

If you confirm there is an issue with 21.10 please send logs of the enb in info mode. Thanks

zxmek, 2022-01-20:
> Thanks for reporting this. @zxmek Can you try with srsENB 21.04 please? @herlesupreeth have you tried VoLTE with 21.10 by any chance?

thank you. now I use 21.04. there are a lot of  message。As shown below.

RACH:  tti=661, cc=0, preamble=1, offset=1, temp_crnti=0xd6
Disconnecting rnti=0xd6.
RACH:  tti=941, cc=0, preamble=38, offset=1, temp_crnti=0xd7
User 0xd7 requesting RRC Reestablishment as 0xcc. Cause: otherFailure
RACH:  tti=2271, cc=0, preamble=27, offset=1, temp_crnti=0xd8
Disconnecting rnti=0xd8.
Disconnecting rnti=0xcc.
RACH:  tti=2321, cc=0, preamble=7, offset=1, temp_crnti=0xd9
Disconnecting rnti=0xd9.
User 0xd7 connected

I use the B210 .  should I need use the gpsdo?

zxmek, 2022-01-20:
I have use the 21.10.  I get lots of  the RACH messages and  RRC Reestablishment as 0xcc. Cause: otherFailure. 
 I use volte .it seems not very stable. After a few minutes , I cannot make volte，it noly can surf the internet .the srsenb log .I have been upload in the first issue.
[enb.log](https://github.com/srsran/srsRAN/files/7905507/enb.log)

I think, the 21.04 is more stable.

caio-amarala, 2022-01-21:
Hi @zxmek and @andrepuschmann.

I'm having a very similar issue:

## Issue Description ##    
When i try to make a volte call srsenb starts to become unstable, reporting several RACH and Disconnecting messages. Even when the call is up and everything looks good, if i move a little bit my COTs phone the same messages appears and instability begins.

I tried to change tx and rx gains, earfcn and the srsran release but the same behavior occurs.

## Setup Details ##

We have a X300 SDR and the services are running all in one physical machine (Ubuntu 18.04.6 LTS, intel 8 core i7-10700F): srsenb (I tried with master and release 21.04) + open5gs + Kamailio IMS (master branch) + UHD (master branch)

## Steps to reproduce the problem ##

1. Start IMS services
2. Start all Open5GS services
3. Start srsenb
4. COTS phones LTE attaches and IMS registers //everything is stable up until this point
5. Start a VoLTE call between the two phones

After the call is established, RRCConnectionReestablishmentRequests are observed frequently. All with specified cause otherFailure (which leads me to believe this is related to QCI 1 DRB )

Config files attached with a pcap log:
[srsran.zip](https://github.com/srsran/srsRAN/files/7915001/srsran.zip)
[confs_open5gs (1).zip](https://github.com/srsran/srsRAN/files/7915002/confs_open5gs.1.zip)
pcap: https://drive.google.com/file/d/17oBq27Om4ZnmmIiQuYxaBL6JN6m0vH32/view?usp=sharing

zxmek, 2022-01-21:
> Thanks for reporting this. @zxmek Can you try with srsENB 21.04 please? @herlesupreeth have you tried VoLTE with 21.10 by any chance?

thank you . I Find the question in the old issue.

caio-amarala, 2022-01-21:
> > Thanks for reporting this. @zxmek Can you try with srsENB 21.04 please? @herlesupreeth have you tried VoLTE with 21.10 by any chance?
> 
> thank you . I Find the question in the old issue.

Hi @zxmek , where did you find this old issue?

Thanks