srsran/srsRAN_4G #783: SRSenb doesn't recognize usrpb210 despite it is known by pc

emadrezz, 2021-12-28:
## Issue Description ##
I have installed UHD_003.010.003.000-0-unknown and all the dependencies for srslte and usrpb210 and downloaded by `uhd_images_downloader ` and loaded into usrp by `uhd_usrp_probe ` and it worked fine and defined accurately to my system and even tasted it by gnu-radio companion and was ok. but when i run the srsran the problem shows up that says  **/home/emadrez/srsRAN/lib/src/phy/rf/rf_imp.c:127: RF device 'UHD' not found. Please check the available srsRAN CMAKE options to verify if this device is being detected in your system** , I also checked the CMakelist as the error says and the uhd is ON and also checked the device args that using the default args and the device name is UHD.

## Setup Details ##
I'm using usrpB210 , UHD_003.010.003.000-0-unknown , ubuntu 18.04 LTS , only changed that ive made was the bandwidth that i change n_prb that i don't think affects on this situation.

## Actual Behaviour ##

emadrez@emadrez-VAIO:~$ uhd_usrp_probe 
linux; GNU C++ version 7.3.0; Boost_106501; UHD_003.010.003.000-0-unknown

-- Detected Device: B210
-- Operating over USB 2.
-- Initialize CODEC control...
-- Initialize Radio control...
-- Performing register loopback test... pass
-- Performing register loopback test... pass
-- Performing CODEC loopback test... pass
-- Performing CODEC loopback test... pass
-- Setting master clock rate selection to 'automatic'.
-- Asking for clock rate 16.000000 MHz... 
-- Actually got clock rate 16.000000 MHz.
-- Performing timer loopback test... pass
-- Performing timer loopback test... pass
  _____________________________________________________
 /
|       Device: B-Series Device
|     _____________________________________________________
|    /
|   |       Mboard: B210
|   |   revision: 4
|   |   product: 2
|   |   serial: 3189F60
|   |   name: MyB210
|   |   FW Version: 8.0
|   |   FPGA Version: 14.0
|   |   
|   |   Time sources:  none, internal, external, gpsdo
|   |   Clock sources: internal, external, gpsdo
|   |   Sensors: ref_locked
|   |     _____________________________________________________
|   |    /
|   |   |       RX DSP: 0
|   |   |   
|   |   |   Freq range: -8.000 to 8.000 MHz
|   |     _____________________________________________________
|   |    /
|   |   |       RX DSP: 1
|   |   |   
|   |   |   Freq range: -8.000 to 8.000 MHz
|   |     _____________________________________________________
|   |    /
|   |   |       RX Dboard: A
|   |   |     _____________________________________________________
|   |   |    /
|   |   |   |       RX Frontend: A
|   |   |   |   Name: FE-RX2
|   |   |   |   Antennas: TX/RX, RX2
|   |   |   |   Sensors: temp, rssi, lo_locked
|   |   |   |   Freq range: 50.000 to 6000.000 MHz
|   |   |   |   Gain range PGA: 0.0 to 76.0 step 1.0 dB
|   |   |   |   Bandwidth range: 200000.0 to 56000000.0 step 0.0 Hz
|   |   |   |   Connection Type: IQ
|   |   |   |   Uses LO offset: No
|   |   |     _____________________________________________________
|   |   |    /
|   |   |   |       RX Frontend: B
|   |   |   |   Name: FE-RX1
|   |   |   |   Antennas: TX/RX, RX2
|   |   |   |   Sensors: temp, rssi, lo_locked
|   |   |   |   Freq range: 50.000 to 6000.000 MHz
|   |   |   |   Gain range PGA: 0.0 to 76.0 step 1.0 dB
|   |   |   |   Bandwidth range: 200000.0 to 56000000.0 step 0.0 Hz
|   |   |   |   Connection Type: IQ
|   |   |   |   Uses LO offset: No
|   |   |     _____________________________________________________
|   |   |    /
|   |   |   |       RX Codec: A
|   |   |   |   Name: B210 RX dual ADC
|   |   |   |   Gain Elements: None
|   |     _____________________________________________________
|   |    /
|   |   |       TX DSP: 0
|   |   |   
|   |   |   Freq range: -8.000 to 8.000 MHz
|   |     _____________________________________________________
|   |    /
|   |   |       TX DSP: 1
|   |   |   
|   |   |   Freq range: -8.000 to 8.000 MHz
|   |     _____________________________________________________
|   |    /
|   |   |       TX Dboard: A
|   |   |     _____________________________________________________
|   |   |    /
|   |   |   |       TX Frontend: A
|   |   |   |   Name: FE-TX2
|   |   |   |   Antennas: TX/RX
|   |   |   |   Sensors: temp, lo_locked
|   |   |   |   Freq range: 50.000 to 6000.000 MHz
|   |   |   |   Gain range PGA: 0.0 to 89.8 step 0.2 dB
|   |   |   |   Bandwidth range: 200000.0 to 56000000.0 step 0.0 Hz
|   |   |   |   Connection Type: IQ
|   |   |   |   Uses LO offset: No
|   |   |     _____________________________________________________
|   |   |    /
|   |   |   |       TX Frontend: B
|   |   |   |   Name: FE-TX1
|   |   |   |   Antennas: TX/RX
|   |   |   |   Sensors: temp, lo_locked
|   |   |   |   Freq range: 50.000 to 6000.000 MHz
|   |   |   |   Gain range PGA: 0.0 to 89.8 step 0.2 dB
|   |   |   |   Bandwidth range: 200000.0 to 56000000.0 step 0.0 Hz
|   |   |   |   Connection Type: IQ
|   |   |   |   Uses LO offset: No
|   |   |     _____________________________________________________
|   |   |    /
|   |   |   |       TX Codec: A
|   |   |   |   Name: B210 TX dual DAC
|   |   |   |   Gain Elements: None

The error when i execute the enb:
[image]
my CMakelist :
[image]

[image]

zxmek, 2021-12-30:
Try to rebuild the srsran use the usrp b210,not use other sdr devices

emadrezz, 2022-01-02:
@zxmek thanks mate
but i didn't catch what do you mean by not use any other sdr devices; please tell me more details if its possible.
I cleaned and rebuild it again but still doesn't seem right!

zxmek, 2022-01-03:
> @zxmek thanks mate but i didn't catch what do you mean by not use any other sdr devices; please tell me more details if its possible. I cleaned and rebuild it again but still doesn't seem right!

from the pic  ,first ,you should solve the ip address. your srsenb cannot bind your IP address. then rebuild the srsenb..  if still have problem. you could search it in google