srsran/srsRAN_4G #786: SRSLTE NOT DETECTING XB300

ramgaik, 2022-01-10:
## Issue Description ##
I have a srsLTE project running with bladeRF X40. Now I want to enhance bladeRF Transmission power so I have purchased amplifier XB300. Using the wiki page I have setup the XB300 with bladeRF X40. The connections are ok, the amplifier enabled successfully, the PA enabled successfully and the LNA also enabled successfully. But when I user to start srsenb it is not recognizing the XB 300.   

## Setup Details ##
My srsLTE is working fine with bladeRF X40 but the srsenb is not reading the amplfier board XB300
## Expected Behavior ##
The srsenb should have start with the XB300

## Actual Behaviour ##
srsenb not reading the amplifier board XB300 and returning with an error that no RF front end found.

## Steps to reproduce the problem ##
setting of srsLTE with bladeRF X40 and XB300

## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]

andrepuschmann, 2022-01-10:
You need to share logs, console output etc. Compare it with the bladeRF wiki etc and other sources that may use the XB300

ramgaik, 2022-01-12:
Thank for the reply.

**1. **Console output is as follows****

sudo srsenb
---  Software Radio Systems LTE eNodeB  ---

Reading configuration file /root/.config/srsran/enb.conf...
WARNING: cpu0 scaling governor is not set to performance mode. Realtime processing could be compromised. Consider setting it to performance mode before running the application.

Built in Release mode using commit 1c6dd8c4a on branch HEAD.

Opening 1 channels in RF device=default with args=default
Available RF device list: bladeRF  zmq 
Trying to open RF device 'bladeRF'
Opening bladeRF...
/home/volte/srsLTE/lib/src/phy/rf/rf_blade_imp.c:191: Unable to open device: No devices available
Unable to open RF device 'bladeRF'
Trying to open RF device 'zmq'
[zmq] Error: No device 'args' option has been set. Please make sure to set this option to be able to use the ZMQ no-RF module
Unable to open RF device 'zmq'
/home/volte/srsLTE/lib/src/phy/rf/rf_imp.c:145: Failed to open a RF frontend device. Please check the available srsRAN CMAKE options to verify what RF frontend devices have been detected in your system
Error initializing radio.

**2. enb logs are as follows**
08:11:19.615637 [ENB    ] [I] Using binary srsenb with arguments: 
08:11:19.615716 [ENB    ] [I] Built in Release mode using commit 1c6dd8c4a on branch HEAD.
08:11:19.643772 [RF     ] [E] Error opening RF device
08:11:19.643775 [RF     ] [E] Error opening RF device 0
08:11:19.645216 [COMN   ] [D] [    0] RxSockets: Closing rx socket handler thread
08:11:19.645260 [COMN   ] [D] [    0] RxSockets: closed.

Thanks in advance

ramgaik, 2022-01-16:
> You need to share logs, console output etc. Compare it with the bladeRF wiki etc and other sources that may use the XB300

Thank for the reply.

1. Console output is as follows

sudo srsenb
--- Software Radio Systems LTE eNodeB ---

Reading configuration file /root/.config/srsran/enb.conf...
WARNING: cpu0 scaling governor is not set to performance mode. Realtime processing could be compromised. Consider setting it to performance mode before running the application.

Built in Release mode using commit 1c6dd8c on branch HEAD.

Opening 1 channels in RF device=default with args=default
Available RF device list: bladeRF zmq
Trying to open RF device 'bladeRF'
Opening bladeRF...
/home/volte/srsLTE/lib/src/phy/rf/rf_blade_imp.c:191: Unable to open device: No devices available
Unable to open RF device 'bladeRF'
Trying to open RF device 'zmq'
[zmq] Error: No device 'args' option has been set. Please make sure to set this option to be able to use the ZMQ no-RF module
Unable to open RF device 'zmq'
/home/volte/srsLTE/lib/src/phy/rf/rf_imp.c:145: Failed to open a RF frontend device. Please check the available srsRAN CMAKE options to verify what RF frontend devices have been detected in your system
Error initializing radio.

2. enb logs are as follows
08:11:19.615637 [ENB ] [I] Using binary srsenb with arguments:
08:11:19.615716 [ENB ] [I] Built in Release mode using commit 1c6dd8c on branch HEAD.
08:11:19.643772 [RF ] [E] Error opening RF device
08:11:19.643775 [RF ] [E] Error opening RF device 0
08:11:19.645216 [COMN ] [D] [ 0] RxSockets: Closing rx socket handler thread
08:11:19.645260 [COMN ] [D] [ 0] RxSockets: closed.

Thanks in advance

andrepuschmann, 2022-01-16:
Well, the blade is not even loading

ramgaik, 2022-01-17:
> Well, the blade is not even loading

Sir, thanks for your interest in the issue.

1. The bladeRF working fine when used alone.Everything is ok while using the sdr alone but when I connect the amplifier and enable it using console the amplifier enable LED, PA LED and LNA LED glows which means that the amplifier is connected and enabled.
2. But when I run srsenb then the problem occurs even though the amplifier LEDs are glowing. 
3. My observation is that when I enable the amplifier using console then the srsenb is assuming that the SDR is being used by another process that's why it is generating the error.
4. Now i am not able to find any way to enable the amplifier using srsenb script

Thanks