srsran/srsRAN_4G #787: UE not connecting to srsran

idonttest, 2022-01-12:
So i am running srsran recent release on ubuntu using USRP b210. 
UE is Motorola G5s Plus. 

Below are the changes I made in the config
>epc.conf

[mme]
mme_code = 0x1a
mme_group = 0x0001
tac = 0x0007
mcc = 001
mnc = 01
mme_bind_addr = 127.0.1.100
apn = srsapn
dns_addr = 8.8.8.8
encryption_algo = EEA0
integrity_algo = EIA1
paging_timer = 2
request_imeisv = false

>enb.conf

[enb]
enb_id = 0x19B
mcc = 001
mnc = 01
mme_addr = 127.0.1.100
gtp_bind_addr = 127.0.1.1
s1c_bind_addr = 127.0.1.1
s1c_bind_port = 0
n_prb = 15
tm = 2
nof_ports = 2

[rf]
dl_earfcn = 1934
#1934
tx_gain = 60
rx_gain = 40
 
>user_data.cfg

ue2,mil,001010123456780,00112233445566778899aabbccddeeff,op,63bfa50ee6523365ff14c1f45f88737d,8000,000000001656,7,dynamic
ue1,xor,001010123456789,00112233445566778899aabbccddeeff,opc,63bfa50ee6523365ff14c1f45f88737d,9001,000000001234,7,dynamic

The SIM card is configured as followed. 
[image]

My phone can detect the 001 01 network as CMCC. but the phone doesn't register to the network. 

Is there something i am doing wrong??

Below is the log 
---  Software Radio Systems EPC  ---

Reading configuration file epc.conf...
HSS Initialized.
MME S11 Initialized
MME GTP-C Initialized
MME Initialized. MCC: 0xf001, MNC: 0xff01
SPGW GTP-U Initialized.
SPGW S11 Initialized.
SP-GW Initialized.
Received S1 Setup Request.
S1 Setup Request - eNB Name: srsenb01, eNB id: 0x19b
S1 Setup Request - MCC:001, MNC:01
S1 Setup Request - TAC 7, B-PLMN 0xf110
S1 Setup Request - Paging DRX v128
Sending S1 Setup Response
Initial UE message: LIBLTE_MME_MSG_TYPE_ATTACH_REQUEST
Received Initial UE message -- Attach Request
Attach request -- IMSI: 001010123456780
Attach request -- eNB-UE S1AP Id: 1
Attach request -- Attach type: 2
Attach Request -- UE Network Capabilities EEA: 11110000
Attach Request -- UE Network Capabilities EIA: 01110000
Attach Request -- MS Network Capabilities Present: true
PDN Connectivity Request -- EPS Bearer Identity requested: 0
PDN Connectivity Request -- Procedure Transaction Id: 82
PDN Connectivity Request -- ESM Information Transfer requested: true
Downlink NAS: Sending Authentication Request
Initial UE message: LIBLTE_MME_MSG_TYPE_ATTACH_REQUEST
Received Initial UE message -- Attach Request
Attach request -- IMSI: 001010123456780
Attach request -- eNB-UE S1AP Id: 2
Attach request -- Attach type: 2
Attach Request -- UE Network Capabilities EEA: 11110000
Attach Request -- UE Network Capabilities EIA: 01110000
Attach Request -- MS Network Capabilities Present: true
PDN Connectivity Request -- EPS Bearer Identity requested: 0
PDN Connectivity Request -- Procedure Transaction Id: 83
PDN Connectivity Request -- ESM Information Transfer requested: true
Attach Request -- Found previously attach UE.
Downlink NAS: Sending Authentication Request
Received UE Context Release Request. MME-UE S1AP Id 1
No UE context to release found. MME-UE S1AP Id: 1
Initial UE message: LIBLTE_MME_MSG_TYPE_ATTACH_REQUEST
Received Initial UE message -- Attach Request
Attach request -- IMSI: 001010123456780
Attach request -- eNB-UE S1AP Id: 3
Attach request -- Attach type: 2
Attach Request -- UE Network Capabilities EEA: 11110000
Attach Request -- UE Network Capabilities EIA: 01110000
Attach Request -- MS Network Capabilities Present: true
PDN Connectivity Request -- EPS Bearer Identity requested: 0
PDN Connectivity Request -- Procedure Transaction Id: 84
PDN Connectivity Request -- ESM Information Transfer requested: true
Attach Request -- Found previously attach UE.
Downlink NAS: Sending Authentication Request
Received UE Context Release Request. MME-UE S1AP Id 2
No UE context to release found. MME-UE S1AP Id: 2
Initial UE message: LIBLTE_MME_MSG_TYPE_ATTACH_REQUEST
Received Initial UE message -- Attach Request
Attach request -- IMSI: 001010123456780
Attach request -- eNB-UE S1AP Id: 4
Attach request -- Attach type: 2
Attach Request -- UE Network Capabilities EEA: 11110000
Attach Request -- UE Network Capabilities EIA: 01110000
Attach Request -- MS Network Capabilities Present: true
PDN Connectivity Request -- EPS Bearer Identity requested: 0
PDN Connectivity Request -- Procedure Transaction Id: 85
PDN Connectivity Request -- ESM Information Transfer requested: true
Attach Request -- Found previously attach UE.
Downlink NAS: Sending Authentication Request
Received UE Context Release Request. MME-UE S1AP Id 3
No UE context to release found. MME-UE S1AP Id: 3
Initial UE message: LIBLTE_MME_MSG_TYPE_ATTACH_REQUEST
Received Initial UE message -- Attach Request
Attach request -- IMSI: 001010123456780
Attach request -- eNB-UE S1AP Id: 5
Attach request -- Attach type: 2
Attach Request -- UE Network Capabilities EEA: 11110000
Attach Request -- UE Network Capabilities EIA: 01110000
Attach Request -- MS Network Capabilities Present: true
PDN Connectivity Request -- EPS Bearer Identity requested: 0
PDN Connectivity Request -- Procedure Transaction Id: 86
PDN Connectivity Request -- ESM Information Transfer requested: true
Attach Request -- Found previously attach UE.
Downlink NAS: Sending Authentication Request
Received UE Context Release Request. MME-UE S1AP Id 4
No UE context to release found. MME-UE S1AP Id: 4
Received UE Context Release Request. MME-UE S1AP Id 5

Jackrabrett, 2022-01-13:
Try upping your resource blocks to 50?

Respectfully,
Brett Jeske

idonttest, 2022-01-13:
@Jackrabrett   After increasing the resource block to 50. Now my phone cannot detect the network all together..  Anything other than 15 doesn't show in the phone

Jackrabrett, 2022-01-13:
Look at your UE config file, it has op instead of opc. On your sim editor you have opc selected try that

Respectfully,
Brett Jeske

Jackrabrett, 2022-01-13:
I’m actually not seeing the SIM data that matches your editor in your ue config file, just ue1 and ue2. Did you add your SIM card info in there?

Respectfully,
Brett Jeske

idonttest, 2022-01-14:
@Jackrabrett   Thanks.. That was a stupid mistake of mine. Even after fixing it, the SIM card won't register with the network.. 
I changed my earfn_dl to 500 which is band 1. same issue.. played around with all setting on enb & epc.. I recreated the same using LimeSDR and same issue persist. I am assuming the problem is either in the UE or the SIM card itself.

JM-Lemmi, 2022-01-20:
Have you tried a different MCC/MNC from 001/01? I have read, that some devices refuse connecting to 001/01

Jackrabrett, 2022-01-20:
I’ve had good luck on Band 8

Respectfully,
Brett Jeske

closed 2023-04-14 by ismagom