srsran/srsRAN_4G #788: Error with SRSeNB: "Failed to initiate S1 connection. Attempting reconnection in 10 seconds"

rgrunbla, 2022-01-14:
## Issue Description ##
After launching srsenb, some errors messages appear, e.g. :

```
connect(): Connection refused
Failed to initiate S1 connection. Attempting reconnection in 10 seconds
```

## Setup Details ##

Operating System:
```
$ cat /etc/lsb-release 
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=20.04
DISTRIB_CODENAME=focal
DISTRIB_DESCRIPTION="Ubuntu 20.04.3 LTS"
```

Software Versions:
```
$ dpkg -l|rg uhd
ii  libgnuradio-uhd:amd64                         3.9.4.0-0~gnuradio~focal-0            amd64        gnuradio universal hardware driver functions
ii  libgnuradio-uhd3.8.1:amd64                    3.8.1.0~rc1-2build2                   amd64        gnuradio universal hardware driver functions
ii  libuhd-dev                                    4.1.0.4-0ubuntu1~focal1               amd64        universal hardware driver for Ettus Research products
ii  libuhd3.15.0:amd64                            3.15.0.0-2build5                      amd64        universal hardware driver for Ettus Research products - library
ii  libuhd4.1.0:amd64                             4.1.0.4-0ubuntu1~focal1               amd64        hardware driver for Ettus Research products
ii  soapysdr0.7-module-uhd:amd64                  0.3.6-2build2                         amd64        UHD device support for SoapySDR
ii  uhd-host                                      4.1.0.4-0ubuntu1~focal1               amd64        hardware driver for Ettus Research products - host apps
$ dpkg -l|rg srs
ii  srsenb                                        21.10-0ubuntu1~srsran1~20.04          amd64        Complete SDR LTE eNodeB application
ii  srsepc                                        21.10-0ubuntu1~srsran1~20.04          amd64        Light-weight LTE core network implementation
ii  srsran                                        21.10-0ubuntu1~srsran1~20.04          all          LTE software suite for software defined radios (metapackage)
ii  srsran-core                                   21.10-0ubuntu1~srsran1~20.04          amd64        Common files for srsRAN
ii  srsue                                         21.10-0ubuntu1~srsran1~20.04          amd64        User Equipment implementation for LTE
$ srsenb --version
---  Software Radio Systems LTE eNodeB  ---

Version 21.10.0
$ uhd_config_info --version
UHD 4.1.0.4-release

```

RF Frontend: 

```
$ uhd_usrp_probe |head -n 20
[INFO] [UHD] linux; GNU C++ version 9.3.0; Boost_107100; UHD_4.1.0.4-release
[INFO] [B200] Detected Device: B210
[INFO] [B200] Operating over USB 3.
[INFO] [B200] Initialize CODEC control...
[INFO] [B200] Initialize Radio control...
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Setting master clock rate selection to 'automatic'.
[INFO] [B200] Asking for clock rate 16.000000 MHz... 
[INFO] [B200] Actually got clock rate 16.000000 MHz.
  _____________________________________________________
 /
|       Device: B-Series Device
|     _____________________________________________________
|    /
|   |       Mboard: B210
|   |   serial: [REDACTED]
|   |   name: MyB210
|   |   product: 2
|   |   revision: 4
|   |   FW Version: 8.0
|   |   FPGA Version: 16.0
|   |   
|   |   Time sources:  none, internal, external, gpsdo
|   |   Clock sources: internal, external, gpsdo
|   |   Sensors: ref_locked

```

## Expected Behavior ##

I expect no error messages on stdout.

## Actual Behaviour ##

Full logs:

```
$ srsenb 
---  Software Radio Systems LTE eNodeB  ---

Couldn't open , trying /home/remy/.config/srsran/enb.conf
Reading configuration file /home/remy/.config/srsran/enb.conf...
Couldn't open sib.conf, trying /home/remy/.config/srsran/sib.conf
Couldn't open rr.conf, trying /home/remy/.config/srsran/rr.conf
Couldn't open rb.conf, trying /home/remy/.config/srsran/rb.conf
WARNING: cpu0 scaling governor is not set to performance mode. Realtime processing could be compromised. Consider setting it to performance mode before running the application.
Failed to `mlockall`: {}
Built in RelWithDebInfo mode using 21.10.0.

srsLog error - Unable to create log file "/tmp/enb.log": Permission denied
connect(): Connection refused
Failed to initiate S1 connection. Attempting reconnection in 10 seconds
Warning: Failed to create thread with real-time priority. Creating it with normal priority: Connection refused
Opening 1 channels in RF device=default with args=default
Available RF device list: UHD 
Trying to open RF device 'UHD'
[INFO] [UHD] linux; GNU C++ version 9.2.1 20200304; Boost_107100; UHD_3.15.0.0-2build5
[INFO] [LOGGING] Fastpath logging disabled at runtime.
Opening USRP channels=1, args: type=b200,master_clock_rate=23.04e6
[INFO] [UHD RF] RF UHD Generic instance constructed
[INFO] [B200] Detected Device: B210
[INFO] [B200] Operating over USB 3.
[INFO] [B200] Initialize CODEC control...
[INFO] [B200] Initialize Radio control...
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Asking for clock rate 23.040000 MHz... 
[INFO] [B200] Actually got clock rate 23.040000 MHz.
RF device 'UHD' successfully opened
Warning: Failed to create thread with real-time priority. Creating it with normal priority: No such file or directory
Warning: Failed to create thread with real-time priority. Creating it with normal priority: No such file or directory
Warning: Failed to create thread with real-time priority. Creating it with normal priority: No such file or directory
Warning: Failed to create thread with real-time priority. Creating it with normal priority: No such file or directory

==== eNodeB started ===
Type <t> to view trace
Setting frequency: DL=2680.0 Mhz, UL=2560.0 MHz for cc_idx=0 nof_prb=50
connect(): Connection refused
Failed to initiate S1 connection. Attempting reconnection in 10 seconds
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 18.9235 >= 18.9234. Starting new burst...
connect(): Connection refused
Failed to initiate S1 connection. Attempting reconnection in 10 seconds
connect(): Connection refused
Failed to initiate S1 connection. Attempting reconnection in 10 seconds
connect(): Connection refused
Failed to initiate S1 connection. Attempting reconnection in 10 seconds
connect(): Connection refused
Failed to initiate S1 connection. Attempting reconnection in 10 seconds
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 55.2454 >= 24.0049. Starting new burst...
^CStopping ..
```

andrepuschmann, 2022-01-19:
Check your EPC config and IPs. Closing this. It's a config problem.

closed 2022-01-19 by andrepuschmann

fatafalihhilmi, 2022-10-13:
do you have a correct config? @andrepuschmann