srsran/srsRAN_4G #789: Importance of AMF, SQN and QCI in repeated authentication failures

almeyras, 2022-01-20:
Hello! After many tries, I keep getting "UL NAS: Authentication Failure" in my phone. I even got "UE is ECM IDLE" in one occasion. I'm using the default values of an oem milenage card with 4G support and opc. MCC and MNC are correct across all devices and apps (001 01). I'm using the default values of the card. APN matches too.

I wonder, what are the values at the end of every SIM entry in user_db.csv (AMF),(SQN),(QCI)? Do any of them have to be unique? There is not much explanation about those values in the docs.

Thank you.

zxmek, 2022-01-21:
may be you should check the  value of Auth algorithm?xor and mil? usim card should be mil.

stackprogramer, 2024-07-27:
Did understand what's three parameters and how we can set them?

stackprogramer, 2024-07-27:
Finally i found solution according link srsran cots ue 
The AMF, SQN, QCI and IP Alloc fields can be populated with the following values:

9000, 000000000000, 9, dynamic
https://docs.srsran.com/projects/4g/en/latest/app_notes/source/cots_ue/source/index.html