srsran/srsRAN_4G #790: How can i make srsRAN run with my Signal Hound SDR?

gaur-yogesh, 2022-01-21:
## Issue Description ##
[Describe the issue in detail]

## Setup Details ##
[Specify details of the test setup. This would help us reproduce the problem reliably]
e.g. Network configuration, Operation System, Hardware, RF front-end, library and driver versions

## Expected Behavior ##
[What you expect to happen]

## Actual Behaviour ##
[What happens instead e.g. error message]

## Steps to reproduce the problem ##
[Tell us how to reproduce this issue e.g. RF setup, application config files]

## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]

jtang613, 2022-01-27:
I do not personally own a SignalHound (donations welcome) but in theory, this should work with any device that is supported by SoapySDR provided the device meets the SRSRAN RF and bandwidth requirements - which the SignalHound appears to.
A quick search revealed at least one project aimed at SoapySDR support for the SignalHound at: https://github.com/SignalHound/soapy-bb60

andrepuschmann, 2022-01-27:
Thanks @jtang613 for the pointers. As you suggest, SoapySDR should work out of the box. Anyway, I don't think its an srsRAN issue. Closing it.

closed 2022-01-27 by andrepuschmann