srsran/srsRAN_4G #794: srsRAN UE on raspberry pi 4

rwangu07, 2022-01-24:
## Issue Description ##
I have a doubt if we could run srsRAN UE on a Raspberry Pi 4 while the EPC and ENB are running on another Raspberry Pi 4.

## Setup Details ##
[Specify details of the test setup. This would help us reproduce the problem reliably]
I'm using 2 Raspberry Pi 4 with Ubuntu 20.04. they are connected to a USRP B210 with UHD 3.15. 
The setup is identical on both the Pi's.
e.g. Network configuration, Operation System, Hardware, RF front-end, library and driver versions

## Expected Behavior ##
srsRAN UE should connect to srsRAN ENB.

## Actual Behaviour ##
srsRAN UE does not attach to the ENB. 
But when i install srsRAN on a Laptop and run srsRAN UE on it while the EPC and ENB are running on the raspberry Pi the system works.

## Steps to reproduce the problem ##
I have just used the default setup using the document on https://docs.srsran.com/en/latest/app_notes/source/pi4/source/index.html

## Additional Information ##
I'm really not sure if we can run the UE on a raspberry Pi, although I don't see any reason for it to not work or any documentation supporting my hypothesis.

andrepuschmann, 2022-02-14:
We don't support the UE on a Pi4. Running a full UE on the Pi4 is quite challenging as the requirements are higher than for the eNB.

closed 2022-02-14 by andrepuschmann

rwangu07, 2022-02-14:
Thank you for the response @andrepuschmann . I had kind of figured it out by now, was just not sure about the reason for it.

nomidaepapi, 2022-02-18:
Hey andre, may i ask what these additional requirements of the UE are? Are they hardware requirements specifically?

rwangu07, 2022-02-18:
Im not sure about the additional requirements of the UE, but yes you can run srsEPC on the Pi4

nomidaepapi, 2022-03-03:
We are trying to run the UE on an RPi but recently came across a post on GitHub which mentions that the UE function can’t be run on the RPi. We have noted that it runs on more powerful systems with larger processing capacity. But we want to be sure of this. There is no documentation that specifies the minimum requirements for the UE running on an SDR. Also, we would like to know what aspects of the UE build prevent it from running on a RPi, given that the eNB can. Our assumption was that the eNB would have a more computationally intensive function related to scheduling etc.

Some clarification on these points will really help.

coinoperated0428, 2022-04-24:
You might want to remove the Rasberry PI application notes or at least give us a warning is going work.

kbarlee, 2022-04-25:
> You might want to remove the Rasberry PI application notes or at least give us a warning it's not going work.

The eNB works per the cell configurations described in the app note. 5MHz cells are pretty stable when the Pi4 is overclocked to 2GHz as well - just requires a heatsink and fan to keep the thing cool.

coinoperated0428, 2022-04-26:
kbarlee - thanks, I was trying to use it as a multiple UE as well and I'm hearing that the UE is too computationally intensive for my PI4's. My experiments/experience is similar to those reported here.

armin485, 2024-10-16:
Hi. I'm trying to run srsue on an ARM SBC that is more powerful than RPi4 (8x core at 2GHz) but I cannot attach to eNB. I was wondering what are the requirements of UE that it cannot be run on small SBC? Where is exactly the point where UE needs high computation power?
In addition, would it be possible to run srsue on lower bandwidths? I couldn't find any config parameter to set it up.