srsran/srsRAN_4G #795: SrsUE with USRP B210 connecting to LTE but not switching to 5G in NSA mode

anasalbakri, 2022-01-31:
## Issue Description ##
[Describe the issue in detail]
I'm trying to connect USRP B210 with SrsUE on a linux PC to a real commercial 5G gNB NSA mode wireless. 
I have the EARFCN of the gNB as follows: 
LTE: EARFCN: 38675 corresponding to 2302.5 MHz (Band 40). 
NR Central (dl_nr_arfcn): 630666 corresponding to 3459.990 MHz (Band 78)
NR SSB (ssb_nr_arfcn): 630624 corresponding to 3459.360 MHz 
Also I'm using their unique sim card because their gNB can't work with virtual sims. 
I'm following the guide of 5G NSA Amari as it is the closest one to my use case:
 https://docs.srsran.com/en/latest/app_notes/source/5g_nsa_amari/source/index.html
 Though, unlike in the guide, I can't use external clock because it is wireless connection and also i don't have control over the commercial gNB. 

## Setup Details ##
[Specify details of the test setup. This would help us reproduce the problem reliably]
e.g. Network configuration, Operation System, Hardware, RF front-end, library and driver versions
I have made the following changes in srsue config file: 
[rf]
freq_offset = -4000
tx_gain = 90
rx_gain = 60
[rat.eutra]
dl_earfcn = 38675
[rat.nr]
 bands = 78
 #dl_earfcn = 630666
 nof_carriers = 1

## Expected Behavior ##
[What you expect to happen]
UE should connect to LTE then siwtch to 5G in NR RRC reconfiguration. 

## Actual Behaviour ##
[What happens instead e.g. error message]
UE is connecting indeed to LTE and getting an IP address but is not conencting to 5G. I chekced the log files and it appears UE is not trying to switch to 5G at all so could this be due to high UE capability requirement from gNB?
also, I'm getting the error: _Error predecoding CCD: Invalid combination of ports 2 and rx antennax 1_ 
I want to try to use another USRP to be able to do MIMO 2x2 with NSA mode to get rid of this error but is this possible with SrsUE?

I attached the log file with debug mode. 

This is the output I'm getting. 
sudo srsue
Couldn't open , trying /root/.config/srsran/ue.conf
Reading configuration file /root/.config/srsran/ue.conf...

Built in Release mode using commit 5275f3336 on branch master.

Opening 2 channels in RF device=default with args=default
Available RF device list: UHD  soapy 
Trying to open RF device 'UHD'
[INFO] [UHD] linux; GNU C++ version 9.3.0; Boost_107100; UHD_3.15.0.HEAD-0-gaea0e2de
[INFO] [LOGGING] Fastpath logging disabled at runtime.
Opening USRP channels=2, args: type=b200,master_clock_rate=23.04e6
[INFO] [UHD RF] RF UHD Generic instance constructed
[INFO] [B200] Detected Device: B210
[INFO] [B200] Operating over USB 3.
[INFO] [B200] Initialize CODEC control...
[INFO] [B200] Initialize Radio control...
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Asking for clock rate 23.040000 MHz... 
[INFO] [B200] Actually got clock rate 23.040000 MHz.
[INFO] [MULTI_USRP]     1) catch time transition at pps edge
[INFO] [MULTI_USRP]     2) set times next pps (synchronously)
RF device 'UHD' successfully opened
Waiting PHY to initialize ... done!
Attaching UE...
.
Found Cell:  Mode=TDD, PCI=42, PRB=25, Ports=2, CP=Normal, CFO=-1.9 KHz
[... 283 lines cut ...]
lte   42   -88  106 -1.6k |   1    6   1.0    13k    0%   22.4 |   1   177k    53k   11%
 nr    0     0    0   0.0 |   0    0   0.0    0.0    0%    0.0 |   0    0.0    0.0    0%
lte   42   -88  106 -1.6k |   1    5   1.0    13k    0%   22.4 |   1   171k    52k   12%
 nr    0     0    0   0.0 |   0    0   0.0    0.0    0%    0.0 |   0    0.0    0.0    0%
^CStopping ..
lte   42   -89  107 -1.6k |   0    5   1.0    12k    0%   21.9 |   1   166k    52k   10%
 nr    0     0    0   0.0 |   0    0   0.0    0.0    0%    0.0 |   0    0.0    0.0    0%
lte   42   -88  106 -1.6k |   1    5   1.0    58k    0%   21.9 |   1   161k    48k   10%
 nr    0     0    0   0.0 |   0    0   0.0    0.0    0%    0.0 |   0    0.0    0.0    0%
lte   42   -88  106 -1.7k |   0    5   1.0   2.2k    0%   21.9 |   1   155k    47k    8%
 nr    0     0    0   0.0 |   0    0   0.0    0.0    0%    0.0 |   0    0.0    0.0    0%
lte   42   -89  107 -1.7k |   1    5   1.0   2.6k    0%   21.9 |   1   149k    67k   16%
 nr    0     0    0   0.0 |   0    0   0.0    0.0    0%    0.0 |   0    0.0    0.0    0%
lte   42   -88  106 -1.6k |   0    5   1.0   3.5k    0%   21.9 |   1   143k    55k    8%
 nr    0     0    0   0.0 |   0    0   0.0    0.0    0%    0.0 |   0    0.0    0.0    0%
Couldn't stop after 5s. Forcing exit.
Killed

## Steps to reproduce the problem ##
[Tell us how to reproduce this issue e.g. RF setup, application config files]
sudo srsue

## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]
Thank you very much for the help.

andrepuschmann, 2022-02-14:
Thanks for the report. I am afraid the TDD support for LTE in srsUE hasn't been a focus for some time so there might be issues with this in the first place. But your console log seems ok, you get an IP. However, after the attach the network doesn't seem to continue with any 5G activation. Also the UE capability exchange isn't happening for 5G. Double check you're setting capabilities and RRC version correctly. Please also share PCAPs.

andrepuschmann, 2022-02-14:
Specifically, use:
```
[rrc]
release           = 15
```

anasalbakri, 2022-02-15:
Thank you very much for the reply. Yes, I noticed there is some limitations for 5G with srsue but it is worth trying :) 
I specified release 15 in rrc and also tried both UE category 4 and 5 with same results (though UE capability information message now is larger 44 bytes and was 19 bytes before). 
So, Indeed UE capability is not containing the EUTRA-NR capabilities which is very odd because the log clearly is saying otherwise: 

From log level info: 
_RRC    ] [I] SRB1 - Rx ueCapabilityEnquiry (16 B)
2022-02-15T16:25:48.797271 [RRC    ] [I] Including EUTRA-NR capabilities in UE Capability Info (31 B)
2022-02-15T16:25:48.797276 [RRC    ] [I] Including NR capabilities in UE Capability Info (12 B)
2022-02-15T16:25:48.797279 [RRC    ] [I] SRB1 - Tx ueCapabilityInformation (48 B)_

from log level debug which shows the full UE capability enquiry and information message. 
] _[D] Integrity check input msg (Bytes=20):
    0000: 03 38 20 1c 83 49 ca 73 60 a0 08 03 41 38 29 80
    0010: 9b 04 10 00
2022-02-15T16:09:04.827139 [RRC    ] [D] RX PDU, LCID: 1
2022-02-15T16:09:04.827158 [RRC    ] [D] SRB1 - Rx ueCapabilityEnquiry (19 B)
    0000: 38 20 1c 83 49 ca 73 60 a0 08 03 41 38 29 80 9b
    0010: 04 10 00
2022-02-15T16:09:04.827164 [RRC    ] [D] Content:
[
  {
    "DL-DCCH-Message": {
      "message": {
        "c1": {
          "ueCapabilityEnquiry": {
            "rrc-TransactionIdentifier": 0,
            "criticalExtensions": {
              "c1": {
                "ueCapabilityEnquiry-r8": {
                  "ue-CapabilityRequest": [
                    "eutra"
                  ],
                  "nonCriticalExtension": {
                    "nonCriticalExtension": {
                      "requestedFrequencyBands-r11": [
                        14,
                        40,
                        42
                      ],
                      "nonCriticalExtension": {
                        "requestReducedFormat-r13": "true",
                        "requestSkipFallbackComb-r13": "true",
                        "requestReducedIntNonContComb-r13": "true",
                        "nonCriticalExtension": {
                          "nonCriticalExtension": {
                            "requestedFreqBandsNR-MRDC-r15": "00803413829809b04100"
                          }
                        }
                      }
                    }
                  }
                }
              }
            }
          }
        }
      }
    }
  }
]
2022-02-15T16:09:04.827164 [RRC    ] [D] Preparing UE Capability Info
2022-02-15T16:09:04.827219 [RRC    ] [D] Requested Freq Bands NR MRDC R15 present
2022-02-15T16:09:04.827231 [RRC    ] [D] SRB1 - Tx ueCapabilityInformation (44 B)
    0000: 38 01 02 8d d8 00 00 9c 00 0f 30 20 80 00 01 00
    0010: 0a 04 00 a2 00 42 50 00 c0 0a a0 00 40 08 a4 80
2022-02-15T16:09:04.827232 [RRC    ] [D] Content:
[
  {
    "UL-DCCH-Message": {
      "message": {
        "c1": {
          "ueCapabilityInformation": {
            "rrc-TransactionIdentifier": 0,
            "criticalExtensions": {
              "c1": {
                "ueCapabilityInformation-r8": {
                  "ue-CapabilityRAT-ContainerList": [
                    {
                      "rat-Type": "eutra",
                      "ueCapabilityRAT-Container": "dd800009c000f3020800001000a0400a200425000c00aa0004008a480001000008254d0004d10000"
                    }
                  ]
                }
              }
            }
          }
        }
      }
    }
  }
]
2022-02-15T16:09:04.827235 [PDCP   ] [D] Integrity gen input: COUNT 4, Bearer ID 1, Direction Uplink
2022-02-15T16:09:04.827235 [PDCP   ] [D] Integrity gen key:
    0000: 4b 61 46 3c 44 f5 b7 d7 60 98 f7 8b ab e6 41 99
    0010: 65 c0 a7 97 84 7b 94 f9 2a a4 e3 ec 78 c2 f3 61
2022-02-15T16:09:04.827235 [PDCP   ] [D] Integrity gen input msg:
    0000: 04 38 01 02 8d d8 00 00 9c 00 0f 30 20 80 00 01
    0010: 00 0a 04 00 a2 00 42 50 00 c0 0a a0 00 40 08 a4
2022-02-15T16:09:04.827237 [PDCP   ] [D] MAC (generated)_

What are the other parameters in the configuration that is related to UE capability? 
when I try to set feature_group to any value, I get error: 
the argument ('0xe6041000') for option 'rrc.feature_group' is invalid

Also, is it possible to connect to 5G SA instead of NSA or it's not supported yet? 

I have attached log files of debug level and the PCAP in a zip file. 

At this point, I know for sure it has to do with UE capability message so I'm trying to look at the source code of rrc to understand what is happening
So, any guide toward where to look specifically is greatly appreciated. 

Thank you very much, 
[ue1.zip](https://github.com/srsran/srsRAN/files/8072956/ue1.zip)

anasalbakri, 2022-02-15:
One more information that could be helpful on what is the UE capability that is sufficient to do 5G reconfiguration. 
We were able to connect another SDR of a different software to the same gNB we are trying to connect o with srsue 
Here is the UE capability information message from the other device logs: 

{
          message c1: ueCapabilityInformation: {
            rrc-TransactionIdentifier 0,
            criticalExtensions c1: ueCapabilityInformation-r8: {
              ue-CapabilityRAT-ContainerList {
                {
                  rat-Type eutra,
                  ueCapabilityRAT-Container {
                    accessStratumRelease rel15,
                    ue-Category 4,
                    pdcp-Parameters {
                      supportedROHC-Profiles {
                        profile0x0001-r15 TRUE,
                        profile0x0002-r15 TRUE,
                        profile0x0003-r15 FALSE,
                        profile0x0004-r15 TRUE,
                        profile0x0006-r15 FALSE,
                        profile0x0101-r15 FALSE,
                        profile0x0102-r15 FALSE,
                        profile0x0103-r15 FALSE,
                        profile0x0104-r15 FALSE
                      }
                    },
                    phyLayerParameters {
                      ue-TxAntennaSelectionSupported FALSE,
                      ue-SpecificRefSigsSupported TRUE
                    },
                    rf-Parameters {
                      supportedBandListEUTRA {
                        {
                          bandEUTRA 40,
                          halfDuplex FALSE
                        }
                      }
                    },
                    measParameters {
                      bandListEUTRA {
                        {
                          interFreqBandList {
                            {
                              interFreqNeedForGaps TRUE
                            }
                          }
                        }
                      }
                    },
                    featureGroupIndicators  '5E001002'H /*
                       2: PUCCH 2a/2b, abs TPC, RB alloc type 1 for PDSCH, Periodic CQI/PMI/RI reporting on PUCCH: Modes 2-0 and 2-1
                       4: Short DRX cycle
                       5: Long DRX cycle, DRX Mac CE
                       6: Prioritized bit rate
                       7: RLC UM
                      20: SRB1 and SRB2 for DCCH + 8x DRB
                      31: MFBI
                    */,
                    interRAT-Parameters {
                    },
                    nonCriticalExtension {
                      phyLayerParameters-v920 {
                        enhancedDualLayerFDD-r9 supported,
                        enhancedDualLayerTDD-r9 supported
                      },
                      interRAT-ParametersGERAN-v920 {
                      },
                      csg-ProximityIndicationParameters-r9 {
                      },
[... 222 lines cut ...]
                                                        }
                                                      }
                                                    }
                                                  }
                                                }
                                              }
                                            }
                                          }
                                        }
                                      }
                                    }
                                  }
                                }
                              }
                            }
                          }
                        }
                      }
                    }
                  }
                }
              }
            }
          }
        }