srsran/srsRAN_4G #796: make test produces 8 failures pdcch_test

Wingqvist, 2022-02-01:
## Issue Description ##
I followed the guide to build srsRAN. When I run make test I get 8 errors regarding pdcch_test

## Setup Details ##
I have a fresh install of Ubuntu 21.10, running on legacy hardware. Specifically I use the Intel i7 2600k. 

## Expected Behavior ##
all tests should pass

## Actual Behaviour ##
99% tests passed, 12 tests failed out of 1463

Label Time Summary:
chest     =  31.44 sec*proc (22 tests)
fec       =   0.60 sec*proc (11 tests)
ldpc      =   0.01 sec*proc (1 test)
lib       = 270.50 sec*proc (1005 tests)
lte       = 244.56 sec*proc (942 tests)
nr        = 133.15 sec*proc (280 tests)
pdcp      =   0.36 sec*proc (3 tests)
phch      =  55.99 sec*proc (485 tests)
phy       = 253.55 sec*proc (1004 tests)
polar     =   0.00 sec*proc (1 test)
rlc       =  26.49 sec*proc (11 tests)
srsenb    =   8.79 sec*proc (10 tests)
srsue     =   1.12 sec*proc (3 tests)
turbo     =   0.59 sec*proc (9 tests)

Total Test time (real) = 411.50 sec

The following tests FAILED:
	251 - pdcch_test-n100-p1-f1-Sauto-R1 (Failed)
	252 - pdcch_test-n100-p1-f1-S15-R1 (Failed)
	254 - pdcch_test-n100-p1-f2-Sauto-R1 (Failed)
	255 - pdcch_test-n100-p1-f2-S15-R1 (Failed)
	257 - pdcch_test-n100-p1-f3-Sauto-R1 (Failed)
	258 - pdcch_test-n100-p1-f3-S15-R1 (Failed)
	260 - pdcch_test-n100-p2-f1-Sauto-R1 (Failed)
	261 - pdcch_test-n100-p2-f1-S15-R1 (Failed)
	263 - pdcch_test-n100-p2-f2-Sauto-R1 (Failed)
	264 - pdcch_test-n100-p2-f2-S15-R1 (Failed)
	266 - pdcch_test-n100-p2-f3-Sauto-R1 (Failed)
	267 - pdcch_test-n100-p2-f3-S15-R1 (Failed)
Errors while running CTest
make: *** [Makefile:148: test] Error 8

## Steps to reproduce the problem ##
Install everything using the guide and run the make test command

## Additional Information ##
I read in another post that there might be some issues related to older cpu architectures that are missing more recent instruction sets, which I think the 2600 do. If so, I would be happy to get this confirmed and some guidance on the implications of running this even with the failed tests. Would it be stable?

Wingqvist, 2022-02-01:
./lib/src/phy/phch/test/pdcch_test
params - pci=1; rnti=0x0046; cfi=2; nof_ports=1; cif_enabled=0; nof_prb=100; snr_db=+nan; repetitions=1; false_check=0;
test_case_1 - format Format0  - passed - 2.1 usec/encode; 10.1 usec/llr; 6.5 usec/decode; min_corr=0.620700; false_alarm_prob=0.000000;
test_case_1 - format Format1A - passed - 1.9 usec/encode; 9.2 usec/llr; 6.0 usec/decode; min_corr=0.615631; false_alarm_prob=0.000000;
test_case_1 - format Format1  - passed - 1.8 usec/encode; 9.2 usec/llr; 6.3 usec/decode; min_corr=0.597290; false_alarm_prob=0.000000;
-- Detected payload was not matched, repeating decode with INFO logs (n0: -15.0 dB, corr: 0.934715)
[INFO]: Decoded DCI: nCCE=2, L=0, format=Format2A, msg_len=48, mean=0.977104, crc_rem=0xbd1
Tx: rnti=0x0046; format=Format2A; L=0; ncce=2; payload=[00 01 01 00 01 01 00 01 01 00 01 00 01 00 01 00 00 00 00 01 00 00 00 01 00 01 01 00 00 01 00 00 00 01 00 00 00 00 00 01 00 00 00 01 00 01 00 01 ];
Rx: rnti=0x0046; format=Format2A; L=0; ncce=2; payload=[00 00 01 01 00 00 01 00 00 00 01 01 01 01 01 01 00 00 00 00 01 00 00 00 01 00 00 00 01 00 01 00 01 01 01 01 00 01 01 01 00 00 01 01 00 00 00 01 ];
test_case1:327: Assertion Failure: "payload_match"
/home/x/workspace/exjobb/srsRAN/lib/src/phy/phch/test/pdcch_test.c:415: Test case 1 failed
Error

Wingqvist, 2022-02-01:
This is my cpu info

cat /proc/cpuinfo
processor	: 0
vendor_id	: GenuineIntel
cpu family	: 6
model		: 42
model name	: Intel(R) Core(TM) i7-2600K CPU @ 3.40GHz
stepping	: 7
microcode	: 0x2f
cpu MHz		: 1600.000
cache size	: 8192 KB
physical id	: 0
siblings	: 8
core id		: 0
cpu cores	: 4
apicid		: 0
initial apicid	: 0
fpu		: yes
fpu_exception	: yes
cpuid level	: 13
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ht tm pbe syscall nx rdtscp lm constant_tsc arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf pni pclmulqdq dtes64 monitor ds_cpl est tm2 ssse3 cx16 xtpr pdcm pcid sse4_1 sse4_2 popcnt tsc_deadline_timer aes xsave avx lahf_lm epb pti ssbd ibrs ibpb stibp xsaveopt dtherm ida arat pln pts md_clear flush_l1d
bugs		: cpu_meltdown spectre_v1 spectre_v2 spec_store_bypass l1tf mds swapgs itlb_multihit
bogomips	: 6799.95
clflush size	: 64
cache_alignment	: 64
address sizes	: 36 bits physical, 48 bits virtual
power management:

processor	: 1
vendor_id	: GenuineIntel
cpu family	: 6
model		: 42
model name	: Intel(R) Core(TM) i7-2600K CPU @ 3.40GHz
stepping	: 7
microcode	: 0x2f
cpu MHz		: 2600.000
cache size	: 8192 KB
physical id	: 0
siblings	: 8
core id		: 1
cpu cores	: 4
apicid		: 2
initial apicid	: 2
fpu		: yes
fpu_exception	: yes
cpuid level	: 13
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ht tm pbe syscall nx rdtscp lm constant_tsc arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf pni pclmulqdq dtes64 monitor ds_cpl est tm2 ssse3 cx16 xtpr pdcm pcid sse4_1 sse4_2 popcnt tsc_deadline_timer aes xsave avx lahf_lm epb pti ssbd ibrs ibpb stibp xsaveopt dtherm ida arat pln pts md_clear flush_l1d
bugs		: cpu_meltdown spectre_v1 spectre_v2 spec_store_bypass l1tf mds swapgs itlb_multihit
bogomips	: 6799.95
clflush size	: 64
cache_alignment	: 64
address sizes	: 36 bits physical, 48 bits virtual
power management:

processor	: 2
vendor_id	: GenuineIntel
cpu family	: 6
model		: 42
model name	: Intel(R) Core(TM) i7-2600K CPU @ 3.40GHz
stepping	: 7
microcode	: 0x2f
cpu MHz		: 1600.000
cache size	: 8192 KB
physical id	: 0
siblings	: 8
core id		: 2
cpu cores	: 4
[... 123 lines cut ...]
vendor_id	: GenuineIntel
cpu family	: 6
model		: 42
model name	: Intel(R) Core(TM) i7-2600K CPU @ 3.40GHz
stepping	: 7
microcode	: 0x2f
cpu MHz		: 2431.107
cache size	: 8192 KB
physical id	: 0
siblings	: 8
core id		: 3
cpu cores	: 4
apicid		: 7
initial apicid	: 7
fpu		: yes
fpu_exception	: yes
cpuid level	: 13
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ht tm pbe syscall nx rdtscp lm constant_tsc arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf pni pclmulqdq dtes64 monitor ds_cpl est tm2 ssse3 cx16 xtpr pdcm pcid sse4_1 sse4_2 popcnt tsc_deadline_timer aes xsave avx lahf_lm epb pti ssbd ibrs ibpb stibp xsaveopt dtherm ida arat pln pts md_clear flush_l1d
bugs		: cpu_meltdown spectre_v1 spectre_v2 spec_store_bypass l1tf mds swapgs itlb_multihit
bogomips	: 6799.95
clflush size	: 64
cache_alignment	: 64
address sizes	: 36 bits physical, 48 bits virtual
power management:

andrepuschmann, 2022-02-02:
Thanks for the report. I agree we should not fail here. But I guess it's because you're using a very old CPU without AVX2 support. Some of the DSP code requires AVX2.