srsran/srsRAN_4G #797: Manual Camping of Phones

neilrep, 2022-02-02:
Issue Description:
Manual camping of all brand of phones after provisioning

We're running an EPC --> enodeB --> SDR (b210) --> duplexer --> antenna . A complete set-up for a box base station. SRSENB 20.10.1 , UHD 3.15.0 

Expected Behavior :
All brands of Phones can camp to the network

Actual Behavior: 

Not all phones can camp to the network automatically 
(Phones that cannot camp) :

 - Huwaei Y7a
 - Xiaomi Redmi note 9 Pro

 (Phones automatically camp either manual camping successful) :
- Realme C12
- Vivo Y15s

Also trying to exchange sims to each phones, and got a same scenario. When I tried to restart the phones (Huwaei & Xiaomi), both unable to connect the network

If you have any request of log files and other configuration files, please feel free to comment on this forum for me to get it posted. Thank you so much all for your future support.

Additional Information :
([UE's_Cannot_Latch.logs.txt](https://github.com/srsran/srsRAN/files/7983796/UE.s_Cannot_Latch.logs.txt))
[Brand_of_UE.can't.latch.txt](https://github.com/srsran/srsRAN/files/7983798/Brand_of_UE.can.t.latch.txt)
[pcap.txt](https://github.com/srsran/srsRAN/files/7983828/pcap.txt)

meremortals70, 2022-02-02:
Have a look at the UE mode of operation. A lot of cheaper phones are "voice centric" when the network does not support voice the UE won't complete the attach.

neilrep, 2022-02-03:
sorry for late reply @meremortals70 , your opinion is highly appreciated, thanks. As I checked, I compare traces from phones that cannot attached and phones that  are automatically attached. see attached file below.

Have you ever encountered this kind of situation? if ever, what should I do?

[Phone_pcap_traces.zip](https://github.com/srsran/srsRAN/files/7993068/Phone_pcap_traces.zip)

neilrep, 2022-02-21:
My apologies, I attached no content pcap traces on my last reply. I replaced new traces (see attached below) as I run through the ticket, I've got the same issue on that sequence. 

[Phone_pcap_traces.zip](https://github.com/srsran/srsRAN/files/8105411/Phone_pcap_traces.zip)

neilrep, 2022-02-21:
My apologies, I attached no content pcap traces on my last reply. I replaced new traces (see attached below) as I run through the ticket, I've got the same issue on that sequence.

[Phone_pcap_traces.zip](https://github.com/srsran/srsRAN/files/8105498/Phone_pcap_traces.zip)
.

taesokjoo, 2022-04-01:
Hi,
Band locking is easier.
And I would check that IMSI  ....1181 configured for IMS as well.