srsran/srsRAN_4G #798: Connecting 2 USRPs B210 together to enable Mimo 2x2 in 5G NSA

anasalbakri, 2022-02-04:
## Issue Description ##
I'm trying to connect two USRP B210 boards together to enable MIMO 2x2 in 5G NSA connection to gnb.

I've checked on UHD website about how to do this step and it appears that I need to redefine channel mapping to enable the necessary 4 RX channels out of the two usrps. 
So  is this possible with srsRAN and what are the steps to achieve that?
Otherwise if not possible, will a UHD script that redefine channel mapping contradict with srs code? 

## Setup Details ##
using srsue with 2 usrps B210 t connect to commercial 5g gNB base station. 

## Expected Behavior ##
[What you expect to happen]

## Actual Behaviour ##
when I just run srsue normally while connecting both devices and I set paramneter nof_antennas = 2  I receive the error  
_[ERROR] [UHD RF] LookupError: IndexError: multi_usrp: RX channel 2 out of range for configured RX frontends
/home/ets/srsRAN/lib/src/phy/rf/rf_uhd_imp.cc:930: uhd_init failed, freeing..._

## Steps to reproduce the problem ##
sudo srsue

## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]

andrepuschmann, 2022-02-14:
Hey, first of all we don't support MIMO in the 5G branch. So what you're trying to achieve is not possible. But the idea of combining 2x B210 to do NSA mode should be possible, yes.

I am closing the issue, please feel free to reopen if you think it's needed.

closed 2022-02-14 by andrepuschmann

hayyanali-cmd, 2023-01-27:
hi @andrepuschmann 
you said "combining 2x B210 to do NSA mode should be possible, yes".
could i ask you how to define the two devices in the enb.conf?

JitheshSuresh, 2024-03-06:
Hello, 
Is there any script / option available now to connect two B210 USRPs for a SA network?