srsran/srsRAN_4G #802: srs support for adalmpluto with soapysdr

eljuligallego, 2022-02-07:
## Issue Description ##
Hi, I know the support for the analog devices is limited due timestamp not implemented with analog devices drivers but from  other issue I know there is some test than can be made with this devices like cellsearch(it's working in my case and cell are detected) and PDSCH PSSCH examples but the last ones are not working due sample rate error.

## Setup Details ##
I have ubuntu 20 with the last srsRAN version installed from source and soapysdr witha soapysdrpluto installed and try to run the pssch_ue and pdsch_enodeb example  from build/lib/examples folder:

## Actual Behaviour ##
the output from running both example are the same:

Soapy has found device #0: device=plutosdr, driver=plutosdr, uri=usb:3.4.5, 
Soapy has found device #1: device=plutosdr, driver=plutosdr, uri=ip:pluto.local, 
Selecting Soapy device: 0
Setting up Rx stream with 1 channel(s)
[INFO] Using format CF32.
[INFO] Auto setting Buffer Size: 32768
[INFO] Set MTU Size: 32768
Setting up Tx stream with 1 channel(s)
[INFO] Using format CF32.
[INFO] Has direct TX copy: 1
[INFO] Auto setting Buffer Size: 32768
[INFO] Set MTU Size: 32768
Available device sensors: 
 - xadc_temp0
 - xadc_voltage0
 - xadc_voltage1
 - xadc_voltage2
 - xadc_voltage3
 - xadc_voltage4
 - xadc_voltage5
 - xadc_voltage6
 - xadc_voltage7
 - xadc_voltage8
 - adm1177_current0
 - adm1177_voltage0
 - ad9361-phy_temp0
 - ad9361-phy_voltage2
Available sensors for Rx channel 0: 
State of gain elements for Rx channel 0 (AGC supported):
 - PGA: 71.00 dB
State of gain elements for Tx channel 0 (AGC not supported):
 - PGA: 80.00 dB
Rx antenna set to A_BALANCED
Tx antenna set to A
RF device 'soapy' successfully opened
Setting sampling rate 5.76 MHz
/home/julian/Documents/SDR/srsRAN/lib/examples/pdsch_enodeb.c:796: Could not set sampling rate

the sample rate could not be set. how should I deal with this?.
what could be done with my setup "out of the box"?.

andrepuschmann, 2022-02-14:
You need to set the number of PRBs to 6 to use a sample rate of 1.92Msps. Hope that helps. Feel free to reopen if you still see the issue.

cloudwiser, 2023-01-12:
@andrepuschmann Andre - I have the same configuration using Pluto & Soapy - I found, with pdsch_ue, that the sample rate requested and that actually set in AD9363 will differ by 2 Hz hence the strict srate comparison fails. Allowing this 2 Hz delta meant the subsequent decodes worked as expected. I'll test with pssch_ue as well and raise a PR with my fix if it helps?