srsran/srsRAN_4G #805: RAR not transmitted within the RA Response window

gaurishdotkdm, 2022-02-10:
**## Issue Description ##**
srsue complains - immediately after Random Access Complete - "Scheduling request failed: releasing RRC connection".
srenb prints "Could not transmit RAR within the window (RA=2261, Window=[2264, 2274), RAR=2274"

**## Setup Details ##**
- 2 x RF-frontends - USRP X300
- 2 x PCs with a Linux based OS and identical configuration (below) - srsRAN installed on both PCs

 Operating System: Ubuntu 20.04.3 LTS
   Kernel: Linux 5.4.0-99-generic
     Architecture: x86-64

- On machine 1, in one console, run srsEPC - in another console, run srsENB
- On machine 2, run srsUE

- RF front-end driver: 3.15.0

**## Expected Behavior ##**
srsUE on (PC2/ USRP2) should attach to the (srsENB + srsEPC) radiated from (PC1/ USRP1)

**## Actual Behaviour ##**

-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=
ENB ==>
-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=

-- screen display after executing command "srsenb"

```
==== eNodeB started ===
Type <t> to view trace
Setting frequency: DL=1930.0 Mhz, UL=1850.0 MHz for cc_idx=0 nof_prb=50
RACH:  tti=2261, cc=0, preamble=17, offset=1, temp_crnti=0x46
RACH:  tti=2261, cc=0, preamble=18, offset=23, temp_crnti=0x47
SCHED: Could not transmit RAR within the window (RA=2261, Window=[2264, 2274), RAR=2274
Disconnecting rnti=0x47.
```

-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=

-- enb.log
 
```
2022-02-09T22:31:31.970238 [PHY    ] [I] [ 2262] PRACH: cc=0, 0/2, preamble=17, offset=1.0 us, peak2avg=154.5, max_offset=30.0 us
2022-02-09T22:31:31.970241 [PHY    ] [I] [ 2262] PRACH: cc=0, 1/2, preamble=18, offset=12.4 us, peak2avg=264.9, max_offset=30.0 us

  2022-02-09T22:31:31.970423 [MAC    ] [I] [ 2261] SCHED: New PRACH tti=2261, preamble=17, temp_crnti=0x46, ta_cmd=1, msg3_size=7

2022-02-09T22:31:31.970423 [MAC    ] [I] [ 2261] RACH:  tti=2261, cc=0, preamble=17, offset=1, temp_crnti=0x46
2022-02-09T22:31:31.971252 [MAC    ] [I] [ 2262] SCHED: RAR, ra-rnti=2, rbgs=[0, 1), dci=(2,0), msg3 grants=[{c-rnti=0x46, rba=102, mcs=0}]

2022-02-09T22:31:31.979082 [MAC    ] [I] [ 2269] SCHED: New PRACH tti=2261, preamble=18, temp_crnti=0x47, ta_cmd=23, msg3_size=7
2022-02-09T22:31:31.979083 [MAC    ] [I] [ 2269] RACH:  tti=2261, cc=0, preamble=18, offset=23, temp_crnti=0x47

2022-02-09T22:31:31.979244 [MAC    ] [W] [ 2270] SCHED: Could not transmit RAR within the window (RA=2261, Window=[2264, 2274), RAR=2274

2022-02-09T22:31:32.006248 [RRC    ] [I] Rx SRB0 PDU, rnti=0x46 - rrcConnectionRequest (6 B)
2022-02-09T22:31:32.006492 [RRC    ] [I] Tx SRB0 PDU, rnti=0x46 - rrcConnectionSetup (20 B)
```

-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=
-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=
-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=
-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=

UE ==>

-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=
-- screen display after executing command "srsue"

[... 435 lines cut ...]
 }
}
     }
   }
 }
]

2022-02-09T22:31:32.014091 [RRC    ] [I] Finished Connection Setup successfully
 
2022-02-09T22:31:46.797740 [NAS    ] [W] Timer T3410 expired after attach attempt 1/5: starting T3411
```

-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=
-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=
-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=
-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=

**## Steps to reproduce the problem ##**

1. start srsepc on (PC1/ USRP1)
2. start srsenb on (PC1/ USRP1)
3. start srsue on (PC2/ USRP2)

**## Conf files Attached ##**
[conf_files.zip](https://github.com/srsran/srsRAN/files/8037438/conf_files.zip)

jtang613, 2022-02-10:
This appears to be the same issue as in https://github.com/srsran/srsRAN/issues/770 and https://github.com/srsran/srsRAN/issues/803 
I'd consider this to now be a verified issue. Issue 803 has a code snippet that claims to resolve this issue on 19.x, but the affected code has changed quite a bit since then and it's not obvious (to me) how to apply that particular fix to the current master branch.

gaurishdotkdm, 2022-02-11:
@FabianEckermann Hi Fabian, would you be able to guide here? I see that you raised similar concern earlier https://lists.srsran.com/pipermail/srsran-users/2017-July/001039.html

gaurishdotkdm, 2022-02-11:
@jtang613 Thanks for your response and suggestions. I checked your ticket at https://github.com/srsran/srsRAN/issues/803. I see that your srsUE fails to complete RA procedure.

I double checked my log. My srsUE is able to transmit _RRC Connection Request_ and receive _RRC Connection Setup_. However it failed further down to transmit _RRC Connection Setup Complete_. So code change at  (https://github.com/srsran/srsRAN/issues/770) may not be applicable to me.

ismagom, 2023-11-06:
Hi all, the issue here is time advance misconfiguration. Due to offsets in the TX and RX paths of the USRPs, the PRACH preamble can arrive before the start of the reception window. When this happens, the detected preamble is either incorrect or with a very high TA. Both result in either undetected RAR or incorrect Msg3. 

We calibrated the time advance for some UHD versions. If using a very old (or very new one) this might not be accurate. 

The solution to the issue is to use the parameter `time_adv_nsamples` in the `rf` section to set it to a value such that the detected preamble is correct and the TA is in the range eg 2-10 us.

closed 2023-11-06 by ismagom

MarkusKluegel, 2023-11-07:
Thanks for clarifying, @ismagom. Is there a way to identify from the logs whether the timing advance is configured too low/high? Or does it need to be found by trial and error? Plus, should the TA be configured on the base station or the UE side?

ismagom, 2023-11-07:
Note the `time_adv_nsamples` is not the same as the `TA` cmd sent by the RAR. The first one is the offset between TX and RX paths at the UE, the later is the offset between the reception of the PRACH and the start of the PRACH window at the eNB. 

It's more trial and error, since the value you see in the `TA` at the eNB is incorrect if the PRACH arrives *before* the start of the window.

MarkusKluegel, 2023-11-08:
Yes, okay. I was also considering the the `time_alignment_calibration` parameter on the eNB, but I see that's not what you were referring to. So I'll try to alter the `time_adv_samples` at the UE side only. If I understand correctly, higher values create more delay and therefore will cause the signal to arrive later at the eNB.

MarkusKluegel, 2023-11-14:
I can confirm that adaptation of the `time_adv_samples` on the UE side solves the problem. I started with `10us * 11.52e6 samples/sec = 115 samples`, then increased in steps of 5 while observing the reported SNR during PRACH. A good final configuration for me was 150 samples, which corresponds to about 13us.

mochammadfikri28, 2025-07-31:
@MarkusKluegel okay, i will try it