srsran/srsRAN_4G #810: srsUE can't attach commercial network with LimeSDR-USB

PYH2017, 2022-02-19:
## Issue Description ##
I'm trying to connect LimeSDR-USB with srsUE on a linux PC to a real commercial LTE network, using a pcsc card reader to read a real commercial sim card. srsUE can find the cell and RRC connections can be implemented, but UE can't attach and get the IP address

## Setup Details ##
srsRAN version 19.12
Soapy version v0.7.2
LimeSuiteGUI version 20.01.0

The UE configuration is as follows:
[rf]
dl_earfcn = 300
freq_offset = 0
tx_gain = 60
rx_gain = 60
device_name = soapy
device_args = rxant=LNAW,txant=BAND1
[usim]
mode = pcsc
algo = xor
opc  = 63BFA50EE6523365FF14C1F45F88737D
k    = 00112233445566778899aabbccddeeff
imsi = 001010123456789
imei = 353490069873319
[rrc]
ue_category       = 4
release           = 10

[ueconf.txt](https://github.com/srsran/srsRAN/files/8101585/ueconf.txt)
[ue.log](https://github.com/srsran/srsRAN/files/8101577/ue.log)

## Expected Behavior ##
[What you expect to happen]

## Actual Behaviour ##
Found Cell:  Mode=FDD, PCI=435, PRB=75, Ports=1, CFO=-0.8 KHz
[INFO] RX LPF configured
Set Rx bandwidth to 11.52 MHz
[INFO] Rx calibration finished
[INFO] Filter calibrated. Filter order-4th, filter bandwidth set to 11.52 MHz.Real pole 1st order filter set to 2.5 MHz. Preemphasis filter not active
[INFO] TX LPF configured
Set Tx bandwidth to 11.52 MHz
[INFO] Filter calibrated. Filter order-2nd, set to 50 MHz
[INFO] TX LPF configured
Set Tx bandwidth to 50.00 MHz
[INFO] Tx calibration finished
RF status: O=0, U=0, L=1
Found PLMN:  Id=46011, TAC=30090
Random Access Transmission: seq=2, ra-rnti=0x2
LRandom Access Transmission: seq=4, ra-rnti=0x2
Random Access Complete.     c-rnti=0x768a, ta=0
RRC Connected
RF status: O=0, U=0, L=2
Scheduling request failed: releasing RRC connection...
LRandom Access Transmission: seq=31, ra-rnti=0x2
LRandom Access Transmission: seq=7, ra-rnti=0x2
LRandom Access Transmission: seq=14, ra-rnti=0x2
LRandom Access Transmission: seq=20, ra-rnti=0x2
LRandom Access Transmission: seq=30, ra-rnti=0x2
LRandom Access Transmission: seq=10, ra-rnti=0x2
LRandom Access Transmission: seq=20, ra-rnti=0x2
LRandom Access Transmission: seq=19, ra-rnti=0x2
LRandom Access Transmission: seq=1, ra-rnti=0x2
LRandom Access Transmission: seq=6, ra-rnti=0x2
Warning: Detected Radio-Link Failure
RF status: O=0, U=0, L=11
RF status: O=0, U=0, L=2
RRC IDLE
Failed to Attach

## Steps to reproduce the problem ##
[Tell us how to reproduce this issue e.g. RF setup, application config files]

## Additional Information ##
In order to set the dl_earfcn correctly, I use a commercial cat4 module with the  same sim card and get the serving cell using AT  commands.

sonium-labs, 2022-03-03:
Hi, I'm by no means an expert but I've spent a good deal of time recently looking into the RA/RRC system within srsUE. It appears as though you've hit the maximum transmission count for RACH (which is 10), without receiving a Random Access Response (RAR), which contains information such as timing advance, temp C-RNTI and others. When RAR is received, the UE then sends MSG3 and continues with the attach sequence. However, after 10 RACH transmissions without an RAR, the system checks for whether or not any of the relevant timers are still running. If it detects that no timers are ongoing, it then triggers an RLF.

While I don't quite have the expertise to know _why_ this is occurring, the above describes the nature of the issue you're experiencing. I'm also working on getting srsUE synced up with a real eNB (see #816), so if you'd like to compare notes we can correspond on the forum, or you can shoot me an email at the address in my profile.

Best of luck!

sunbytez, 2022-07-01:
Same Problem here, is there any news?