srsran/srsRAN_4G #817: Connecting 2 USRP B210 in 5G NSA

hawk933, 2022-02-28:
## Issue Description ##
I am trying to connect two USRP B210 boards together to run a base station in 5G NSA mode to GNB. I found information on the UHD website on how to access the USRP serial number. 

## Setup Details ##
I am using srsENB with 2 USRP b210 on Ubuntu 20.04 operating system to run a 5G NSA base station.

## Expected Behavior ##
Run a 5G NSA base station

## Actual Behaviour ##
After the launch, the following log is displayed in the terminal:

> 
> sudo UHD_IMAGES_DIR=/usr/share/uhd/images srsenb
> ---  Software Radio Systems LTE eNodeB  ---
> 
> Couldn't open , trying /root/.config/srsran/enb.conf
> Reading configuration file /root/.config/srsran/enb.conf...
> Couldn't open sib.conf, trying /root/.config/srsran/sib.conf
> Couldn't open rr.conf, trying /root/.config/srsran/rr.conf
> Couldn't open rb.conf, trying /root/.config/srsran/rb.conf
> WARNING: cpu0 scaling governor is not set to performance mode. Realtime processing could be compromised. Consider setting it to performance mode before running the application.
> 
> Built in Release mode using commit 5275f3336 on branch master.
> 
> Opening 2 channels in RF device=default with args=serial = F5D83D, serial = F5D80E
> Available RF device list: UHD 
> Trying to open RF device 'UHD'
> [INFO] [UHD] linux; GNU C++ version 9.3.0; Boost_107100; UHD_3.15.0.0-74-ge35f66e8
> [INFO] [LOGGING] Fastpath logging disabled at runtime.
> Opening USRP channels=2, args: serial=F5D80E,type=b200,master_clock_rate=23.04e6
> [INFO] [UHD RF] RF UHD Generic instance constructed
> [INFO] [B200] Detected Device: B210
> [INFO] [B200] Operating over USB 3.
> [INFO] [B200] Initialize CODEC control...
> [INFO] [B200] Initialize Radio control...
> [INFO] [B200] Performing register loopback test... 
> [INFO] [B200] Register loopback test passed
> [INFO] [B200] Performing register loopback test... 
> [INFO] [B200] Register loopback test passed
> [INFO] [B200] Asking for clock rate 23.040000 MHz... 
> [INFO] [B200] Actually got clock rate 23.040000 MHz.
> [INFO] [MULTI_USRP]     1) catch time transition at pps edge
> [INFO] [MULTI_USRP]     2) set times next pps (synchronously)
> RF device 'UHD' successfully opened
> 
> ==== eNodeB started ===
> Type <t> to view trace
> Starting plot for worker_id=0
> Creating plot window 'srsENB PCI 1'...
> QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
> Setting frequency: DL=2680,0 Mhz, UL=2560,0 MHz for cc_idx=0 nof_prb=25
> Setting frequency: DL=4400,0 Mhz, UL=4400,0 MHz for cc_idx=1 nof_prb=25
> [INFO] [UHD RF] Tx while waiting for EOB, timed out... 43.6126 >= 43.6124. Starting new burst...
> [INFO] [UHD RF] Tx while waiting for EOB, timed out... 128.371 >= 128.371. Starting new burst...
> [INFO] [UHD RF] Tx while waiting for EOB, timed out... 128.99 >= 128.371. Starting new burst...
> [INFO] [UHD RF] Tx while waiting for EOB, timed out... 134.205 >= 134.205. Starting new burst...
> [INFO] [UHD RF] Tx while waiting for EOB, timed out... 147.278 >= 147.278. Starting new burst...

At a frequency of 2680.0 Mhz, the analyzer is empty, at a frequency of 4400.0 Mhz there is radiation

## Steps to reproduce the problem ##
I connected two USRP B210 boards,
> sudo UHD_IMAGES_DIR=/usr/share/uhd/images uhd_find_devices
> [INFO] [UHD] linux; GNU C++ version 9.3.0; Boost_107100; UHD_3.15.0.0-74-ge35f66e8
> [INFO] [B200] Loading firmware image: /usr/share/uhd/images/usrp_b200_fw.hex...
> --------------------------------------------------
> -- UHD Device 0
> --------------------------------------------------
> Device Address:
>     serial: F5D80E
>     name: 
>     product: B210
>     type: b200
> 
> 
> --------------------------------------------------
> -- UHD Device 1
> --------------------------------------------------
> Device Address:
>     serial: F5D83D
>     name: 
>     product: B210
>     type: b200
    
Specified arguments in the env.conf configuration:
> device_args = serial = F5D83D, serial = F5D80E
Received with the command:
> sudo UHD_IMAGES_DIR=/usr/share/uhd/images uhd_find_devices

Added the code responsible for GNB to rr.conf:

> nr_cell_list =
> (
> {
> rf_port = 1;
> cell_id =0x02;
> so = 0x0007;
> pci = 500;
> root_seq_id = 204;
> dl_arfcn =693334;
> band = 79;
> }
> );
> 

## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]

hawk933, 2022-02-28:
sudo srsepc

[image]

hawk933, 2022-02-28:
sudo UHD_IMAGES_DIR=/usr/share/uhd/images srsenb

[image]

hawk933, 2022-02-28:
LTE 2680.0 MHz:
[image]
USRP B210:
[image]

hawk933, 2022-02-28:
5G NSA 4400.0 MHz:
[image]

USRP b210:
[image]

hawk933, 2022-02-28:
LTE mode without making changes to nr_cell_list works fine, the SIM card is authorized in phones based on Android and iOS

joelsernamoreno, 2022-05-05:
Hi @hawk933 !

Did you manage to fix this problem? I'm trying to get this now

Thanks

hawk933, 2022-05-24:
no, it didn’t work out, as I understand it, there is no way to implement it on two USRP b210

closed 2022-08-01 by hawk933

hayyanali-cmd, 2023-01-27:
hi, is there any updates on your idea, i am trying to connect two usrp b210 to have NSA 5G netwrok. Is it possible or not !

hayyanali-cmd, 2023-01-27:
hi @joelsernamoreno 
did you solve this issue?